openwrt/packages #4258: adblock-[2.5.1]"can not load full block lists on my wndr4300 router"

jack338c, 2017-04-13:
Maintainer: Dirk Brenken <dev@brenken.org>
Environment: LEDE Reboot SNAPSHOT r3956-ffd055d on Router Wndr4300

Description:

# Here are logs of adblock and system logs of router,someone tell me how to solve this ,thank you

Wed Apr 12 22:09:09 2017 user.notice adblock-[2.5.1] info : start adblock processing ...
Wed Apr 12 22:09:09 2017 user.notice adblock-[2.5.1] debug: name: adaway, enabled: 1, backup: 1, dns: dnsmasq, fetch: /usr/bin/wget, memory: 125084
Wed Apr 12 22:09:11 2017 user.notice adblock-[2.5.1] debug: name: adaway, mode: backup, count: 379, in_rc: 0, out_rc: 0
Wed Apr 12 22:09:11 2017 user.notice adblock-[2.5.1] debug: name: adguard, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:09:11 2017 user.notice adblock-[2.5.1] debug: name: blacklist, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:09:11 2017 user.notice adblock-[2.5.1] debug: name: disconnect, enabled: 1, backup: 1, dns: dnsmasq, fetch: /usr/bin/wget, memory: 125084
Wed Apr 12 22:09:21 2017 user.notice adblock-[2.5.1] debug: name: disconnect, mode: backup, count: 5119, in_rc: 0, out_rc: 0
Wed Apr 12 22:09:22 2017 user.notice adblock-[2.5.1] debug: name: dshield, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:09:22 2017 user.notice adblock-[2.5.1] debug: name: feodo, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:09:22 2017 user.notice adblock-[2.5.1] debug: name: hphosts, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:09:22 2017 user.notice adblock-[2.5.1] debug: name: malware, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:09:22 2017 user.notice adblock-[2.5.1] debug: name: malwarelist, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:09:22 2017 user.notice adblock-[2.5.1] debug: name: openphish, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:09:22 2017 user.notice adblock-[2.5.1] debug: name: palevo, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:09:22 2017 user.notice adblock-[2.5.1] debug: name: ransomware, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:09:22 2017 user.notice adblock-[2.5.1] debug: name: reg_ch, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:09:22 2017 user.notice adblock-[2.5.1] debug: name: reg_pl, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:09:22 2017 user.notice adblock-[2.5.1] debug: name: reg_ro, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:09:22 2017 user.notice adblock-[2.5.1] debug: name: reg_ru, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:09:22 2017 user.notice adblock-[2.5.1] debug: name: securemecca, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:09:22 2017 user.notice adblock-[2.5.1] debug: name: shalla, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:09:22 2017 user.notice adblock-[2.5.1] debug: name: spam404, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:09:22 2017 user.notice adblock-[2.5.1] debug: name: sysctl, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:09:22 2017 user.notice adblock-[2.5.1] debug: name: whocares, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:09:22 2017 user.notice adblock-[2.5.1] debug: name: winspy, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:09:22 2017 user.notice adblock-[2.5.1] debug: name: winhelp, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:09:22 2017 user.notice adblock-[2.5.1] debug: name: yoyo, enabled: 1, backup: 1, dns: dnsmasq, fetch: /usr/bin/wget, memory: 125084
Wed Apr 12 22:09:26 2017 user.notice adblock-[2.5.1] debug: name: yoyo, mode: backup, count: 2319, in_rc: 0, out_rc: 0
Wed Apr 12 22:09:26 2017 user.notice adblock-[2.5.1] debug: name: zeus, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:09:31 2017 user.notice adblock-[2.5.1] info : block lists with overall 5549 domains loaded successfully (LEDE Reboot SNAPSHOT r3956-ffd055d)

# system logs of router

Wed Apr 12 22:08:58 2017 user.notice adblock-[2.5.1] debug: name: disconnect, mode: backup, count: 5119, in_rc: 0, out_rc: 0
Wed Apr 12 22:08:58 2017 user.notice adblock-[2.5.1] debug: name: dshield, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:08:58 2017 user.notice adblock-[2.5.1] debug: name: feodo, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:08:58 2017 user.notice adblock-[2.5.1] debug: name: hphosts, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:08:58 2017 user.notice adblock-[2.5.1] debug: name: malware, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:08:58 2017 user.notice adblock-[2.5.1] debug: name: malwarelist, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:08:58 2017 user.notice adblock-[2.5.1] debug: name: openphish, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:08:58 2017 user.notice adblock-[2.5.1] debug: name: palevo, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:08:58 2017 user.notice adblock-[2.5.1] debug: name: ransomware, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:08:58 2017 user.notice adblock-[2.5.1] debug: name: reg_ch, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:08:58 2017 user.notice adblock-[2.5.1] debug: name: reg_pl, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:08:58 2017 user.notice adblock-[2.5.1] debug: name: reg_ro, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:08:58 2017 user.notice adblock-[2.5.1] debug: name: reg_ru, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:08:58 2017 user.notice adblock-[2.5.1] debug: name: securemecca, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:08:58 2017 user.notice adblock-[2.5.1] debug: name: shalla, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:08:58 2017 user.notice adblock-[2.5.1] debug: name: spam404, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:08:58 2017 user.notice adblock-[2.5.1] debug: name: sysctl, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:08:59 2017 user.notice adblock-[2.5.1] debug: name: whocares, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:08:59 2017 user.notice adblock-[2.5.1] debug: name: winspy, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:08:59 2017 user.notice adblock-[2.5.1] debug: name: winhelp, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:08:59 2017 user.notice adblock-[2.5.1] debug: name: yoyo, enabled: 1, backup: 1, dns: dnsmasq, fetch: /usr/bin/wget, memory: 125084
Wed Apr 12 22:09:03 2017 user.notice adblock-[2.5.1] debug: name: yoyo, mode: backup, count: 2319, in_rc: 0, out_rc: 0
Wed Apr 12 22:09:03 2017 user.notice adblock-[2.5.1] debug: name: zeus, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:09:04 2017 daemon.info dnsmasq[7661]: exiting on receipt of SIGTERM
Wed Apr 12 22:09:08 2017 user.notice adblock-[2.5.1] info : block lists with overall 5549 domains loaded successfully (LEDE Reboot SNAPSHOT r3956-ffd055d)
Wed Apr 12 22:09:09 2017 daemon.info dnsmasq[8049]: started, version 2.77test4 cachesize 150
Wed Apr 12 22:09:09 2017 daemon.info dnsmasq[8049]: DNS service limited to local subnets
Wed Apr 12 22:09:09 2017 daemon.info dnsmasq[8049]: compile time options: IPv6 GNU-getopt no-RTC no-DBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify
Wed Apr 12 22:09:09 2017 daemon.info dnsmasq-dhcp[8049]: DHCP, IP range 192.168.11.100 -- 192.168.11.249, lease time 12h
Wed Apr 12 22:09:09 2017 daemon.info dnsmasq[8049]: using local addresses only for domain test
[... 94 lines cut ...]
Wed Apr 12 22:09:22 2017 user.notice adblock-[2.5.1] debug: name: malware, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:09:22 2017 user.notice adblock-[2.5.1] debug: name: malwarelist, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:09:22 2017 user.notice adblock-[2.5.1] debug: name: openphish, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:09:22 2017 user.notice adblock-[2.5.1] debug: name: palevo, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:09:22 2017 user.notice adblock-[2.5.1] debug: name: ransomware, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:09:22 2017 user.notice adblock-[2.5.1] debug: name: reg_ch, mode: remove, count: 0, in_rc: 0, out_rc: 0
Wed Apr 12 22:09:22 2017 user.notice adblock-[2.5.1] debug: name: reg_pl, mode: remove, count: 0, in_rc: 0, out_rc: 0
[... cut ...]
Wed Apr 12 22:09:28 2017 authpriv.info dropbear[8260]: Child connection from 192.168.11.179:32769
Wed Apr 12 22:09:30 2017 authpriv.notice dropbear[8260]: Password auth succeeded for 'root' from 192.168.11.179:32769
Wed Apr 12 22:09:31 2017 user.notice adblock-[2.5.1] info : block lists with overall 5549 domains loaded successfully (LEDE Reboot SNAPSHOT r3956-ffd055d)
Wed Apr 12 22:09:32 2017 daemon.info dnsmasq[8314]: started, version 2.77test4 cachesize 150
Wed Apr 12 22:09:32 2017 daemon.info dnsmasq[8314]: DNS service limited to local subnets

dibdot, 2017-04-13:
@jack338c Maybe I miss something ... what's your problem actually?
To enable/disable blocklists edit /etc/config/adblock (via ssh session or via LUCI app). BTW, release 2.6.0 is the current version.

jack338c, 2017-04-13:
sorry,I did tell in details
please find above log,it should load lists 379+5119+2319=7817 (adaway disconnect yoyo),but actually are 5549. 2268 lists are not load

thank you dibdot

dibdot, 2017-04-13:
@jack338c This is not a bug, it's a feature!
adblock supports overall sort (make all lists entries unique/remove duplicates) and includes a feature called "top level domain compression" which eliminates subdomains when the top level is already in the list. 
This save valuable space for the dns backends.

Please close this one & happy blocking! :smiley:

jack338c, 2017-04-13:
hi dibdot
actually,same setting for adblock on my another router dir505 is working fine
the difference is my wndr4300's dnsmasq works pdnsd ,ipset etc
So I want to know is there a way to optimize adblock or dnsmasq to solve my problem
thank you dibdot

# here is the logs on my another router dir505 and it's working fine

Thu Apr 13 20:57:37 2017 user.notice adblock-[2.4.0-2] info : start adblock processing ...
Thu Apr 13 20:57:37 2017 user.notice adblock-[2.4.0-2] debug: name: adaway, enabled: 1, backup: 1, dns: dnsmasq, fetch: /usr/bin/wget, memory: 60444
Thu Apr 13 20:57:49 2017 user.notice adblock-[2.4.0-2] debug: name: adaway, mode: backup, count: 379, in_rc: 0, out_rc: 0
Thu Apr 13 20:57:49 2017 user.notice adblock-[2.4.0-2] debug: name: adguard, mode: remove, count: 0, in_rc: 0, out_rc: 0
Thu Apr 13 20:57:49 2017 user.notice adblock-[2.4.0-2] debug: name: blacklist, mode: remove, count: 0, in_rc: 0, out_rc: 0
Thu Apr 13 20:57:49 2017 user.notice adblock-[2.4.0-2] debug: name: disconnect, enabled: 1, backup: 1, dns: dnsmasq, fetch: /usr/bin/wget, memory: 60444
Thu Apr 13 20:57:56 2017 user.notice adblock-[2.4.0-2] debug: name: disconnect, mode: backup, count: 5116, in_rc: 0, out_rc: 0
Thu Apr 13 20:57:56 2017 user.notice adblock-[2.4.0-2] debug: name: dshield, mode: remove, count: 0, in_rc: 0, out_rc: 0
Thu Apr 13 20:57:57 2017 user.notice adblock-[2.4.0-2] debug: name: feodo, mode: remove, count: 0, in_rc: 0, out_rc: 0
Thu Apr 13 20:57:57 2017 user.notice adblock-[2.4.0-2] debug: name: hphosts, mode: remove, count: 0, in_rc: 0, out_rc: 0
Thu Apr 13 20:57:57 2017 user.notice adblock-[2.4.0-2] debug: name: malware, mode: remove, count: 0, in_rc: 0, out_rc: 0
Thu Apr 13 20:57:57 2017 user.notice adblock-[2.4.0-2] debug: name: malwarelist, mode: remove, count: 0, in_rc: 0, out_rc: 0
Thu Apr 13 20:57:57 2017 user.notice adblock-[2.4.0-2] debug: name: openphish, mode: remove, count: 0, in_rc: 0, out_rc: 0
Thu Apr 13 20:57:57 2017 user.notice adblock-[2.4.0-2] debug: name: palevo, mode: remove, count: 0, in_rc: 0, out_rc: 0
Thu Apr 13 20:57:57 2017 user.notice adblock-[2.4.0-2] debug: name: ransomware, mode: remove, count: 0, in_rc: 0, out_rc: 0
Thu Apr 13 20:57:57 2017 user.notice adblock-[2.4.0-2] debug: name: reg_ch, mode: remove, count: 0, in_rc: 0, out_rc: 0
Thu Apr 13 20:57:57 2017 user.notice adblock-[2.4.0-2] debug: name: reg_pl, mode: remove, count: 0, in_rc: 0, out_rc: 0
Thu Apr 13 20:57:57 2017 user.notice adblock-[2.4.0-2] debug: name: reg_ro, mode: remove, count: 0, in_rc: 0, out_rc: 0
Thu Apr 13 20:57:57 2017 user.notice adblock-[2.4.0-2] debug: name: reg_ru, mode: remove, count: 0, in_rc: 0, out_rc: 0
Thu Apr 13 20:57:57 2017 user.notice adblock-[2.4.0-2] debug: name: securemecca, mode: remove, count: 0, in_rc: 0, out_rc: 0
Thu Apr 13 20:57:57 2017 user.notice adblock-[2.4.0-2] debug: name: shalla, mode: remove, count: 0, in_rc: 0, out_rc: 0
Thu Apr 13 20:57:57 2017 user.notice adblock-[2.4.0-2] debug: name: spam404, mode: remove, count: 0, in_rc: 0, out_rc: 0
Thu Apr 13 20:57:57 2017 user.notice adblock-[2.4.0-2] debug: name: sysctl, mode: remove, count: 0, in_rc: 0, out_rc: 0
Thu Apr 13 20:57:57 2017 user.notice adblock-[2.4.0-2] debug: name: whocares, mode: remove, count: 0, in_rc: 0, out_rc: 0
Thu Apr 13 20:57:57 2017 user.notice adblock-[2.4.0-2] debug: name: winspy, mode: remove, count: 0, in_rc: 0, out_rc: 0
Thu Apr 13 20:57:57 2017 user.notice adblock-[2.4.0-2] debug: name: winhelp, mode: remove, count: 0, in_rc: 0, out_rc: 0
Thu Apr 13 20:57:57 2017 user.notice adblock-[2.4.0-2] debug: name: yoyo, enabled: 1, backup: 1, dns: dnsmasq, fetch: /usr/bin/wget, memory: 60444
Thu Apr 13 20:58:02 2017 user.notice adblock-[2.4.0-2] debug: name: yoyo, mode: backup, count: 2319, in_rc: 0, out_rc: 0
Thu Apr 13 20:58:02 2017 user.notice adblock-[2.4.0-2] debug: name: zeus, mode: remove, count: 0, in_rc: 0, out_rc: 0
Thu Apr 13 20:58:02 2017 daemon.info dnsmasq[2421]: exiting on receipt of SIGTERM
Thu Apr 13 20:58:06 2017 user.notice adblock-[2.4.0-2] info : block lists with overall 7814 domains loaded successfully (LEDE Reboot 17.01-SNAPSHOT r3305-c6e7998)

dibdot, 2017-04-13:
@jack338c As I said before, you have no problem - protection level on both routers is the same. Your  second router has only < 64 MByte RAM. Adblock enables overall sort/duplicate removal only on devices with > 64MB RAM, this prevents sort related kernel dumps.

jack338c, 2017-04-13:
ok, I got it
thank you very much for your help

closed 2017-04-13 by jack338c