openwrt/packages #4260: Fatal errors in privoxy v. 3.0.26-2

Tamila-2017, 2017-04-13:
Installed package privoxy v.**3.0.26-2** for **LEDE**

1) This program is unworkable.
When privoxy launched -

`# privoxy /etc/config/privoxy`

it says:

```
root@LEDE:~# privoxy /etc/config/privoxy 
2017-04-13 03:06:37.886 77454c90 Error: Ignoring unrecognized directive 'config	privoxy	'privoxy'' (395678U) in line 16 in configuration file (/etc/config/privoxy).
2017-04-13 03:06:37.887 77454c90 Error: Ignoring unrecognized directive 'option	confdir		/etc/privoxy' (434665U) in line 17 in configuration file (/etc/config/privoxy).
2017-04-13 03:06:37.888 77454c90 Error: Ignoring unrecognized directive 'option	logdir		/var/log' (434665U) in line 18 in configuration file (/etc/config/privoxy).
2017-04-13 03:06:37.889 77454c90 Error: Ignoring unrecognized directive 'option	logfile		privoxy.log' (434665U) in line 19 in configuration file (/etc/config/privoxy).
2017-04-13 03:06:37.891 77454c90 Error: Ignoring unrecognized directive 'list	filterfile	'default.filter'' (16816U) in line 20 in configuration file (/etc/config/privoxy).
2017-04-13 03:06:37.892 77454c90 Error: Ignoring unrecognized directive 'list	actionsfile	'match-all.action'' (16816U) in line 22 in configuration file (/etc/config/privoxy).
2017-04-13 03:06:37.893 77454c90 Error: Ignoring unrecognized directive 'list	actionsfile	'default.action'' (16816U) in line 23 in configuration file (/etc/config/privoxy).
2017-04-13 03:06:37.894 77454c90 Error: Ignoring unrecognized directive 'list	listen_address	'192.168.1.1:8118'' (16816U) in line 26 in configuration file (/etc/config/privoxy).
2017-04-13 03:06:37.895 77454c90 Error: Ignoring unrecognized directive 'option	toggle		'1'' (434665U) in line 27 in configuration file (/etc/config/privoxy).
2017-04-13 03:06:37.897 77454c90 Error: Ignoring unrecognized directive 'option	enable_remote_toggle	'1'' (434665U) in line 28 in configuration file (/etc/config/privoxy).
2017-04-13 03:06:37.898 77454c90 Error: Ignoring unrecognized directive 'option	enable_remote_http_toggle	'0'' (434665U) in line 29 in configuration file (/etc/config/privoxy).
2017-04-13 03:06:37.899 77454c90 Error: Ignoring unrecognized directive 'option	enable_edit_actions	'1'' (434665U) in line 30 in configuration file (/etc/config/privoxy).
2017-04-13 03:06:37.900 77454c90 Error: Ignoring unrecognized directive 'option	enforce_blocks		'0'' (434665U) in line 31 in configuration file (/etc/config/privoxy).
2017-04-13 03:06:37.902 77454c90 Error: Ignoring unrecognized directive 'option	buffer_limit		'4096'' (434665U) in line 32 in configuration file (/etc/config/privoxy).
2017-04-13 03:06:37.903 77454c90 Error: Ignoring unrecognized directive 'option	forwarded_connect_retries	'0'' (434665U) in line 33 in configuration file (/etc/config/privoxy).
2017-04-13 03:06:37.904 77454c90 Error: Ignoring unrecognized directive 'option	accept_intercepted_requests	'0'' (434665U) in line 34 in configuration file (/etc/config/privoxy).
2017-04-13 03:06:37.905 77454c90 Error: Ignoring unrecognized directive 'option	allow_cgi_request_crunching	'0'' (434665U) in line 35 in configuration file (/etc/config/privoxy).
2017-04-13 03:06:37.907 77454c90 Error: Ignoring unrecognized directive 'option	split_large_forms	'0'' (434665U) in line 36 in configuration file (/etc/config/privoxy).
2017-04-13 03:06:37.908 77454c90 Error: Ignoring unrecognized directive 'option	keep_alive_timeout	'300'' (434665U) in line 37 in configuration file (/etc/config/privoxy).
2017-04-13 03:06:37.909 77454c90 Error: Ignoring unrecognized directive 'option	socket_timeout		'300'' (434665U) in line 38 in configuration file (/etc/config/privoxy).
2017-04-13 03:06:37.910 77454c90 Error: Ignoring unrecognized directive 'list	permit_access		'192.168.1.0/24'' (16816U) in line 39 in configuration file (/etc/config/privoxy).
2017-04-13 03:06:37.912 77454c90 Error: Ignoring unrecognized directive 'option	debug_1		'0'' (434665U) in line 40 in configuration file (/etc/config/privoxy).
2017-04-13 03:06:37.913 77454c90 Error: Ignoring unrecognized directive 'option	debug_512	'1'' (434665U) in line 41 in configuration file (/etc/config/privoxy).
2017-04-13 03:06:37.914 77454c90 Error: Ignoring unrecognized directive 'option	debug_1024	'0'' (434665U) in line 42 in configuration file (/etc/config/privoxy).
2017-04-13 03:06:37.915 77454c90 Error: Ignoring unrecognized directive 'option	debug_4096	'1'' (434665U) in line 43 in configuration file (/etc/config/privoxy).
2017-04-13 03:06:37.917 77454c90 Error: Ignoring unrecognized directive 'option	debug_8192	'1'' (434665U) in line 44 in configuration file (/etc/config/privoxy).
root@LEDE:~# 
```

This is a **fatal errors**. Privoxy cannot be used.
Please correct these mistakes as soon as possible!

2) Please, pay special attention to the operability of the **forward_socks5** directive.
This directive also does not work

3) At start by default this program incorrectly searches for a config file -

`Fatal error: can not verify configuration file '/root/config': invalid argument`

In fact, it should look for it in **/etc/config/privoxy** 

Environment:

**Model: GL-MT300A
Firmware: Version LEDE Reboot 17.01.0 r3205-59508e3 / LuCI lede-17.01 branch (git-17.051.53299-a100738)
Kernel: Version 4.4.50
Config file: default**

chris5560, 2017-04-15:
On LEDE and OpenWrt use:

#>/etc/init.d/privoxy enable
to enable autostart on reboot

#>/etc/init.d/privoxy start
to start the service

#>/etc/init.d/privoxy stop
to stop the service

and everything will work fine.

Tamila-2017, 2017-04-15:
Thank you! :)

Nevertheless, Privoxy still does not work.
He does not understand the directive **forward_socks5** and gives such an error:

```
root@LEDE:~# /etc/init.d/privoxy start
2017-04-14 01:06:00.562 7718cc90 Error: Wrong number of parameters for forward-socks5 in configuration file.
```

Full config:

```
config  privoxy 'privoxy'
        option  confdir         '/etc/privoxy'
        option  logdir          '/var/log'
        option  logfile         'privoxy.log'
        list    filterfile      'default.filter'
        list    actionsfile     'match-all.action'
        list    actionsfile     'default.action'
        list    forward_socks5  '/'     '127.0.0.1:1080'
        list    listen_address  '192.168.8.3:8118'
        option  toggle          '1'
        option  enable_remote_toggle    '1'
        option  enable_remote_http_toggle       '0'
        option  enable_edit_actions     '1'
        option  enforce_blocks          '0'
        option  buffer_limit            '4096'
        option  forwarded_connect_retries       '0'
        option  accept_intercepted_requests     '0'
        option  allow_cgi_request_crunching     '0'
        option  split_large_forms       '0'
        option  keep_alive_timeout      '300'
        option  socket_timeout          '300'
        list    permit_access           '192.168.8.0/24'
        option  debug_1         '0'
        option  debug_512       '1'
        option  debug_1024      '0'
        option  debug_4096      '1'
        option  debug_8192      '1'
```

chris5560, 2017-04-19:
your
list forward_socks5 '/' '127.0.0.1:1080'

should look like:
list forward_socks5 'target_pattern socks_proxy[:port] http_parent[:port]'

uci only accept type (list), name (forward_socks5) and only ONE value (/) in your sample
privoxy requires THREE space separated parameters
http://www.privoxy.org/user-manual/config.html#SOCKS
all these parameters are located between ONE pair of single quotes inside UCI configuration

dibdot, 2018-08-17:
no further feedback - close this by now.

closed 2018-08-17 by dibdot