openwrt/packages #4261: adblock: 2.6.0 procd implementation does not boot start and may crash DNS backend

EricLuehrsen, 2017-04-14:
Maintainer: @dibdot 
Environment: LEDE Trunk
Description:
Something in 2.6.0 has regressed. Adblock does not appear to start up at boot (completely?). No files are created for the DNS back end. The event triggers still occur, may restart DNS back end in short periods, and cause back end to crash. Adblock does start completely with manual trigger. Changing the restart delay to 5 seconds solved the crashing. I can't seem to pin it down.

dibdot, 2017-04-14:
@EricLuehrsen adblock 2.6.0 solely relies on "real" interface trigger on boot (no raw trigger or hard timeouts any more). If adblock doesn't start, please make an appropriate "adb_iface" entry (ubus list). By default 'wan' interface is set in config.

Where did you change a restart delay? You mean the raw trigger timeout in unbound init?

EricLuehrsen, 2017-04-14:
I was using default as installed UCI. I haven't done anything special so my WAN is named 'wan'. 

> Adblock itself does not produce any interface trigger events.
When adblock is executed by its own trigger events it eventually calls the respective `init.d/... restart` which forces a hard restart.

> Where did you change a restart delay? Did you mean the raw trigger timeout in unbound init?
You made a UCI / LuCI configuration parameter to set a procd script variable before setting the adblock triggers. I changed `adb_triggerdelay` to 5.
```
local delay="$(uci -q get adblock.global.adb_triggerdelay)"
PROCD_RELOAD_DELAY=$((${delay:=1} * 1000))
procd_add_interface_trigger "interface.*.up" "${name}" "${adb_init}" start
```

dibdot, 2017-04-14:
I can't see any adblock problem here. Just re-tested with a default unbound installation on a TP-LINK RE450:

1. Boot without adblock:
root@adb2go:~# logread | grep "start of"
```
Fri Apr 14 17:25:46 2017 daemon.info unbound: [2279:0] info: start of service (unbound 1.6.1).
Fri Apr 14 17:25:53 2017 daemon.info unbound: [2379:0] info: start of service (unbound 1.6.1).

```
I've got 2 unbound reloads during normal startup within 7 seconds!?

Now Boot with enabled adblock:
```
Fri Apr 14 17:28:29 2017 daemon.info unbound: [2330:0] info: start of service (unbound 1.6.1).
Fri Apr 14 17:29:34 2017 daemon.info unbound: [2451:0] info: start of service (unbound 1.6.1).
Fri Apr 14 17:29:42 2017 daemon.info unbound: [2665:0] info: start of service (unbound 1.6.1).
```
Two unbound reloads plus the third one (restart) triggered by adblock and all looks OK.

EricLuehrsen, 2017-04-14:
>I can't see any adblock problem here.

This baffles me also. It _should_ start as you have written it but it just doesn't.

>I've got two unbound reloads during normal startup within 7 seconds!?

DNSSEC needs time, time needs NTP, and NTP needs DNS. Chicken-and-egg. The BOOT instance of Unbound enables DNSSEC but without time validation. NTP has a hotplug event. Once NTP is satisfied, then Unbound is entirely restarted with DNSSEC at full strength. Same is done for dnsmasq-full package.

EricLuehrsen, 2017-04-14:
>I can't see any adblock problem here.

This baffles me also. It _should_ start as you have written it but it just doesn't. **UPDATE:** This is strange. I now tried this two ways. If I build LEDE from source with adblock included, then it will not auto start. If I install adblock afterwards on the running router, then it works perfectly.

>I've got two unbound reloads during normal startup within 7 seconds!?

DNSSEC needs time, time needs NTP, and NTP needs DNS. Chicken-and-egg. The BOOT instance of Unbound enables DNSSEC but without time validation. NTP has a hotplug event. Once NTP is satisfied, then Unbound is entirely restarted with DNSSEC at full strength. Same is done for dnsmasq-full package.

dibdot, 2017-04-16:
I've changed the default trigger timeout with #4267

EricLuehrsen, 2017-04-17:
- Thanks for updating the delay. It would seem the practical requirement is 2 to 5 seconds. My experience shows race conditions near 1 second. 
- For the frequency of restarts, it appears some coincident change in Trunk is creating frequent ifup, and it doesn't occur in 17.01. [lede/netifd](https://git.lede-project.org/?p=project/netifd.git;a=commit;h=a03216660797173fbe67866f75564e3fec9c1e8d) and [lede/FS#713](https://bugs.lede-project.org/index.php?do=details&task_id=713)
- I still haven't worked out why adblock auto-starts fine when separate install but not when built-in.

dibdot, 2017-04-17:
Thanks for your feedback! Did you change the start priority to 50, too? I've added this to the above mentioned PR because of restart issues on an APU2 (with LAN trigger).
Thanks again.

EricLuehrsen, 2017-04-17:
The start priority does fix it. I am not sure about the ubus command side effects. When you use only careful procd triggers with boot-start trick, then you can set priority to 19 (before netifd 20).

closed 2017-04-17 by EricLuehrsen