openwrt/packages #4294: zsh : please build with --enable-regex and --enable-pcre

neskweek, 2017-04-24:
### Package: zsh
Hello

- expected behavior: be able to use some Oh My Zsh plugins and themes like powerlevel9k

- actual behavior: 
when starting zsh with powerlevel9k I get this error :  

> /root/.oh-my-zsh/custom/themes/powerlevel9k/powerlevel9k.zsh-theme:29: failed to load module: zsh/regex
> /root/.oh-my-zsh/custom/themes/powerlevel9k/powerlevel9k.zsh-theme:29: -regex-match not available for regex

The problem is that the operator **=~** is not supported with the actual build
reference : 
https://github.com/robbyrussell/oh-my-zsh/issues/227

Simple test : 
>\# [[ "$remote" =~ "github" ]]
zsh: failed to load module: zsh/regex
zsh: -regex-match not available for regex

Thank you
Best regards

diizzyy, 2017-06-07:
@msva 
Is this reasonable or perhaps possible to add as an option?

msva, 2017-06-08:
@diizzyy well, it is reasonable, just as some other options.

I've added many things to zsh's Makefile locally, and now performing some testing on router. I'll make PR in some near time.

// btw, I started writing of that answer two days ago, and got distracted to work, and just returned to it

dibdot, 2018-08-17:
@msva can this be closed?

neheb, 2019-03-30:
regex was removed or renamed while pcre is enabled. closing.

closed 2019-03-30 by neheb

nobk, 2019-12-10:
regex and pcre are different module, only regex module provide =~ operator in zsh.
zsh-syntax-highlighting plugins need =~

```
~/git/zsh-syntax-highlighting
grep "=~" * -R 
docs/highlighters/regexp.md:See also: [regular expressions tutorial][perlretut], zsh regexp operator `=~`
highlighters/main/main-highlighter.zsh:         [[ ${parameter_name} =~ ^${~parameter_name_pattern}$ ]] &&
highlighters/main/main-highlighter.zsh:        while [[ $arg[i+1] == [\^=~#+] ]]; do
highlighters/main/main-highlighter.zsh:        if $highlight_glob && [[ ${arg[$i]} =~ ^[*?] || ${arg:$i-1} =~ ^\<[0-9]*-[0-9]*\> ]]; then
highlighters/main/main-highlighter.zsh:            if [[ ${arg:$i} =~ ^([A-Za-z_][A-Za-z0-9_]*|[0-9]+) ]] ; then
highlighters/main/main-highlighter.zsh:            elif [[ ${arg:$i} =~ ^[{]([A-Za-z_][A-Za-z0-9_]*|[0-9]+)[}] ]] ; then
highlighters/main/main-highlighter.zsh:            if [[    "$AA" =~ "^(x|X)[0-9a-fA-F]{1,2}"
highlighters/main/main-highlighter.zsh:                  || "$AA" =~ "^[0-7]{1,3}"
highlighters/main/main-highlighter.zsh:                  || "$AA" =~ "^u[0-9a-fA-F]{1,4}"
highlighters/main/main-highlighter.zsh:                  || "$AA" =~ "^U[0-9a-fA-F]{1,8}"
highlighters/regexp/regexp-highlighter.zsh:    [[ "$buf" =~ "$pat" ]] || return;

```

neheb, 2019-12-13:
regex is unconditionally enabled now. You sure you're not using an old version?

nobk, 2019-12-13:
Finally I find why zsh-syntax-highlighting do not work. 
--disable-dynamic in Makefile cause it can not load.
regex module can be not present in static build of Zsh.
one command line to test if regex work or not.
`[[ "asdf" =~ "/*s*" ]] && echo "regex ok"`