openwrt/packages #4299: gpsd fails to comipile with build error related to missing curses library

mrnuke-adaptrum, 2017-04-25:
-------------------------------

Maintainer: @psidhu @p-wassi 
Environment: OMAP, BeagleBone, LEDE-17.01

Description:

```
Formating code blocks by wrapping them with pairs of ```
```

mrnuke-adaptrum, 2017-04-25:
The same issue happens when selecting glibc as the the C library. Error in [this gist](https://gist.github.com/mrnuke-adaptrum/4416c82b693aae843a31f5ccf3e9650b)

thornley-touchstar, 2017-05-02:
This also fixed cross-compilation on macOS for OMAP, but inadvertently by the look of it. I patched it a different way, but my problem was with SConstruct in that it checks sys.platform() against darwin and if detected assumes ncurses will be "libcurses" as opposed to "libncurses" even though it's cross compiling for OMAP.  

I was getting this linking error: -

> lede-source-ts/staging_dir/toolchain-mips_24kc_gcc-5.4.0_musl-1.1.16/lib/gcc/mips-openwrt-linux-musl/5.4.0/../../../../mips-openwrt-linux-musl/bin/ld: cannot find -lcurses
> 

The change you made is forcing "ncurseslibs" variable via the configuration script prior to the platform check, which on my macOS environment is currently using brew which results in "libncursesw", so I get this: -

> /usr/local/Cellar/ncurses/6.0/bin/ncurses6-config --libs --flags
> -L/usr/local/Cellar/ncurses/6.0/lib -lncursesw
> -D_DARWIN_C_SOURCE -I/usr/local/Cellar/ncurses/6.0/include/ncursesw -I/usr/local/Cellar/ncurses/6.0/include

FYI, my version of macOS (10.12.4) looks to include ncurses 5.4: -

> /usr/bin/ncurses5.4-config --libs --flags
> -lncurses
> -I/usr/include/ncurses
> 

So the original assumption in SConstruct that the macOS is "libcurses" might be wrong or dated... I don't know enough to comment further, but your change fixed my issue! 👍 

Just letting you know :)

thornley-touchstar, 2017-05-02:
One question I did have was regarding the mix-matching of ncursesw (unicode) and ncurses, do you think this would be a problem?

dibdot, 2018-08-18:
Reported issue seems to be fixed.

PR #4300 (2017-05-08): utils/gpsd: Backport ncurses6 support from master
Because gpsd FTBTS without this patch.

Maintainer: @psidhu @p-wassi
Compile tested: OMAP+glibc, BeagleBone, LEDE-17.01
Run tested: N/A

Description:
Fixes #4299
closed 2018-08-18 by dibdot