openwrt/packages #4307: kmod-usb-serial-ch341: garbage outpout on serial printer

outlawx, 2017-04-27:
I have some old receipt printers (Epson TM-T88 III and TM-U220), both with serial port and default settings 9600 8-N-1. Doing just fine.
Also I have a budget CH341 USB to serial adapter and working corect on my daily Linux workstation (kernel 4.10) & a Windows-box.

With the CH341 connected to either a Netgear WNDR3700v1 or a TP-LINK TL-WR710Nv1 the printers only produce garbage when echoing text to /dev/ttyUSB0. For most sent lines the printers do nothing and the rest is only trash. Settings are correct, checked with stty.

Trying a PL2303 usb-to serial evertything works just fine on these boxes.
Firmware tested on both devices OpenWrt 15.05.1 and LEDE 17.01.1

karlp, 2017-04-27:
4.10 is just _barely_ new enough to have a working ch341.  this isn't a problem in lede/openwrt, it's a kernel problem with those devices. (and iirc, not even all teh fixes made it to 4.10)

closed 2017-04-27 by karlp

outlawx, 2017-04-27:
Hi karlp, thanks for your reply :)

Wasn't sure whether the sources for a working ch341-module are compiled against the kernel or included in the kernel.

Hadn't known about this cheap ch341 usb-to-serial til I first stumbled over an Olympia cash register last year - scary hardware designed to be hacked vs zero tolerance austrian & german tax law...