openwrt/packages #4310: Uhttpd: sh: broken pipe spaming

LipkeGu, 2017-04-28:
A long time we get when using Luci... something is badly with uhttpd and sh
```
...
Fri Apr 28 17:16:31 2017 daemon.err uhttpd[1078]: sh: write error: Broken pipe
Fri Apr 28 17:16:31 2017 daemon.err uhttpd[1078]: sh: write error: Broken pipe
Fri Apr 28 17:16:31 2017 daemon.err uhttpd[1078]: sh: write error: Broken pipe
Fri Apr 28 17:16:31 2017 daemon.err uhttpd[1078]: sh: write error: Broken pipe
Fri Apr 28 17:16:31 2017 daemon.err uhttpd[1078]: sh: write error: Broken pipe
Fri Apr 28 17:16:31 2017 daemon.err uhttpd[1078]: sh: write error: Broken pipe
Fri Apr 28 17:16:31 2017 daemon.err uhttpd[1078]: sh: write error: Broken pipe
Fri Apr 28 17:16:31 2017 daemon.err uhttpd[1078]: sh: write error: Broken pipe
Fri Apr 28 17:16:31 2017 daemon.err uhttpd[1078]: sh: write error: Broken pipe
Fri Apr 28 17:16:31 2017 daemon.err uhttpd[1078]: sh: write error: Broken pipe
Fri Apr 28 17:16:31 2017 daemon.err uhttpd[1078]: sh: write error: Broken pipe
Fri Apr 28 17:16:31 2017 daemon.err uhttpd[1078]: sh: write error: Broken pipe
Fri Apr 28 17:16:32 2017 daemon.err uhttpd[1078]: sh: write error: Broken pipe
Fri Apr 28 17:16:32 2017 daemon.err uhttpd[1078]: sh: write error: Broken pipe
Fri Apr 28 17:16:32 2017 daemon.err uhttpd[1078]: sh: write error: Broken pipe
...
```

hnyman, 2017-04-28:
That is a generic error that is typically coming from a package's init script or hotplug script, where an uninitialised (or empty) variable is used. Or alternatively some command run by the page fails. Typically e.g. a config option has no default value and if the option is missing from the config file, the variable remains empty. When that shell variable is then used in a comparison like "is it greater than", the shell complains about empty value, and uhttpd passes the error. Same goes for failing output from a command run by LuCI to create info for some config page.

Such errors have been seen in the last few years e.g. in ddns-scripts, mwan3 etc. packages. If the package has a status component on the LuCI overview page and that is refreshed every 5 seconds, the log spam gets visible.

See e.g.
https://github.com/openwrt/packages/issues/4246
https://github.com/openwrt/packages/pull/2798#issuecomment-223598889

So, the error itself has nothing to do with uhttpd. 
As others are not complaining about that error in large scale, it is likely something specific to your config or values. Is that error coming from **every** page? or just from the front page (that has automatically refreshing components)? Or some other specific page(s)?

If it is coming from every page, then you might check any special characters in router name etc. generic errors. Otherwise, try to narrow 

I am closing this issue because neither uhttpd or LuCI is located in this packages repo.

but if you figure out any specific item causing that for you (some special package combination, special settings?), please add the info here.

closed 2017-04-28 by hnyman

LipkeGu, 2018-02-15:
OK i did some investigation to this broken pipe thing...

- login to luci and locate to the Main page (status overview)let it open!!!
- now login via ssh and type 'logread -f' and in luci navigate to 'status -> firewall'
- logread should now spaming 'brocken pipe'

hnyman, 2018-02-15:
>   login to luci and locate to the Main page (status overview)let it open!!!
>   now login via ssh and type 'logread -f' and in luci navigate to 'status -> firewall'
>   logread should now spaming 'brocken pipe'

That does not cause any log messages for me.

Like I said earlier:
> As others are not complaining about that error in large scale, it is likely something specific to your config or values.

The general reason is that for some reason your config leads a shell script used by LuCI to stumble into an invalid value.

LipkeGu, 2018-02-15:
you should repeat it (it happens at the time where the page is showing up and while the entries

hnyman, 2018-02-15:
No, that does not generate errors for me.

Have you removed ipv6, or anything like that, which would make your build or config non-standard? 

(So that when LuCI tries to evaluate the firewall config, the ipv6 side is missing, or some other firewall table is missng.)

LipkeGu, 2018-02-15:
no i have ipv6 included in my build.

hnyman, 2018-02-15:
Ok, but it is likely still something similar, where your specific build / config leads the shell script used by LuCI for that firewall status page to read an empty value, as something that is usually there is not present in your system.

LipkeGu, 2018-02-15:
what is happening when you do **repeatly** the following:
1. go to overview (main)
2. switch to firewall status

hnyman, 2018-02-15:
Nothing special. No new messages to system log.

Tried a dozen times...

LipkeGu, 2018-02-15:
do you have firefox?

hnyman, 2018-02-15:
> do you have firefox?

Yes. 59.0b9

LipkeGu, 2018-02-15:
when im switching to overview and switching back to firewall status (before the load of the overview page is done) i get this.

hnyman, 2018-02-15:
>  it seems uhttpd tries to write into a stream which is already closed.

No, 
This is about a shell script or command running in your router, triggered by that LuCI page load. Parsing iptables values, or something similar. The script does not get an expected value and that causes the error at the next step, when the empty value is passed as a parameter to the next command in the script. 

uhttpd is the innocent party here, like I already explained in my first message in April 2017.
It just passes along the error message from the shell where the command runs.

LipkeGu, 2018-02-15:
it is not the firewall status it is guarantee related to the overview page (when it tries to request swap, local time) and you switching to another page.