openwrt/packages #4314: ipt-netflow: Feature request

nL1ght5, 2017-04-30:
Request to add netflow capabilities for LEDE/OpenWrt/... 

The ipt-netflow package (https://github.com/aabc/ipt-netflow) can apparently already  be built for LEDE/OpenWrt, could it be considered to be added as a standard package?

hnyman, 2017-04-30:
Feel free contribute: 
* craft the needed package Makefile, compile and test it, 
* volunteer to be the maintainer for the package, and 
* submit a Pull Request for importing the package here.

If others show interest for the package, it might get accepted to the repo.)

nL1ght5, 2017-04-30:
Update, apparently someone did the work and the needed Makefile has already been crafted.
https://github.com/alex-eri/ipt-netflow/blob/openwrt-makefile/openwrt/Readme.md

dibdot, 2018-08-18:
feel free to open a PR.

closed 2018-08-18 by dibdot