openwrt/packages #4318: Netatalk 3.1.10-1 : Wrong file permissions on afp.conf

Mrengles, 2017-05-01:
Maintainer: @dangowrt
Environment: mvebu, arm_cortex-a9_vfpv3 , WRT1900ACS, LEDE Reboot 17.01.1 r3316-7eb58cf109

The netatalk 3.1.10-1 package applies the wrong file permission to `/etc/afp.conf` and `/etc/extmap.conf` which make them unreadable by daemon. These files should have a POSIX file permission of 644 `-rw-r--r--  1 root  root` but currently use 600 `-rw-------  1 root  root`.

The files located. in `/var/netatalk/` have the proper permissions and function as intended.

Current users can change permissions with `chmod` to workaround this problem.
```
chmod 644 /etc/afp.conf
chmod 644 /etc/extmap.conf
```

commodo, 2017-05-29:
@robberize 
are you sure this is an issue ?
usually LEDE/OpenWrt processes are run as root
this one also runs as root 
```
root@LEDE:/# ps w | grep afp
 1856 root     20716 S    /usr/sbin/afpd -d -F /etc/afp.conf
 1885 root      3072 R    grep afp
```

this can be adapted to run as a non-root user;
but i guess that's up to the maintainer

robberize, 2017-05-31:
@commodo 

I can confirm this issue has effected myself and at least one other. The issue was discussed over at the LEDE-Project forum. [Help with Apple filesharing protocol for Time Machine](https://forum.lede-project.org/t/help-with-apple-filesharing-protocol-for-time-machine/3259/6)

Let me clarify by saying; the process running as root is not the issue. please disregard my statement above about deamon reading the configuration files.

For whatever reason Netatalk `/usr/sbin/afpd` can not read `/etc/afp.conf` and `/etc/extmap.conf` with the permissions at 600 `-rw------- 1 root root`. The root "group" and "world" also need read privileges for everything to work as intended.

closed 2017-06-05 by dangowrt

Neustradamus, 2024-11-25:
@ffainelli: For the history:

Netatalk 3.1.7 (not merged):
- https://github.com/openwrt/packages/pull/1550

Netatalk 3.1.9:
- https://github.com/openwrt/packages/pull/3135

Netatalk 3.1.11:
- https://github.com/openwrt/packages/pull/4402

Netatalk 3.1.12:
- https://github.com/openwrt/packages/pull/7850

netatalk: drop package:
- https://github.com/openwrt/packages/pull/15873

Netatalk 3.1.13:
- https://github.com/openwrt/packages/pull/18687
- Backport: https://github.com/openwrt/packages/pull/18696

Netatalk 3.1.18 (not merged):
- https://github.com/openwrt/packages/pull/24206
- https://github.com/openwrt/packages/pull/24249

Netatalk 3.2.0:
- https://github.com/openwrt/packages/pull/24257

Netatalk 3.2.5:
- https://github.com/openwrt/packages/pull/24762

Netatalk 4.0.2 (not merged):
- https://github.com/openwrt/packages/pull/25195

Netatalk 4.0.4:
- Master: https://github.com/openwrt/packages/pull/25200 + https://github.com/openwrt/packages/pull/25251
- 24.10 backport: https://github.com/openwrt/packages/pull/25246 + https://github.com/openwrt/packages/pull/25260