openwrt/packages #4319: Netatalk 3.1.11 : Upgrade Available

Mrengles, 2017-05-01:
Maintainer: @dangowrt
Environment: LEDE Reboot 17.01.1 r3316-7eb58cf109

## **Netatalk 3.1.11 is available!**

**Changes:**

- NEW: Global option "zeroconf name", FR#99
- NEW: show Zeroconf support by "netatalk -V", FR#100
- UPD: gentoo: Switch openrc init script to openrc-run, GitHub#77
- FIX: log message: name of function doese not match, GitHub#78
- UPD: volume capacity reporting to match Samba behavior, GitHub#83
- FIX: debian: sysv init status command exits with proper exit code, GitHub#84
- FIX: dsi_stream_read: len:0, unexpected EOF, GitHub#82
- UPD: dhx uams: OpenSSL 1.1 support, GitHub#87

**Availability:**

Netatalk tar-balls can be found at:

http://sourceforge.net/project/showfiles.php?group_id=8642

https://github.com/Netatalk

Mrengles, 2017-05-01:
Should have linked my previous #4318 issues. Kill two birds with one stone.

diizzyy, 2017-06-02:
@robberize 
Update notifications are being sent via uscan runs by @sdwalker, no need to create issues for update requests. Please close this.

robberize, 2017-06-02:
@diizzyy 
Closed per request.

closed 2017-06-02 by robberize

Neustradamus, 2024-11-25:
@ffainelli: For the history:

Netatalk 3.1.7 (not merged):
- https://github.com/openwrt/packages/pull/1550

Netatalk 3.1.9:
- https://github.com/openwrt/packages/pull/3135

Netatalk 3.1.11:
- https://github.com/openwrt/packages/pull/4402

Netatalk 3.1.12:
- https://github.com/openwrt/packages/pull/7850

netatalk: drop package:
- https://github.com/openwrt/packages/pull/15873

Netatalk 3.1.13:
- https://github.com/openwrt/packages/pull/18687
- Backport: https://github.com/openwrt/packages/pull/18696

Netatalk 3.1.18 (not merged):
- https://github.com/openwrt/packages/pull/24206
- https://github.com/openwrt/packages/pull/24249

Netatalk 3.2.0:
- https://github.com/openwrt/packages/pull/24257

Netatalk 3.2.5:
- https://github.com/openwrt/packages/pull/24762

Netatalk 4.0.2 (not merged):
- https://github.com/openwrt/packages/pull/25195

Netatalk 4.0.4:
- Master: https://github.com/openwrt/packages/pull/25200 + https://github.com/openwrt/packages/pull/25251
- 24.10 backport: https://github.com/openwrt/packages/pull/25246 + https://github.com/openwrt/packages/pull/25260