openwrt/packages #4325: mwan3: Do not exec because ip path

Flawioo, 2017-05-04:
Maintainer: @Jeroen Louwes <jeroen.louwes@gmail.com>, Florian Eckert <fe@dev.tdt.de>
Environment: (mips_24kc, TP-Link TL-WR1043ND V1, LEDE 17.01.1)

Description:
I installed mwan3 using "opkg install mwan3" and it does not show any error but the command mwan3 does not show any response. 
Checking script with command "bash -x /usr/sbin/mwan3 status" we can see a fail.
```
root@LEDE:/# bash -x /usr/sbin/mwan3 status
+ '[' -x /usr/bin/ip ']'
+ exit 4
root@LEDE:/#
```
It occur because ip command is not located at /usr/bin/ip but /sbin/ip.

To solve this problem i created a link from /usr/bin/ip apointing to /sbin/ip with command:
```
root@LEDE:/# ln -s /sbin/ip /usr/bin/ip
root@LEDE:/# which ip
/usr/bin/ip
root@f8-d1-11-be-2f-f0:/# ls -lrt /usr/bin/ip
lrwxrwxrwx    1 root     root             8 May  4 12:58 /usr/bin/ip -> /sbin/ip
root@f8-d1-11-be-2f-f0:/#
```

hnyman, 2017-05-04:
@feckert

feckert, 2017-05-05:
I have not noticed that the path of ip was changed again.
openwrt-15.05 -> /usr/sbin/ip
lede-17.01 -> /usr/bin/ip
lede master -> /sbin/ip

I think i will change the script to not look at hard coded location like in `mwan3.sh`. The scripts should in instead use the PATH env to get the right binary.

```

#!/bin/sh
IP4="/usr/sbin/ip -4"
IP6="/usr/sbin/ip -6"
IPS="/usr/sbin/ipset"
IPT4="/usr/sbin/iptables -t mangle -w"
IPT6="/usr/sbin/ip6tables -t mangle -w"
LOG="/usr/bin/logger -t mwan3 -p"

```

to 

```

#!/bin/sh
IP4="ip -4"
IP6="ip -6"
IPS="ipset"
IPT4="iptables -t mangle -w"
IPT6="ip6tables -t mangle -w"
LOG="logger -t mwan3 -p"

```

@hnyman Is this the right way to go? Does this will work? The needed binaries the scripts are using are set in the Makefile and get installed in the right place during image generation.

hnyman, 2017-05-05:
Sounds right.
The absolute path will depend on the ip installed. Busybox ip has different path than the one from ip-full

closed 2017-05-12 by champtar