openwrt/packages #4328: unbound: without dnsmasq UCI dhcp.domain stances are ignored

poranje, 2017-05-05:
Maintainer: @EricLuehrsen 
Environment: LEDE all branches
Description: without dnsmasq UCI dhcp.domain stances are ignored

The UCI for dnsmasq allows domain names to be added to DNS (A record) with UCI dhcp.domain stances; unbound does not provide this functionality. Having this possibility to add domain names in DNS for hosts that do not receive their IP config by the local DHCP service (static IP config) or to overrule the address record for a foreign host serves many scenario's.

This UCI is documented in [doc/uci/dhcp](https://wiki.openwrt.org/doc/uci/dhcp#custom_domain), where also UCI for other DNS RRs such as CNAME, MX and SRV are documented).
e.g.
```
config domain
        option name 'myhost.somenet'
        option ip '10.12.14.16'
```

With unbound and odhcpd now being able to supplant dnsmasq, having unbound handle the UCI for the mentioned DNS RR would help migrating to a system without dnsmasq (that mostly works well but which handles things like DNSSEC not as good as unbound does).

Obiter:
The UCI toplevel dhcp is kind of misnomer for DNS related UCI. The DNS related UCI should be better placed in a config named dns (thus /etc/config/dns), but that would currently entail more than a sane person would want to.

EricLuehrsen, 2017-05-07:
This should not be too difficult in terms of scripting. The host record DNS entry for DHCP exceptions is basic. 

However, the SRV, MX, and CNAME entries may not be as feasible. These are normally authoritative responses. BIND is an all-in-one and dnsmasq pretends to be. Unbound on the other hand is a recursive server only. [NLNetLabs](https://nlnetlabs.nl/) chose to split the functions and NSD is the authoritative server. There may be a necessary break in Unbound vs dnsmasq transparency.

The CNAME is the most concern. Unbound doesn't solve the CNAME in "local-data:" entry. It will return the record "IN CNAME" but it will not perform the lookup alias. UCI scripts could execute the CNAME and enter more "local-data:" to simulate it. The problem is two parts: (1) if the CNAME is to a server you don't control, then it can move and the scripts won't run. (2) if the CNAME is to a load-balanced DNS record or anycast server group, then similar problems can occur. Users probably won't expect these behaviors and won't understand why the CNAME record doesn't work reliably. [Unbound conf](https://www.unbound.net/documentation/unbound.conf.html).

poranje, 2017-05-07:
Informative and helpful as ever ! Thanks.

The probably most relevant functionality would be adding A, AAAA and PTR in the local zone based on UCI dhcp.domain stances which will allow serving those records for hosts that are not configured with DHCP (with either odhcpd or dnsmasq).
So for those the same kind of entries as for DHCP derived records (with indefinite lifetime ?) could be used.

The other type of RRs are most likely of lesser importance.
Although unbound can act as authoritative server for **local data** (see [0]), lets assume that for local zones having a separate “authoritative” DNS server like dnsmasq is preferred. In that case the UCI documentation may best be extended with a note that these RR require dnsmasq to handle the local zone.

For my understanding, is the following setup feasible ?
a) odhcpd serving DHCP for both IPv4 and IPv6 and RA
b) unbound handles DNS (A, AAAA and PTR) for DHCP leases provided by odhcpd
c) dnsmasq act as the authoritative DNS server for (static) local RRs.
Which forwards/stubs/views in the unbound configuration will accomplish this ?
(forwards point to other recursive server, stubs points to authoritative servers and views are suitable for ..., I have no experience with those).

Just using a host file would probably satisfy most use-cases and trying to add UCI support for adding records to the local is maybe not worth the effort. On the other hand ...

[0] sections on local-zone and local-date in https://www.unbound.net/documentation/unbound.conf.html 

Bye,
-- 
Paul Oranje

EricLuehrsen, 2017-05-09:
Trying to work out the best UCI / LuCI structure so users can control side effects. If dnsmasq is present, then we could assume on first guess dnsmasq handles all of UCI.dhcp, and Unbound surplants your ISP upstream server only. (the reason dhcp and dns are in uci.dhcp tigether is dnsmasq). When Unbound is the only DNS server to be found, then we need to take over these stances. The A~ and PTR are easy. The authoratative RR have side effects. These should be managed with Unbound options ... and then consider use cases in the middle.

EricLuehrsen, 2017-05-10:
Trying a few things including the "www.facebook.com IN A 1.2.3.4" example from openwrt.org wiki. Unbound creates automatic local-domain: where one doesn't exist and depending on the number this may be inefficient. This isn't a concern for "mycomputer.lan" where ".lan" already exists. But otherwise, we probably want a two-name-tier "local-domain: abcd.tld transparent" clause to lead it. We also don't want these local-domain: duplicated. This isn't terribly complex, but it is tedious.

EricLuehrsen, 2017-05-13:
Not ready for PR, but you can review so far...
https://github.com/EricLuehrsen/openwrt-packages/tree/unbound_may2017

EricLuehrsen, 2017-05-15:
@poranje please close this issue and continue discussion and review with PR #4361

closed 2017-05-15 by poranje