openwrt/packages #4333: postfix: fails to start before netifd enables network interfaces

ardrabczyk, 2017-05-06:
Maintainer: @Shulyaka 
Environment: mips, TP-LINK TL-WDR3600, OpenWrt Chaos Calmer 15.05

Description:

On my system `netifd` enables network interfaces *after* `postfix` starting script is run, making it fail:

```Sat May  6 21:04:32 2017 mail.crit postfix[1362]: fatal: could not find any active network interfaces
Sat May  6 21:04:33 2017 kern.info kernel: [   26.920000] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
Sat May  6 21:04:33 2017 kern.info kernel: [   26.940000] device eth0.1 entered promiscuous mode
Sat May  6 21:04:33 2017 kern.info kernel: [   26.940000] device eth0 entered promiscuous mode
Sat May  6 21:04:33 2017 daemon.notice netifd: Interface 'lan' is enabled
Sat May  6 21:04:33 2017 daemon.notice netifd: Interface 'lan' is setting up now
Sat May  6 21:04:33 2017 kern.info kernel: [   26.960000] IPv6: ADDRCONF(NETDEV_UP): br-lan: link is not ready
Sat May  6 21:04:33 2017 daemon.notice netifd: Interface 'lan' is now up
Sat May  6 21:04:33 2017 daemon.notice netifd: Interface 'loopback' is enabled
Sat May  6 21:04:33 2017 daemon.notice netifd: Interface 'loopback' is setting up now
Sat May  6 21:04:33 2017 daemon.notice netifd: Interface 'loopback' is now up
Sat May  6 21:04:33 2017 daemon.notice netifd: Interface 'wan' is enabled
Sat May  6 21:04:33 2017 daemon.notice netifd: Interface 'wan6' is enabled
Sat May  6 21:04:33 2017 kern.info kernel: [   27.000000] IPv6: ADDRCONF(NETDEV_UP): eth0.2: link is not ready
Sat May  6 21:04:33 2017 daemon.notice netifd: Network device 'lo' link is up
Sat May  6 21:04:33 2017 daemon.notice netifd: Interface 'loopback' has link connectivity
Sat May  6 21:04:34 2017 kern.info kernel: [   27.610000] eth0: link up (1000Mbps/Full duplex)
Sat May  6 21:04:34 2017 kern.info kernel: [   27.610000] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
Sat May  6 21:04:34 2017 daemon.notice netifd: Network device 'eth0' link is up
Sat May  6 21:04:34 2017 daemon.notice netifd: VLAN 'eth0.2' link is up
Sat May  6 21:04:34 2017 daemon.notice netifd: Interface 'wan' has link connectivity
Sat May  6 21:04:34 2017 daemon.notice netifd: Interface 'wan' is setting up now
Sat May  6 21:04:34 2017 daemon.notice netifd: Interface 'wan6' has link connectivity
Sat May  6 21:04:34 2017 daemon.notice netifd: Interface 'wan6' is setting up now
Sat May  6 21:04:34 2017 daemon.notice netifd: VLAN 'eth0.1' link is up
Sat May  6 21:04:34 2017 kern.info kernel: [   27.670000] br-lan: port 1(eth0.1) entered forwarding state
Sat May  6 21:04:34 2017 kern.info kernel: [   27.670000] br-lan: port 1(eth0.1) entered forwarding state
Sat May  6 21:04:34 2017 kern.info kernel: [   27.680000] IPv6: ADDRCONF(NETDEV_CHANGE): eth0.2: link becomes ready
Sat May  6 21:04:34 2017 kern.info kernel: [   27.680000] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
Sat May  6 21:04:34 2017 daemon.notice netifd: Bridge 'br-lan' link is up
Sat May  6 21:04:34 2017 daemon.notice netifd: Interface 'lan' has link connectivity
```

Adding `ifconfig lo up` or `sleep` before `postfix -D start` helps but
it's more a workaround than a real solution. A similar problem with
the program running before interfaces are set up has been reported
here https://dev.openwrt.org/ticket/14199.

I also set `debugger_command` in `/etc/postfix/main.cf` but from what
I see it's not working when `postfix` fails to start. Other users
https://mail-index.netbsd.org/netbsd-bugs/2008/09/24/msg006151.html
have reported that `SIOCGIFNETMASK: Device not configured` error
occurs here.

Just to make it clear - I didn't modify `netifd` priority - it's still
has `START=20`.  This issue might not be reproducible on a different
hardware.

lucize, 2017-05-12:
is something that blocks you from using LEDE 17x release ? have you tried it?

ardrabczyk, 2017-05-21:
Hello,

Nothing blocks me from using Lede and honestly I'd prefer to use it
instead of OpenWRT because it's more actively developed but last time
I looked into it it wasn't available in a stable version. It's good to
know it is now.

Unfortunately, the same happens:

Sun May 21 15:31:21 2017 mail.crit postfix[960]: fatal: could not find any active network interfaces

root@LEDE:~# cat /etc/openwrt_release
DISTRIB_ID='LEDE'
DISTRIB_RELEASE='17.01.1'
DISTRIB_REVISION='r3316-7eb58cf109'
DISTRIB_CODENAME='reboot'
DISTRIB_TARGET='ar71xx/generic'
DISTRIB_ARCH='mips_24kc'
DISTRIB_DESCRIPTION='LEDE Reboot 17.01.1 r3316-7eb58cf109'
DISTRIB_TAINTS='no-all'
root@LEDE:~# cat /etc/openwrt_version
r3316-7eb58cf109

Adding `ifconfig lo up' "solves" the issue:

start() {
        ifconfig lo up
	test -f "$config_directory"/opkg_postinst && postinst
	postfix start
}

I did that test on a clean Lede installation in order to exclude my
configuration. I just flashed Lede onto my router, setup extroot and
installed postfix.

BTW, installing postfix on Lede gives this:

postfix/postfix-script: starting the Postfix mail system
Collected errors:
 * pkg_run_script: package "postfix" postinst script returned status
 1.
  * opkg_configure: postfix.postinst returned 1.

lucize, 2017-07-24:
sorry for the delay
look into /etc/rc.d and see if postfix has a lower number to start than the network, maybe is starting ahead of networking, usually postfi has start and stop set to 50

lucize, 2018-08-29:
@ardrabczyk I encounter the same problem on some CF card system, raising the `START=70` should fix the problem, followed by 
```
/etc/init.d/postfix disable
/etc/init.d/postfix enable
```
you should have  `S70postfix` in  `/etc/rc.d/`

anyhow I'll make a PR

Shulyaka, 2018-08-29:
Do you have a clue why it makes a difference? Just a time delay, or you have an important service between 50 and 70 that might cause such behavior?

lucize, 2018-08-29:
@Shulyaka my actual test system has a slow CF card, and is a single core sempron, maybe the devices initialize slower than usual

feckert, 2019-07-19:
Change https://github.com/openwrt/packages/pull/6906/files#diff-33aa02ffdecf16e628a16bdbb3f0032dR4  should fixed the issue.

closed 2019-07-19 by feckert