openwrt/packages #4343: VPNC-SCRIPTS: Default gateway overrides route to VPN access server

voltivo, 2017-05-10:
Maintainer: Nikos Mavrogiannopoulos <n.mavrogiannopoulos@gmail.com>
Environment: Latest commit a0391c4  on Dec 21, 2015
Description:

When VPN-C connects, it sets a default route to the tap interface:
proto_add_ipv4_route "0.0.0.0" 0

This drops the connection to the VPN Server.

We should first add a host route to vpn servers address via our normal gateway:
do_connect() {
        DFGW="$(netstat -r -n | awk '/:/ { next; } /^(default|0\.0\.0\.0)/ { print $2; }')"
        route add -host $VPNGATEWAY gw $DFGW

This is the same the default vpnc-script does as well.
http://git.infradead.org/users/dwmw2/vpnc-scripts.git/blob_plain/HEAD:/vpnc-script

jow-, 2017-05-10:
This is exactly what is supposed to happen with https://github.com/openwrt/packages/blob/master/net/vpnc/files/vpnc.sh#L46

Do you use vpnc standalone (outside of netifd / network) by any chance?

voltivo, 2017-05-10:
Thank you for the reply!

No, I am using the two standard packages "vpnc" and "vpnc-scripts". The route does not seem to get installed. 

After successful connect, i only see the hostroute for my vpn assigned address added.
'''
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         *               0.0.0.0         U     0      0        0 vpn-cisco
10.1.1.0        *               255.255.255.0   U     0      0        0 br-lan
10.24.85.241    *               255.255.255.255 UH    0      0        0 vpn-cisco
'''

I am wondering about the "interface" option. What should it be set to?
I tried "lan" and "br-lan", but same result with both: No static route to the server.

'''
config interface 'lan'
	option ifname 'eth0'
	option type 'bridge'
	option proto 'static'
	option ipaddr '10.1.1.3'
	option netmask '255.255.255.0'
	option ip6assign '60'
	option gateway '10.1.1.1'
	option dns '10.1.1.1'

config globals 'globals'

config interface 'cisco'
	option proto 'vpnc'
	option server 'removed'
	option username 'removed'
	option authgroup 'removed'
	option passgroup 'removed'
	option interface 'br-lan'
	option password 'removed'
'''

yousong, 2017-05-10:
You can try just commenting out the `interface` option.

voltivo, 2017-05-10:
Dear Yousong, I tried that. No difference. Also, the script Jo was referring to actually seems to require the "interface' as a parameter as well:
( proto_add_host_dependency "vpn-$config" "$ip" $interface )

yousong, 2017-05-13:
Hi @voltivo 

`$interface` is actually optional.  It just came across me that I have actually posted to openwrt-devel before about this [1] ;) 

Actually, the interface config option is now stale for proto openconnect which also depends on vpnc-script.  See [2]

 \[1] https://lists.openwrt.org/pipermail/openwrt-devel/2016-May/041438.html
 \[2] https://github.com/openwrt/packages/pull/4315/commits

27o, 2017-07-21:
Try replacing
``` ( proto_add_host_dependency "vpn-$config" "$ip" $interface )```
with
``` ( proto_add_host_dependency "$config" "$ip" )```

$interface gets set to "-" if it is empty and the call fails.

This worked for me on lede and could be the same with OpenWRT.

laurimyllari, 2017-10-05:
I'm having the same issue on LEDE 17.01.2 and changing 
`( proto_add_host_dependency "vpn-$config" "$ip" $interface )`
to
`( proto_add_host_dependency "$config" "$ip" $interface )`
fixes it.

commit d438de95 (2018-01-19): vpnc: fix using proto_add_host_dependency

Fixes #4343
commit f5046db6 (2018-01-19): vpnc: fix using proto_add_host_dependency

Fixes #4343
closed 2018-01-19 by yousong