openwrt/packages #4345: bind: fails with permission denied error when resolving fqdn recursively

perazim2, 2017-05-10:
-------------------------------

Maintainer: @psycho-nico nico
Environment: ar71xx  openwrt 15.05.1 built from Image Maker to include packages nano and bind

Description: Bind generates permission denied errors when writing to /tmp and then stops resolving. This CAN be intermittent. Below is the syslog after I restarted bind. NOTE: I am using bind with recursion ON, With recursion off, it appears to work.

Let me know if you need more details.

Perazim
perazim@inbox.lt

```
Tue May  9 08:04:55 2017 daemon.notice named[1397]: available at https://www.isc.org/support
Tue May  9 08:04:55 2017 daemon.notice named[1397]: ----------------------------------------------------
Tue May  9 08:04:55 2017 daemon.info named[1397]: using 1 UDP listener per interface
Tue May  9 08:04:55 2017 daemon.info named[1397]: using up to 4096 sockets
Tue May  9 08:04:55 2017 daemon.notice named[1313]: exiting
Tue May  9 08:04:55 2017 daemon.info named[1397]: loading configuration from '/etc/bind/named.conf'
Tue May  9 08:04:56 2017 daemon.info named[1397]: using default UDP/IPv4 port range: [32768, 61000]
Tue May  9 08:04:56 2017 daemon.info named[1397]: using default UDP/IPv6 port range: [32768, 61000]
Tue May  9 08:04:56 2017 daemon.info named[1397]: listening on IPv4 interface lo, 127.0.0.1#53
Tue May  9 08:04:56 2017 daemon.info named[1397]: listening on IPv4 interface eth0, 192.168.1.18#53
Tue May  9 08:04:56 2017 daemon.info named[1397]: listening on IPv4 interface br-lan, 192.168.2.1#53
Tue May  9 08:04:56 2017 daemon.info named[1397]: generating session key for dynamic DNS
Tue May  9 08:04:56 2017 daemon.info named[1397]: sizing zone task pool based on 5 zones
Tue May  9 08:04:56 2017 daemon.info named[1397]: set up managed keys zone for view _default, file 'managed-keys.bind'
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 10.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 16.172.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 17.172.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 18.172.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 19.172.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 20.172.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 21.172.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 22.172.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 23.172.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 24.172.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 25.172.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 26.172.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 27.172.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 28.172.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 29.172.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 30.172.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 31.172.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 168.192.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 64.100.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 65.100.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 66.100.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 75.100.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 76.100.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 77.100.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 78.100.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 79.100.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 80.100.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 81.100.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 82.100.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 83.100.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 84.100.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 85.100.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 86.100.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 87.100.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 88.100.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 89.100.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 90.100.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 91.100.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 92.100.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 93.100.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 94.100.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 95.100.IN-ADDR.ARPA
Tue May  9 08:04:56 2017 daemon.info named[1397]: automatic empty zone: 96.100.IN-ADDR.ARPA
[... 85 lines cut ...]
Tue May  9 08:05:05 2017 daemon.info named[1397]: error (permission denied) resolving 'ns1.first-ns.de/A/IN': 2001:500:a8::e#53
Tue May  9 08:05:05 2017 daemon.info named[1397]: error (permission denied) resolving 'robotns2.second-ns.de/A/IN': 2001:500:a8::e#53
Tue May  9 08:05:05 2017 daemon.info named[1397]: error (permission denied) resolving 'robotns2.second-ns.de/AAAA/IN': 2001:500:a8::e#53
Tue May  9 08:05:05 2017 daemon.info named[1397]: error (permission denied) resolving 'ns1.first-ns.de/AAAA/IN': 2001:500:a8::e#53
Tue May  9 08:05:06 2017 daemon.info named[1397]: error (permission denied) resolving 'luci.subsignal.org/AAAA/IN': 2001:67c:192c::add:a3#53
Tue May  9 08:05:06 2017 daemon.info named[1397]: error (permission denied) resolving 'luci.subsignal.org/A/IN': 2001:67c:192c::add:a3#53
Tue May  9 08:05:06 2017 daemon.info named[1397]: error (permission denied) resolving 'ns1.your-server.de/A/IN': 2001:500:2f::f#53
Tue May  9 08:05:06 2017 daemon.info named[1397]: error (permission denied) resolving 'ns1.your-server.de/A/IN': 2001:503:c27::2:30#53
Tue May  9 08:05:06 2017 daemon.info named[1397]: error (permission denied) resolving 'ns1.your-server.de/A/IN': 2001:668:1f:11::105#53
Tue May  9 08:05:06 2017 daemon.info named[1397]: error (permission denied) resolving 'ns1.your-server.de/A/IN': 2001:67c:192c::add:b3#53
Tue May  9 08:05:06 2017 daemon.info named[1397]: error (permission denied) resolving 'ns1.your-server.de/A/IN': 2a01:4f8:0:a101::b:1#53
Tue May  9 08:05:06 2017 daemon.info named[1397]: error (permission denied) resolving 'z.nic.de/AAAA/IN': 2a02:568:121:6:2::2#53
Tue May  9 08:05:08 2017 daemon.info named[1397]: error (permission denied) resolving 'robotns2.second-ns.de/A/IN': 2001:503:ba3e::2:30#53
Tue May  9 08:05:08 2017 daemon.info named[1397]: error (permission denied) resolving 'robotns2.second-ns.de/AAAA/IN': 2001:503:ba3e::2:30#53
Tue May  9 08:05:12 2017 daemon.info named[1397]: error (permission denied) resolving 'ns1.first-ns.de/A/IN': 2001:67c:1011:1::53#53
Tue May  9 08:05:12 2017 daemon.info named[1397]: error (permission denied) resolving 'ns1.first-ns.de/A/IN': 2001:678:2::53#53
Tue May  9 08:05:12 2017 daemon.info named[1397]: error (permission denied) resolving 'ns1.first-ns.de/A/IN': 2a01:4f8:d0a:2006::2#53
Tue May  9 08:05:36 2017 daemon.info named[1397]: error (permission denied) resolving 'www.stevequayle.com.cdn.cloudflare.net/AAAA/IN': 2400:cb00:2049:1::c629:de83#53
Tue May  9 08:05:37 2017 daemon.notice named[1397]: DNS format error from 38.108.81.8#53 resolving www.kitconet.com/AAAA for client 192.168.2.164#40830: Name kitconet.com (SOA) not subdomain of zone www.kitconet.com -- invalid response
[... cut ...]
Tue May  9 08:05:37 2017 daemon.info named[1397]: error (FORMERR) resolving 'www.kitconet.com/AAAA/IN': 8.21.40.8#53
Tue May  9 08:05:39 2017 daemon.info named[1397]: error (permission denied) resolving 'support.mozilla.org/A/IN': 2001:500:f::1#53
Tue May  9 08:05:39 2017 daemon.info named[1397]: error (permission denied) resolving 'support.mozilla.org/AAAA/IN': 2001:500:f::1#53
Tue May  9 08:05:39 2017 daemon.info named[1397]: error (permission denied) resolving 'sumo.external.zlb.scl3.mozilla.com/A/IN': 2001:503:231d::2:30#53

hnyman, 2017-05-10:
Does that happen also with the current master version? Bind package has got lots of changes after CC15.05 was branched off two years ago.
https://github.com/openwrt/packages/commits/master/net/bind/Makefile

You should test with the current LEDE master snapshot (or stable LEDE 17.01.1) and see if the error materialises with the current versions.

perazim2, 2017-05-17:
Trying to build from master version. I posted the following to the openwrt forum but received nothing I could use:

Building dd from snapshots using ImageBuilder. When I built cc earlier, I got the list of packages to include from a running router. This time with dd from snapshots I don't have the list of packages to use with ImageBuilder.

I tried using the package list that I used with cc and it produced the following build errors:

Collected errors:
* opkg_install_cmd: Cannot install package kmod-crypto-aes.
* opkg_install_cmd: Cannot install package kmod-crypto-arc4.
* opkg_install_cmd: Cannot install package kmod-crypto-core.
* opkg_install_cmd: Cannot install package kmod-ipv6.
make[2]: *** [package_install] Error 255

I am just guessing here that these packages either are not included with dd or the package names have changed.

Can someone point me to the correct package list to use for dd?
I will then test bind on dd and report the results.

Thanks,

Perazim
 

 

 

/*<![CDATA[*/
 p.c1 {color: #666; font-size: small}
/*]]>*/

 
----- Reply to message -----
Subject: Re: [openwrt/packages] bind: fails with permission denied error when resolving fqdn recursively (#4345)
Date: Wednesday, 10 May 2017 at 10:41:41

https://github.com/openwrt/packages/commits/master/net/bind/Makefile

perazim2, 2017-05-24:
Built trunk from lede and errors from bind are still in syslog(see below). I have only added the bind and nano packages. No other config changes. Anything else I should do to resolve this? 

Thanks,

Perazim

Mon May 15 11:06:51 2017 daemon.notice named[711]: starting BIND 9.10.4-P5 <id:2b12043> -u bind -f -c /etc/bind/named.conf
Mon May 15 11:06:51 2017 daemon.notice named[711]: running on Linux mips 4.4.61 #0 Sat Apr 15 16:13:45 2017
Mon May 15 11:06:51 2017 daemon.notice named[711]: built with '--target=mips-openwrt-linux' '--host=mips-openwrt-linux' '--build=x86_64-linux-gnu' '--program-prefix=' '--program-suffix=' '--prefix=/usr' '--exec-prefix=/usr' '--bindir=/usr/bin' '--sbindir=/usr/sbin' '--libexecdir=/usr/lib' '--sysconfdir=/etc' '--datadir=/usr/share' '--localstatedir=/var' '--mandir=/usr/man' '--infodir=/usr/info' '--disable-nls' '--enable-shared' '--enable-static' '--with-randomdev=/dev/urandom' '--disable-threads' '--disablMon May 15 11:06:51 2017 daemon.notice named[711]: ----------------------------------------------------
Mon May 15 11:06:51 2017 daemon.notice named[711]: BIND 9 is maintained by Internet Systems Consortium,
Mon May 15 11:06:51 2017 daemon.notice named[711]: Inc. (ISC), a non-profit 501(c)(3) public-benefit
Mon May 15 11:06:51 2017 daemon.notice named[711]: corporation.  Support and training for BIND 9 are
Mon May 15 11:06:51 2017 daemon.notice named[711]: available at https://www.isc.org/support
Mon May 15 11:06:51 2017 daemon.notice named[711]: ----------------------------------------------------
Mon May 15 11:06:51 2017 daemon.info named[711]: using 1 UDP listener per interface
Mon May 15 11:06:51 2017 daemon.info named[711]: using up to 4096 sockets
Mon May 15 11:06:52 2017 kern.debug kernel: [   20.719062] ar71xx: pll_reg 0xb8050010: 0x1099
Mon May 15 11:06:52 2017 kern.info kernel: [   20.719093] eth0: link up (100Mbps/Full duplex)
Mon May 15 11:06:52 2017 kern.info kernel: [   20.724114] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
Mon May 15 11:06:52 2017 daemon.notice netifd: Network device 'eth0' link is up
Mon May 15 11:06:52 2017 daemon.notice netifd: Interface 'wan' has link connectivity
Mon May 15 11:06:52 2017 daemon.notice netifd: Interface 'wan' is setting up now
Mon May 15 11:06:52 2017 daemon.notice netifd: Interface 'wan6' has link connectivity
Mon May 15 11:06:52 2017 daemon.notice netifd: Interface 'wan6' is setting up now
Mon May 15 11:06:52 2017 daemon.info odhcpd[645]: Using a RA lifetime of 0 seconds on br-lan
Mon May 15 11:06:52 2017 daemon.notice odhcpd[645]: Failed to send to ff02::1%br-lan (Address not available)
Mon May 15 11:06:52 2017 daemon.notice netifd: wan (854): udhcpc: started, v1.25.1
Mon May 15 11:06:52 2017 daemon.info named[711]: loading configuration from '/etc/bind/named.conf'
Mon May 15 11:06:52 2017 daemon.info named[711]: using default UDP/IPv4 port range: [32768, 60999]
Mon May 15 11:06:52 2017 daemon.info named[711]: using default UDP/IPv6 port range: [32768, 60999]
Mon May 15 11:06:52 2017 daemon.notice netifd: wan (854): udhcpc: sending discover
Mon May 15 11:06:52 2017 daemon.notice netifd: wan (854): udhcpc: sending select for 192.168.1.11
Mon May 15 11:06:52 2017 daemon.notice netifd: wan (854): udhcpc: lease of 192.168.1.11 obtained, lease time 86400
Mon May 15 11:06:52 2017 daemon.info named[711]: listening on IPv6 interfaces, port 53
Mon May 15 11:06:52 2017 daemon.err odhcp6c[856]: Failed to send DHCPV6 message to ff02::1:2 (Address not available)
Mon May 15 11:06:53 2017 daemon.info named[711]: listening on IPv4 interface lo, 127.0.0.1#53
Mon May 15 11:06:53 2017 daemon.info named[711]: listening on IPv4 interface br-lan, 192.168.2.1#53
Mon May 15 11:06:53 2017 daemon.info named[711]: generating session key for dynamic DNS
Mon May 15 11:06:53 2017 daemon.info odhcpd[645]: Using a RA lifetime of 0 seconds on br-lan
Mon May 15 11:06:53 2017 daemon.notice netifd: Interface 'wan' is now up
Mon May 15 11:06:53 2017 daemon.info named[711]: sizing zone task pool based on 5 zones
Mon May 15 11:06:53 2017 kern.info kernel: [   21.939243] br-lan: port 1(eth1) entered forwarding state
Mon May 15 11:06:53 2017 user.notice firewall: Reloading firewall due to ifup of wan (eth0)
Mon May 15 11:06:53 2017 daemon.notice odhcpd[645]: Got DHCPv6 request
Mon May 15 11:06:53 2017 daemon.warn odhcpd[645]: DHCPV6 CONFIRM IA_NA from 0004f267e842ba4cd1f92f6ed923f0790c5b on br-lan: not on-link
Mon May 15 11:06:53 2017 daemon.info named[711]: set up managed keys zone for view _default, file 'managed-keys.bind'
Mon May 15 11:06:53 2017 daemon.info named[711]: automatic empty zone: 10.IN-ADDR.ARPA
Mon May 15 11:06:53 2017 daemon.info named[711]: automatic empty zone: 16.172.IN-ADDR.ARPA
Mon May 15 11:06:53 2017 daemon.info named[711]: automatic empty zone: 17.172.IN-ADDR.ARPA
Mon May 15 11:06:53 2017 daemon.info named[711]: automatic empty zone: 18.172.IN-ADDR.ARPA
Mon May 15 11:06:53 2017 daemon.info named[711]: automatic empty zone: 19.172.IN-ADDR.ARPA
Mon May 15 11:06:53 2017 daemon.info named[711]: automatic empty zone: 20.172.IN-ADDR.ARPA
Mon May 15 11:06:53 2017 daemon.info named[711]: automatic empty zone: 21.172.IN-ADDR.ARPA
Mon May 15 11:06:53 2017 daemon.info named[711]: automatic empty zone: 22.172.IN-ADDR.ARPA
Mon May 15 11:06:53 2017 daemon.info named[711]: automatic empty zone: 23.172.IN-ADDR.ARPA
Mon May 15 11:06:53 2017 daemon.notice odhcpd[645]: Got DHCPv6 request
[... cut ...]
Tue May 23 12:48:16 2017 daemon.info named[711]: permission denied resolving 'ns-576.awsdns-08.net/A/IN': 2001:503:ba3e::2:30#53
Tue May 23 12:48:16 2017 daemon.info named[711]: permission denied resolving 'ns-576.awsdns-08.net/A/IN': 2001:500:2d::d#53

 

 

/*<![CDATA[*/
 p.c1 {color: #666; font-size: small}
/*]]>*/

 
----- Reply to message -----
Subject: Re: [openwrt/packages] bind: fails with permission denied error when resolving fqdn recursively (#4345)
Date: Wednesday, 10 May 2017 at 10:41:41

https://github.com/openwrt/packages/commits/master/net/bind/Makefile

neheb, 2020-01-12:
Closing out of age. Please open a new Issue if this is still a problem.

closed 2020-01-12 by neheb