openwrt/packages #4349: perl-html-tagset: no rebuild when perl version changes

nwf, 2017-05-11:
Maintainer: @Naoir @michailf 
Environment: LEDE head
Description:

Upon `lang/perl` being bumped from 5.22 to 5.24, `perl-html-tagset` failed to build:

```
make[3]: Entering directory '/tank/openwrt/scratch/builder-webserver/feeds/packages/lang/perl-html-tagset'
mkdir -p /tank/openwrt/scratch/builder-webserver/staging_dir/target-arm_xscale_musl_eabi/root-kirkwood/stamp
SHELL= flock /tank/openwrt/scratch/builder-webserver/tmp/.root-copy.flock -c 'cp -fpR /tank/openwrt/scratch/builder-webserver/build_dir/target-arm_xscale_musl_eabi/perl/HTML-Tagset-3.20/.pkgdir/perl-html-tagset/. /tank/openwrt/scratch/builder-webserver/staging_di
r/target-arm_xscale_musl_eabi/root-kirkwood/'
touch /tank/openwrt/scratch/builder-webserver/staging_dir/target-arm_xscale_musl_eabi/root-kirkwood/stamp/.perl-html-tagset_installed
mkdir -p /tank/openwrt/scratch/builder-webserver/bin/targets/kirkwood/generic/packages /tank/openwrt/scratch/builder-webserver/build_dir/target-arm_xscale_musl_eabi/perl/HTML-Tagset-3.20/ipkg-arm_xscale/perl-html-tagset/CONTROL /tank/openwrt/scratch/builder-webse
rver/staging_dir/target-arm_xscale_musl_eabi/pkginfo
install -d -m0755 /tank/openwrt/scratch/builder-webserver/build_dir/target-arm_xscale_musl_eabi/perl/HTML-Tagset-3.20/ipkg-arm_xscale/perl-html-tagset/usr/lib/perl5/5.24
(cd /tank/openwrt/scratch/builder-webserver/build_dir/target-arm_xscale_musl_eabi/perl/HTML-Tagset-3.20/ipkg-install/usr/lib/perl5/5.24 && rsync --relative -rlHp --itemize-changes --exclude=\*.pod --exclude=.packlist  --prune-empty-dirs HTML /tank/openwrt/scratch
/builder-webserver/build_dir/target-arm_xscale_musl_eabi/perl/HTML-Tagset-3.20/ipkg-arm_xscale/perl-html-tagset/usr/lib/perl5/5.24)
bash: line 0: cd: /tank/openwrt/scratch/builder-webserver/build_dir/target-arm_xscale_musl_eabi/perl/HTML-Tagset-3.20/ipkg-install/usr/lib/perl5/5.24: No such file or directory
Makefile:48: recipe for target '/tank/openwrt/scratch/builder-webserver/bin/packages/arm_xscale/packages/perl-html-tagset_3.20-3_arm_xscale.ipk' failed
make[3]: *** [/tank/openwrt/scratch/builder-webserver/bin/packages/arm_xscale/packages/perl-html-tagset_3.20-3_arm_xscale.ipk] Error 1
```

Instead, the old 5.22 directory remains around; this can be worked around by nuking `/tank/openwrt/scratch/builder-webserver/build_dir/target-arm_xscale_musl_eabi/perl/HTML-Tagset-3.20` and rebuilding.

hnyman, 2017-05-11:
> Upon lang/perl being bumped from 5.22 to 5.24, perl-html-tagset failed to build
> ...
> Instead, the old 5.22 directory remains around; this can be worked around by nuking /tank/openwrt/scratch/builder-webserver/build_dir/target-arm_xscale_musl_eabi/perl/HTML-Tagset-3.20 and rebuilding.

It sounds quite normal that you need to do a "make clean" before building after a version bump in related components.

nwf, 2017-05-11:
That strikes me as somewhat at odds with the effort the system goes through with, e.g. PKG_CONFIG_DEPENDS, which seeks to eliminate the need to run "make clean" after config changes.  "make clean" is also really heavy-handed when only the perl modules need to be recompiled, for example; most of the rest of the build tree is fine and unaffected.

neheb, 2020-01-12:
This is outdated AFAIK. Open a new issue if still a problem.

closed 2020-01-12 by neheb