openwrt/packages #4350: coova-chilli: Enhanced Feed Makefile

zelda14, 2017-05-11:
Please make sure that the issue subject starts with `<package-name>: ` so that it's easily identifiable.

This repo here is ONLY for packages maintained in this repo.  For base packages residing in the same repo as the build system and maintained by core devs, please consider opening tickets there for more timely responses

 - OpenWrt: https://dev.openwrt.org/newticket
 - LEDE: https://bugs.lede-project.org/
 - Most LuCI packages: https://github.com/openwrt/luci/issues

Thanks for your contribution
Please remove this text (before ---) and fill the following template
-------------------------------

Maintainer: @<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWRT/LEDE version)

Description:

```
Formating code blocks by wrapping them with pairs of ```
```

zelda14, 2017-05-11:
The default feed Makefile for coova-chilli-1.3.0+20141128 available at OpenWRT/LEDE exposes a minimal set of compile options. The following attachments expose all the available compile option. I provide either the needed files for coova-chilli-1.3.0+20141128 and coova-chilli-1.4.
Thanks
[coova-chilli-1.3.0+20141128.tar.gz](https://github.com/openwrt/packages/files/992638/coova-chilli-1.3.0.20141128.tar.gz)
[coova-chilli-1.4.tar.gz](https://github.com/openwrt/packages/files/992642/coova-chilli-1.4.tar.gz)

dibdot, 2018-08-18:
feel free to open a PR.

closed 2018-08-18 by dibdot