openwrt/packages #4352: luci-app-sqm: Link Layer Adaptation settings not applied to (cake) qdisc

HenkPoley, 2017-05-12:
Maintainer: @tohojo (find it by checking history of the package Makefile)
Environment: ar71xx, TP-Link WDR4300, LEDE 17.01.1

Description:
When setting the Link Layer Adaptation to ATM & 'Per Packet Overhead (byte):' to 44, this is not shown in the output of `tc qdisc`. I can currently only assume that this setting is lost somewhere along the way when it's being applied, and _not_ actually set but hidden.

```
root@OpenWrt:/etc/config# tc qdisc
qdisc noqueue 0: dev lo root refcnt 2 
qdisc fq_codel 0: dev eth0 root refcnt 2 limit 10240p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn 
qdisc noqueue 0: dev br-lan root refcnt 2 
qdisc noqueue 0: dev eth0.1 root refcnt 2 
qdisc noqueue 0: dev br-wan root refcnt 2 
qdisc cake 8032: dev eth0.2 root refcnt 2 bandwidth 1029Kbit besteffort triple-isolate rtt 100.0ms raw 
qdisc ingress ffff: dev eth0.2 parent ffff:fff1 ---------------- 
qdisc noqueue 0: dev wlan0 root refcnt 2 
qdisc noqueue 0: dev wlan1 root refcnt 2 
qdisc cake 8033: dev ifb4eth0.2 root refcnt 2 bandwidth 19Mbit besteffort triple-isolate wash rtt 100.0ms raw 
```

Expected:
Something along the lines of "overhead 44 atm" showing up in the output of `tc qdisc` at eth0.2. E.g. with the 'Advanced option string' set:

```
root@OpenWrt:/etc/config# tc qdisc
qdisc noqueue 0: dev lo root refcnt 2 
qdisc fq_codel 0: dev eth0 root refcnt 2 limit 10240p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn 
qdisc noqueue 0: dev br-lan root refcnt 2 
qdisc noqueue 0: dev eth0.1 root refcnt 2 
qdisc noqueue 0: dev br-wan root refcnt 2 
qdisc cake 8035: dev eth0.2 root refcnt 2 bandwidth 1029Kbit besteffort triple-isolate rtt 100.0ms atm overhead 44 via-ethernet 
qdisc ingress ffff: dev eth0.2 parent ffff:fff1 ---------------- 
qdisc noqueue 0: dev wlan0 root refcnt 2 
qdisc noqueue 0: dev wlan1 root refcnt 2 
qdisc cake 8036: dev ifb4eth0.2 root refcnt 2 bandwidth 19Mbit besteffort triple-isolate wash rtt 100.0ms raw 
```

tohojo, 2017-05-12:
@moeller0 thoughts? :)

HenkPoley, 2017-05-12:
Output of log-level 'trace' with 'Advanced option string' unset. The number 44 does show up on two lines.

```
Fri May 12 10:24:21 2017 user.notice SQM: 
Fri May 12 10:24:21 2017 user.notice SQM: Fri May 12 10:24:21 CEST 2017: Stopping.
Fri May 12 10:24:21 2017 user.notice SQM: Stopping SQM on eth0.2
Fri May 12 10:24:21 2017 user.notice SQM: /usr/sbin/tc -p filter show parent ffff: dev eth0.2
Fri May 12 10:24:21 2017 user.notice SQM: ifb associated with interface eth0.2: 
Fri May 12 10:24:21 2017 user.notice SQM: /usr/sbin/tc -p filter show parent ffff: dev eth0.2
Fri May 12 10:24:21 2017 user.notice SQM: Currently no ifb is associated with eth0.2, this is normal during starting of the sqm system.
Fri May 12 10:24:21 2017 user.notice SQM: /usr/sbin/tc qdisc del dev eth0.2 ingress
Fri May 12 10:24:21 2017 user.notice SQM: /usr/sbin/tc qdisc del dev eth0.2 root
Fri May 12 10:24:21 2017 user.notice SQM: /usr/sbin/tc qdisc del dev ifb4eth0.2 root
Fri May 12 10:24:22 2017 user.notice SQM: /usr/lib/sqm/stop-sqm: ifb4eth0.2 shaper deleted
Fri May 12 10:24:22 2017 user.notice SQM: iptables -t mangle -D POSTROUTING -o ifb4eth0.2 -m mark --mark 0x00 -g QOS_MARK_eth0.2
Fri May 12 10:24:22 2017 user.notice SQM: ip6tables -t mangle -D POSTROUTING -o ifb4eth0.2 -m mark --mark 0x00 -g QOS_MARK_eth0.2
Fri May 12 10:24:22 2017 user.notice SQM: iptables -t mangle -D POSTROUTING -o eth0.2 -m mark --mark 0x00/0xff -g QOS_MARK_eth0.2
Fri May 12 10:24:22 2017 user.notice SQM: ip6tables -t mangle -D POSTROUTING -o eth0.2 -m mark --mark 0x00/0xff -g QOS_MARK_eth0.2
Fri May 12 10:24:22 2017 user.notice SQM: iptables -t mangle -D PREROUTING -i vtun+ -p tcp -j MARK --set-mark 0x2/0xff
Fri May 12 10:24:22 2017 user.notice SQM: ip6tables -t mangle -D PREROUTING -i vtun+ -p tcp -j MARK --set-mark 0x2/0xff
Fri May 12 10:24:22 2017 user.notice SQM: iptables -t mangle -D PREROUTING -i eth0.2 -m dscp ! --dscp 0 -j DSCP --set-dscp-class be
Fri May 12 10:24:22 2017 user.notice SQM: ip6tables -t mangle -D PREROUTING -i eth0.2 -m dscp ! --dscp 0 -j DSCP --set-dscp-class be
Fri May 12 10:24:22 2017 user.notice SQM: iptables -t mangle -D PREROUTING -i eth0.2 -m mark --mark 0x00/0xff -g QOS_MARK_eth0.2
Fri May 12 10:24:22 2017 user.notice SQM: ip6tables -t mangle -D PREROUTING -i eth0.2 -m mark --mark 0x00/0xff -g QOS_MARK_eth0.2
Fri May 12 10:24:22 2017 user.notice SQM: iptables -t mangle -D OUTPUT -p udp -m multiport --ports 123,53 -j DSCP --set-dscp-class AF42
Fri May 12 10:24:22 2017 user.notice SQM: ip6tables -t mangle -D OUTPUT -p udp -m multiport --ports 123,53 -j DSCP --set-dscp-class AF42
Fri May 12 10:24:22 2017 user.notice SQM: iptables -t mangle -F QOS_MARK_eth0.2
Fri May 12 10:24:22 2017 user.notice SQM: ip6tables -t mangle -F QOS_MARK_eth0.2
Fri May 12 10:24:22 2017 user.notice SQM: iptables -t mangle -X QOS_MARK_eth0.2
Fri May 12 10:24:22 2017 user.notice SQM: ip6tables -t mangle -X QOS_MARK_eth0.2
Fri May 12 10:24:22 2017 user.notice SQM: /sbin/ip link set dev ifb4eth0.2 down
Fri May 12 10:24:22 2017 user.notice SQM: /sbin/ip link delete ifb4eth0.2 type ifb
Fri May 12 10:24:22 2017 user.notice SQM: /usr/lib/sqm/stop-sqm: ifb4eth0.2 interface deleted
Fri May 12 10:24:22 2017 user.notice SQM: 
Fri May 12 10:24:22 2017 user.notice SQM: Fri May 12 10:24:22 CEST 2017: Starting.
Fri May 12 10:24:22 2017 user.notice SQM: Starting SQM script: piece_of_cake.qos on eth0.2, in: 19000 Kbps, out: 1029 Kbps
Fri May 12 10:24:22 2017 daemon.err insmod: module is already loaded - sch_cake
Fri May 12 10:24:22 2017 daemon.err insmod: module is already loaded - sch_ingress
Fri May 12 10:24:22 2017 daemon.err insmod: module is already loaded - act_mirred
Fri May 12 10:24:22 2017 daemon.err insmod: module is already loaded - cls_fw
Fri May 12 10:24:22 2017 daemon.err insmod: module is already loaded - cls_flow
Fri May 12 10:24:22 2017 daemon.err insmod: module is already loaded - cls_u32
Fri May 12 10:24:22 2017 daemon.err insmod: module is already loaded - sch_htb
Fri May 12 10:24:22 2017 daemon.err insmod: module is already loaded - sch_hfsc
Fri May 12 10:24:22 2017 user.notice SQM: /sbin/ip link add name TMP_IFB_4_SQM type ifb
Fri May 12 10:24:22 2017 user.notice SQM: /usr/sbin/tc qdisc replace dev TMP_IFB_4_SQM root cake
Fri May 12 10:24:22 2017 user.notice SQM: QDISC cake is useable.
Fri May 12 10:24:22 2017 user.notice SQM: /sbin/ip link set dev TMP_IFB_4_SQM down
Fri May 12 10:24:22 2017 user.notice SQM: /sbin/ip link delete TMP_IFB_4_SQM type ifb
Fri May 12 10:24:22 2017 user.notice SQM: Starting piece_of_cake.qos
Fri May 12 10:24:22 2017 user.notice SQM: /usr/sbin/tc -p filter show parent ffff: dev eth0.2
Fri May 12 10:24:22 2017 user.notice SQM: ifb associated with interface eth0.2: 
Fri May 12 10:24:22 2017 user.notice SQM: /usr/sbin/tc -p filter show parent ffff: dev eth0.2
Fri May 12 10:24:22 2017 user.notice SQM: Currently no ifb is associated with eth0.2, this is normal during starting of the sqm system.
Fri May 12 10:24:22 2017 user.notice SQM: /sbin/ip link add name ifb4eth0.2 type ifb
Fri May 12 10:24:22 2017 user.notice SQM: egress
Fri May 12 10:24:22 2017 user.notice SQM: /usr/sbin/tc qdisc del dev eth0.2 root
Fri May 12 10:24:23 2017 user.notice SQM: STAB: stab mtu 2047 tsize 512 mpu 0 overhead 44 linklayer atm
Fri May 12 10:24:23 2017 user.notice SQM: /usr/sbin/tc qdisc add dev eth0.2 root stab mtu 2047 tsize 512 mpu 0 overhead 44 linklayer atm cake bandwidth 1029kbit besteffort
[... cut ...]
Fri May 12 10:24:23 2017 user.notice SQM: /usr/sbin/tc qdisc add dev ifb4eth0.2 root stab mtu 2047 tsize 512 mpu 0 overhead 44 linklayer atm cake bandwidth 19000kbit besteffort wash
Fri May 12 10:24:23 2017 user.notice SQM: /sbin/ip link set dev ifb4eth0.2 up
Fri May 12 10:24:23 2017 user.notice SQM: /usr/sbin/tc filter add dev eth0.2 parent ffff: protocol all prio 10 u32 match u32 0 0 flowid 1:1 action mirred egress redirect dev ifb4eth0.2
Fri May 12 10:24:23 2017 user.notice SQM: ingress shaping activated
Fri May 12 10:24:23 2017 user.notice SQM: piece_of_cake.qos was started on eth0.2 successfully
```

moeller0, 2017-05-12:
Well, in lede we still default to the tc_stab method for link layer accounting for all qdiscs, so unless the user manually selects cake as link layer adjustment method cake will not report the configured overhead at all, so this is expected. And for the stab method tc will only report details if asked nicely. See the difference in output between "tc qdisc" and "tc -d qdisc" below:

```
root@router:/usr/lib/sqm# tc qdisc
qdisc noqueue 0: dev lo root refcnt 2 
qdisc fq_codel 0: dev eth0 root refcnt 2 limit 10240p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn 
qdisc fq_codel 0: dev eth1 root refcnt 2 limit 10240p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn 
qdisc noqueue 0: dev br-lan root refcnt 2 
qdisc noqueue 0: dev eth0.1 root refcnt 2 
qdisc noqueue 0: dev wlan0 root refcnt 2 
qdisc noqueue 0: dev wlan1 root refcnt 2 
qdisc cake 800a: dev pppoe-wan root refcnt 2 bandwidth 9545Kbit diffserv3 dual-srchost nat rtt 100.0ms raw 
qdisc ingress ffff: dev pppoe-wan parent ffff:fff1 ---------------- 
qdisc cake 800b: dev ifb4pppoe-wan root refcnt 2 bandwidth 46246Kbit diffserv3 dual-dsthost nat rtt 100.0ms raw 

root@router:/usr/lib/sqm# tc -d qdisc
qdisc noqueue 0: dev lo root refcnt 2 
qdisc fq_codel 0: dev eth0 root refcnt 2 limit 10240p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn 
qdisc fq_codel 0: dev eth1 root refcnt 2 limit 10240p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn 
qdisc noqueue 0: dev br-lan root refcnt 2 
qdisc noqueue 0: dev eth0.1 root refcnt 2 
qdisc noqueue 0: dev wlan0 root refcnt 2 
qdisc noqueue 0: dev wlan1 root refcnt 2 
qdisc cake 800a: dev pppoe-wan root refcnt 2 bandwidth 9545Kbit diffserv3 dual-srchost nat rtt 100.0ms raw 
 linklayer ethernet overhead 34 mpu 64 mtu 2047 tsize 128 
qdisc ingress ffff: dev pppoe-wan parent ffff:fff1 ---------------- 
qdisc cake 800b: dev ifb4pppoe-wan root refcnt 2 bandwidth 46246Kbit diffserv3 dual-dsthost nat rtt 100.0ms raw 
 linklayer ethernet overhead 34 mpu 64 mtu 2047 tsize 128 

```

As you can see cake does never report tc stab configured overhead (as it is not aware of that at all, all it sees are already increased packet length values in the skb), and tc needs the "-d" flag to report details. I hope this clears things up. One hint that tc-stab is working can be gleaned from looking at the reported max_len values from cakes statistics output (tc -s qdisc), say for a pppoe interface with maxMTU of 1492 and overhead 34, if you might see max_len = 1526 indicating that stab is doing its thing...

Hope that clears things up...

Please note that newer sqm-scripts should default to using cake's in-built method for link layer adjustements if cake is selected as qdisc, but the "tc -d" requirement still exists for non-cake as qdisc...

moeller0, 2017-05-12:
So @HenkPoley try setting the link layer adjustment setting to cake (bottom selection box in the link layer adjustment tab in the GUI).

moeller0, 2017-05-12:
As I hinted at, try “tc -d qdisc” to see whether tc_stab is configured correctly.

Best Regards

moeller0, 2017-05-12:
BTW, advanced option string and tc_stab (if set as $LLAM) will both be active at te same time, probably not something anybody expects, but the warnings around the advanced option strings are there for a reason, as they give our users enough rope to shoot themselves in the foot...

HenkPoley, 2017-05-12:
Okay, this checks out. Sorry for the noise 🤔

```
root@OpenWrt:/etc/config# tc -d qdisc
qdisc noqueue 0: dev lo root refcnt 2 
qdisc fq_codel 0: dev eth0 root refcnt 2 limit 10240p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn 
qdisc noqueue 0: dev br-lan root refcnt 2 
qdisc noqueue 0: dev eth0.1 root refcnt 2 
qdisc noqueue 0: dev br-wan root refcnt 2 
qdisc cake 8038: dev eth0.2 root refcnt 2 bandwidth 1029Kbit besteffort triple-isolate rtt 100.0ms raw 
 linklayer atm overhead 44 mtu 2047 tsize 512 
qdisc ingress ffff: dev eth0.2 parent ffff:fff1 ---------------- 
qdisc noqueue 0: dev wlan0 root refcnt 2 
qdisc noqueue 0: dev wlan1 root refcnt 2 
qdisc cake 8039: dev ifb4eth0.2 root refcnt 2 bandwidth 19Mbit besteffort triple-isolate wash rtt 100.0ms raw 
 linklayer atm overhead 44 mtu 2047 tsize 512
```

And with the Advanced Linklayer ATM option set to Cake:
```
root@OpenWrt:/etc/config# tc -d qdisc
qdisc noqueue 0: dev lo root refcnt 2 
qdisc fq_codel 0: dev eth0 root refcnt 2 limit 10240p flows 1024 quantum 1514 target 5.0ms interval 100.0ms ecn 
qdisc noqueue 0: dev br-lan root refcnt 2 
qdisc noqueue 0: dev eth0.1 root refcnt 2 
qdisc noqueue 0: dev br-wan root refcnt 2 
qdisc cake 803b: dev eth0.2 root refcnt 2 bandwidth 1029Kbit besteffort triple-isolate rtt 100.0ms atm overhead 44 via-ethernet 
qdisc ingress ffff: dev eth0.2 parent ffff:fff1 ---------------- 
qdisc noqueue 0: dev wlan0 root refcnt 2 
qdisc noqueue 0: dev wlan1 root refcnt 2 
qdisc cake 803c: dev ifb4eth0.2 root refcnt 2 bandwidth 19Mbit besteffort triple-isolate wash rtt 100.0ms atm overhead 44 via-ethernet 
```

closed 2017-05-12 by HenkPoley

moeller0, 2017-05-12:
Great, that it works, and kudos for your detail-oriented approach of looking at sqm-scripts. We tried to make this issue less noticeable by selecting cake's LLAM automatically if cake is selected as qdisc, but it seems that change has not made it into LEDE's sqm-scripts/luci-app-sqm yet.

HenkPoley, 2017-05-12:
I'm seeing odd upload saw-toothing on DSLreports speedtest that I didn't notice before. And huge bufferbloat when running the test over HTTPS. Recently updated to a new LEDE. So I went digging. 

I think the problems I'm seeing are at their servers.