openwrt/packages #4357: jq: assert failed on ar71xx (mips_24kc)

luizluca, 2017-05-13:
Maintainer: @ratkaj
Environment: LEDE 17.01 ar71xx (mips_24kc)
Description: jq fails on any call

Hi,

I noticed the addition of jq to openwrt-packages. I compiled it to lede 17.01. However, when I run it, I only get a assertion error:

```
# jq
Assertion failed: jv_get_kind(a) == JV_KIND_STRING (jv.c: jvp_string_ptr: 435)
Aborted
```
Only '--help' and '--version' seems to be working.

For testing, I also downloaded the binary from lede snapshot and installed it in 17.01. The result is the same.

ratkaj, 2017-05-13:
Hi luizluca,

thank you for reporting this issue.
I cant reproduce the issue on my mvebu test board, all examples from the below link work fine:
https://stedolan.github.io/jq/tutorial/

However, image im using is bloated, so i will rebuild and test on mips if necessary. I will get back to you soon.

luizluca, 2017-05-14:
Sure, my system is a tl-wr2543nd.

It seems that it is not a syntax problem. So, no example case will work. Even running jq without args gives me the assertion failure. At least with my ubuntu linux, jq should give me the help msg when there are no args.

I can run a gdb if you need to grab any var info at the point assertion fails.

ratkaj, 2017-05-14:
Unfortunately i cant get the mips image to work with Qemu and i don't have a mips board nearby at the moment :/

Could you please try adding the following line in the jq/Makefile ?
PKG_USE_MIPS16:=0

Then rebuild the package. This might fix the issue.

luizluca, 2017-05-16:
Sorry, no difference.

luizluca, 2017-05-16:
I runned gdb of jq without args. This is where assert fails:

```
(gdb) bt full
#0  __assert_fail (expr=<optimized out>, file=<optimized out>, line=<optimized out>, func=<optimized out>) at src/exit/assert.c:5
No locals.
#1  0x00401890 in jvp_string_ptr (a=...) at jv.c:435
No locals.
#2  0x004101c8 in jvp_string_ptr (a=...) at jv.c:540
No locals.
#3  jvp_string_hash (jstr=...) at jv.c:541
        data = 0x4391c8 "jv.c"
        len = <optimized out>
        nblocks = <optimized out>
        h1 = <optimized out>
        tail = 0x1 <error: Cannot access memory at address 0x1>
        k1 = <optimized out>
#4  0x00410580 in jvp_object_find_bucket (object=..., key=...) at jv.c:912
No locals.
#5  0x00413094 in jvp_object_write (key=..., object=0x7ffff9b0) at jv.c:1019
        bucket = <optimized out>
        slot = <optimized out>
#6  jv_object_set (object=..., key=..., value=...) at jv.c:1112
        __func__ = "jv_object_set"
#7  0x004078ec in jq_set_attr (jq=jq@entry=0x77ffef00, attr=..., val=...) at execute.c:1126
No locals.
#8  0x00402c00 in main (argc=<optimized out>, argv=0x7ffffdd4) at main.c:445
        jq = 0x77ffef00
        ret = 0
        compiled = 0
        parser_flags = 0
        nfiles = 0
        badwrite = <optimized out>
        program_arguments = {kind_flags = 6 '\006', pad_ = 0 '\000', offset = 0, size = 0, u = {ptr = 0x77ffede0, number = 1.0542513331754238e+270}}
        dumpopts = 645
        program = 0x0
        input_state = 0x453010
        further_args_are_files = <optimized out>
        jq_flags = 0
        short_opts = 0
        lib_search_paths = {kind_flags = 6 '\006', pad_ = 0 '\000', offset = 0, size = 3, u = {ptr = 0x77f581a0, number = 7.1010536571870043e+269}}
        origin = <optimized out>

```

ratkaj, 2017-05-16:
Thank you for testing and running gdb. 

I got my hands on a mips board so i was able to reproduce the problem. Jq seems badly broken on MIPS and im gona be a bit busy next few days so i might take a while.

ratkaj, 2017-05-17:
@luizluca 

I managed to fix the issue. Pull request has been created and should be accepted soon. I also imported some general improvements from debian. 

Pull request on the link below:
https://github.com/openwrt/packages/pull/4364

Thank you for reporting the issue.

luizluca, 2017-05-18:
Thanks @ratkaj ! This kind of bug is not the easiest one to fix. compiler normally is the last one you should blame.

I guess the problem is not only with MIPS16. I do not have USE_MIPS16 set and the result was the same. I tested removing ifdef and I can confirm it fixed the problem with mips32. Maybe you can change the "ifdef CONFIG_USE_MIPS16" to something more general that matches both mips16 and mips32. Maybe CONFIG_ARCH?

Also, you might need to bump release in order to let users upgrade their packages.

This seems to be a bug in gcc. I did a quick search and found pretty old references to it (it dates are correct):
https://github.com/soapdog/b2g-mips/blob/master/emulator/gecko-0006-MIPS-Disable-ipa-sra-optimization-beacuse-of-gcc-bug.patch

Maybe the bug should be submitted to gcc as well if not caused by OpenWRT/LEDE optimizations and still present on latest gcc release. If not, the gcc bug should be submitted to LEDE.

closed 2018-08-26 by luizluca