openwrt/packages #4376: python not installable under LEDE, missing hostpkg/pip

WereCatf, 2017-05-23:
Python compiles fine, but fails when it's supposed to get installed with "/blabla/staging_dir/hostpkg/bin/pip: No such file or directory" -- indeed, there is no pip-binary there.

hnyman, 2017-05-23:
which LEDE version, which python?

cc @commodo

WereCatf, 2017-05-23:
Seems to be Python-2.7.13. And whatever is the latest LEDE-version available on their github.

commodo, 2017-05-23:
Will check

commodo, 2017-05-23:
@hkallweit 
seems to work fine for me

i managed to build an x86 image (vm) and ran the python & pip commands successfully
this was with LEDE trunk

was this with a clean build that you saw it ?
and for which target ?
what about the host machine ? [ i'm using xubuntu 16.04 amd64 ]

WereCatf, 2017-05-23:
Maybe it's because I was building for the nbg6716, ie. ar71xx. Tried with both Ubuntu 16.04 and 17.10 amd 64

WereCatf, 2017-05-24:
Actually, the issue is apparently the python-pip - package, not python itself. I forgot I had python-pip selected and now that I took another look at it, python compiles and install fine if python-pip isn't selected. Why it breaks python, I can't say.

WereCatf, 2017-05-24:
I was wrong, it's not specifically the python-package itself which is broken, it's python-setuptools and/or python-pip which break the installation of python. A clean build without either of those packages selected works, but a build with either of them or both fails at installation-time.

commodo, 2017-05-24:
`python-pip` depends on `python-setuptools`

normally the `make defconfig` & `make menuconfig` commands would enforce the dependency chain of `python-pip` requiring `python-setuptools` requiring `python`

that means:
* disabling `python-setuptools` but enabling `python-pip` is an invalid use-case, because that means you probably disabled it manually from `.config` file ; and you should run `make defconfig` to make all dependency chains be sane
* disabling `python-pip but enabling `python-setuptools` is valid; ; that also seems to build for me

atm, i have trouble trying to identify the issue 
and your details are a bit vague sometimes
as for buildbot faillogs [ i.e. here http://downloads.lede-project.org/snapshots/faillogs/ ] i don't see any python/python-pip failing there [ tho, not all targets have built recently ]

the ar71xx [ or mips_24kc ] seem to have built yesterday: http://downloads.lede-project.org/snapshots/faillogs/mips_24kc/packages/
and no failure there regarding python

[ tho, i do admit, I have to fix OVS :p  ; been meaning to do it for a while ; maybe i'll make time real quick now ]

WereCatf, 2017-05-24:
Okay, you're right, I have been a bit confusing, but I'll try to be clearer: here is a config-file that fails to install python https://pastebin.com/UJB5jWrX

I simply copied the LEDE github, updated and installed the feeds, went to menuconfig and the only things I changed was enable python and python-setuptools and went to compile. It fails when compiling libubox for the target-machine, but that's easy enough to fix, it's just looking for the json-headers in the wrong place. After that's fixed, it'll continue on and then fail at trying to install python with:

> # Use host pip to install python-setuptools
> /home/nita/Työpöytä/source/staging_dir/hostpkg/bin/pip install --ignore-installed --root=/home/nita/Työpöytä/source/build_dir/target-mips_24kc_musl/Python-2.7.13/install-setuptools --prefix=. /home/nita/Työpöytä/source/build_dir/target-mips_24kc_musl/Python-2.7.13/Lib/ensurepip/_bundled/setuptools-*.whl
> bash: /home/nita/Työpöytä/source/staging_dir/hostpkg/bin/pip: No such file or directory
> 

I just tested it like I mentioned above. I hope this clarifies things up.

commodo, 2017-05-24:
ok ; will try
thanks :)

commodo, 2017-05-24:
i downloaded from pastebin via download link

just with that config i get a lot of 
```
MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINLEN) [6] (NEW) Minimum password length (PASSWORD_MINL [... cut ...]
* download your config + copy over my .config
* run `make defconfig`
* make V=99 -j12
* this builds for me

at this point, i think it may be something related to your locale settings and/or maybe your build path

could you try creating a non-UTF-8 path for testing ?
i'm seeing you have `/home/nita/Työpöytä/`  
maybe try `/home/nita/lede`  ?
i do know that python is not particularly good with non-ASCII characters, and needs some extra handling ; but, not sure if that's the problem here ; i will try with some non-ASCII paths later

WereCatf, 2017-05-25:
I'm just being an idiot. I had completely, totally forgotten about "make defconfig"

Apologies. Imma go in the corner and don my dunce-hat.

closed 2017-05-25 by WereCatf

commodo, 2017-05-25:
No worries :)
I've had my share of goofs.

I'm just glad it's not a weird/hard-to-replicate issue (like maybe UTF stuff).
I hate those