openwrt/packages #4386: dnscrypt-proxy not starting up without ipv6 support

simontretter, 2017-05-26:
Maintainer: @damianorenfer 
Environment: broken on all architectures / models (ar71xx, dir-825, LEDE r3398-215c1d05b8)

Description:
When dnscrypt-proxy plugin support is selected (DNSCRYPT_ENABLE_PLUGINS=y) but LEDE is built without IPV6 support for packages (CONFIG_IPV6), dnscrypt-proxy doesn't start up anymore.

This is probably due to the init script creating /var/etc/dnscrypt-proxy-*.conf with the "BlockIPv6 yes|no" line that is not being understood by dnscrypt-proxy lacking IPV6 support. 

/etc/init.d/dnscrypt-proxy
```
config_get_bool block_ipv6      $1 'block_ipv6'     '0'                                               
```

damianorenfer, 2017-05-26:
Hi @simontretter, thanks for reporting this issue.

Do you have any error or warning in syslog ?

simontretter, 2017-05-26:
logread output has no mention of dnscrypt-proxy, it fails silently. manually invoking dnscrypt-proxy /var/etc/dnscrypt-proxy-whatever.conf throws the right error though.

damianorenfer, 2017-05-26:
Ok, maybe what we can is to check if IPv6 is supported in init script and ignore block_ipv6 parameter

simontretter, 2017-05-26:
Maybe checking dnsmasq ipv6 (or the absence of such) would make sense? 
`dnsmasq  --version | grep 'no-IPv6' | wc -l`

jow-, 2017-05-27:
No, IPv6 in dnsmasq can be disabled while IPv6 in general is still available.
I think you can check with `-e /proc/net/ipv6_route` for general IPv6 support on the system or directly probe the dnscrypt binary itself for support:

    if grep -sq block_ipv6 /usr/sbin/dnscrypt-proxy; then
        # seems to have IPv6 support
    fi

simontretter, 2017-05-28:
Checking the dnscrypt-proxy binary itself would be the best solution. I don't see how greping for block_ipv6 helps there though. The only switch that provides details on what the binary supports is --version. This is used in the init script to check for general plugin support, but gives no details on ipv6. 

LEDE/openwrt package compiled without plugin support:
```dnscrypt-proxy -V
dnscrypt-proxy 1.9.5

Compilation date: May 26 2017
Support for ldns-based plugins: present
Support for the XChaCha20-Poly1305 cipher: present
```

Arch Linux package with plugin support:
```dnscrypt-proxy -V
dnscrypt-proxy 1.9.4

Compilation date: Mar 22 2017
Support for plugins: present
Plugins root directory: [/usr/lib/dnscrypt-proxy/]
Support for ldns-based plugins: present
Support for systemd socket activation: present
Support for the XChaCha20-Poly1305 cipher: present
```

damianorenfer, 2017-05-28:
@simontretter did you try the versions 1.9.5-3 from PR #4393 there was an error in the init script regarding block_ipv6 param.

starcms, 2017-06-02:
This should be closed.  Issue was fixed in version 1.9.5-3.

closed 2017-06-02 by hnyman