openwrt/packages #4390: fwknop: stops working when interface goes down

badfish, 2017-05-26:
Maintainer: @jp-bennett 
Environment: arch:any, model:any, version:DD

Description:
I'm using fwknop to get remote access to my router. Whenever the wan connection breaks and is restored (due to problems at the ISP), fwknopd continues to run but no longer detects the incoming magic packet.
The fix which works for me is to add a file /etc/hotplug.d/iface/90-fwknop containing:

```
#!/bin/sh
[ "$ACTION" = ifup ] || exit 0
/etc/init.d/fwknopd enabled && /etc/init.d/fwknopd restart
```

InkblotAdmirer, 2017-05-27:
When the router boots (among other possibilities) there seem to be multiple interface state toggles due to dnsmasq starting, etc.  Some reload firewall, others don't but to prevent restarting unnecessarily I have added another filter:

> [ $(iptables -L | grep FWKNOP_INPUT | wc -l) == 0 ] &&

So the test could be:  if fwknopd is enabled && iptables rules are not there, then restart.  This seems to keep remote access while reducing log spam and unnecessary restarts.

jp-bennett, 2017-06-10:
I will get to this.  I have about 2 more weeks of work, then will have free time again.

jp-bennett, 2017-07-29:
I think I misunderstood what is going on here.  Is the issue that Fwknopd continues to run, and its firewall rules are removed?  If that's the case, I would think that fwknopd would see the SPA packet, and try to act on it, but that action would be useless without the iptable rules in place.

Fwknopd is actually intended to end its process when the interface it's listening to goes down.

badfish, 2017-07-31:
Ah, so maybe my fault report should have said "when the wan connection breaks and is restored, fwknopd stops running".
The actual problem that I had was that I could not get remote access to my machine, and when I returned home the log showed that the wan had gone down and then up again. I thought fwknopd was still running but not responding, but perhaps I was mistaken. Either way, restarting it fixed the problem. My wan has gone down/up since I installed my proposed fix, but I've still got remote access.

jp-bennett, 2017-10-16:
#4924 might fix this issue entirely.  If you guys get a chance to update, let me know if there are any problems with the new stuff.

guidosarducci, 2018-03-08:
Hi @jp-bennett,
I wanted to also note that I've seen this issue several times on long-running LEDE stable systems with PCAP_INTF as the home ISP WAN. These systems have stable power and uptimes of weeks to months, with the only network changes expected to be an IP renewal or (less likely) cable modem reboot. Invariably, the systems eventually stop responding to fwknop access packets until the local fwknopd instance can be restarted. 

Unfortunately this makes fwknopd a little unreliable for securing remote access to these unattended routers. Having seen many promising improvements and updates get merged into master, can I ask if you might backport some of the most helpful ones to LEDE stable? Many thanks.

jp-bennett, 2018-03-14:
@guidosarducci There will be a unified Openwrt/Lede release within 1-2 months which will include these fixes.  An issue like this isn't considered critical enough to backport.  Feel free to test on current Openwrt bleeding edge.  If you run into further problems there, please let me know.

guidosarducci, 2018-03-17:
@jp-bennett Thanks, I appreciate the reply.  I can also understand limiting the scope of consideration to critical issues, and there are certainly more important problems than this one.

In that vein, I traced my most serious loss-of-communication incident to a nasty bug during sysupgrade, where the correct fwknopd credentials were wiped due to commit https://github.com/openwrt/luci/commit/4b7a00858ed6ac33cf546cee1d8bce375f63fe69 being missing from the stable branch.

If you're open to backporting that critical fix I will take care of the PR and testing for you. Take care and thanks again.

oldium, 2018-04-10:
I improved the startup script, it should more reliable now - you could try the solution from pull request #5897. You just need to specify the `network` (defaults to `wan`) in `/etc/config/fwknopd` and omit completely the `PCAP_INTF` configuration like the following:

```
config global
    option uci_enabled '1'

config network
    option network 'wan' # your logical network name

config access
   ... your options ...

config config
   #option PCAP_INTF 'eth0' # comment-out or delete the physical interface
   ... your options ...
```

I tested this on PPPoE, which completely loses the interface when the connection is restarted. Tested on OpenWrt master, but there are chances that it will work for you.

neheb, 2020-01-12:
The linked PR has been merged. Closing.

closed 2020-01-12 by neheb