openwrt/packages #4391: tor: conffiles stored in tmpfs

badfish, 2017-05-26:
Maintainer: @hauke 
Environment: openwrt DD

Description:
Makefile for 'tor' specifies /var/lib/tor/fingerprint and /var/lib/tor/keys/* as conffiles. But /var is a symlink to /tmp; these configuration files get destroyed at every reboot. Correct operation of tor requires them to be preserved.
I've worked around this on my machine by moving the files to /etc/tor and adding a line to /etc/init.d/tor to copy them to /var/lib/tor whenever tor is started. This requires the user to manually create them in /etc/tor in the first place, so perhaps /etc/init.d/tor or the package installation script should display a warning if they do not exist.

EricLuehrsen, 2017-06-01:
The proposed process is used for net/unbound for jail and to protect flash. DNSSEC root keys are continuously tracked and this could harm flash. The start/stop scripts copy from `/etc/unbound` to `/var/lib/unbound` and back.

Jookia, 2017-07-22:
It's worth noting that this may be a security issue as well: Tor stores its long-term guard node in the configuration, intended to help prevent traffic attacks by not rotating it often. Having this reset every boot certainly ups the chance of an attacker getting a guard node.

neheb, 2020-02-07:
ping @ja-pa since this seems up your alley.