openwrt/packages #4394: openvswitch: build fails because "debian/ovs-monitor-ipsec': No such file or directory"

jmguzmanc, 2017-05-27:
OpenWRT build fails because some issue with openvswitch

I am using last code from master

```
install -m0755 /home/ubuntu/openwrt_kvm_last/build_dir/target-x86_64_musl-1.1.16/linux-x86_64/openvswitch-2.7.0/debian/ovs-monitor-ipsec /home/ubuntu/openwrt_kvm_last/build_dir/target-x86_64_musl-1.1.16/linux-x86_64/openvswitch-2.7.0/ipkg-x86_64/openvswitch-ipsec/usr/sbin/
install: cannot stat '/home/ubuntu/openwrt_kvm_last/build_dir/target-x86_64_musl-1.1.16/linux-x86_64/openvswitch-2.7.0/debian/ovs-monitor-ipsec': No such file or directory

```
I am not familiar with Makefiles.
Could any of you guys provide me some guidance?

Thanks a lot
JM

commodo, 2017-05-29:
my bad for this ; 
PR https://github.com/openwrt/packages/pull/4381  should fix this

diizzyy, 2017-06-02:
@jmguzmanc 
Please test, if it works close this issue report.

commodo, 2017-06-30:
it seems it wasn't completely fixed ;
i submitted a PR yesterday that seems to have fixed it for buildbot

i no longer see it in buildbot's faillogs
http://downloads.lede-project.org/snapshots/faillogs/mipsel_24kc/packages/

jmguzmanc, 2017-07-02:
It works like a charm!
All packages generated
`./bin/x86/packages/packages/kmod-openvswitch_4.4.14+2.7.0-1_x86_64.ipk
./bin/x86/packages/packages/openvswitch_2.7.0_x86_64.ipk
./bin/x86/packages/packages/openvswitch-base_2.7.0_x86_64.ipk
./bin/x86/packages/packages/openvswitch-ovs-appctl_2.7.0_x86_64.ipk
./bin/x86/packages/packages/openvswitch-ovsdb-client_2.7.0_x86_64.ipk
./bin/x86/packages/packages/openvswitch-ovs-dpctl_2.7.0_x86_64.ipk
./bin/x86/packages/packages/openvswitch-ovs-ofctl_2.7.0_x86_64.ipk
./bin/x86/packages/packages/openvswitch-ovs-vsctl_2.7.0_x86_64.ipk
./bin/x86/packages/packages/openvswitch-python_2.7.0_x86_64.ipk`

Thanks a lot, and my apologies for the delay.
JM

closed 2017-07-02 by jmguzmanc