openwrt/packages #4395: neon: davfs2 can't create a file

bpbiszkopt, 2017-05-27:
Maintainer: @fededim , @nbd168 , @champtar 
Environment: CC 15.05 (ar71xx)

Description:

Hi,
i notice that i can't create a file by using davfs2 on CC.
Folder creates without problem. I was trying diffrent version neon, davfs but without reason.
In logs there is someting like "neon error Premature EOF in request body file "

```
User-Agent: davfs2/1.5.4 neon/0.30.2
Connection: TE
TE: trailers
Host: dav.box.com
If-None-Match: *
Content-Length: 6424947544477018334
Authorization: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

Sun May 28 00:37:52 2017 mark.debug syslog: Sending request-line and headers:
Sun May 28 00:37:52 2017 mark.debug syslog: Sending request body:
Sun May 28 00:37:52 2017 mark.debug syslog: Request body provider failed with
Sun May 28 00:37:52 2017 mark.debug syslog: -
Sun May 28 00:37:52 2017 mark.debug syslog: 1
Sun May 28 00:37:52 2017 mark.debug syslog:
Sun May 28 00:37:52 2017 mark.debug syslog: sess: Closing connection.
Sun May 28 00:37:52 2017 mark.debug syslog: sess: Connection closed.
Sun May 28 00:37:52 2017 mark.debug syslog: Request ends, status
Sun May 28 00:37:52 2017 mark.debug syslog: 0
Sun May 28 00:37:52 2017 mark.debug syslog:  class
Sun May 28 00:37:52 2017 mark.debug syslog: 0
Sun May 28 00:37:52 2017 mark.debug syslog: xx, error line:
Sun May 28 00:37:52 2017 mark.debug syslog: Premature EOF in request body file
Sun May 28 00:37:52 2017 mark.debug syslog:
Sun May 28 00:37:52 2017 mark.debug syslog: Running destroy hooks.
Sun May 28 00:37:52 2017 mark.debug syslog: Request ends.
Sun May 28 00:37:52 2017 mark.debug syslog: close: neon error
Sun May 28 00:37:52 2017 mark.debug syslog:       Premature EOF in request body file
Sun May 28 00:37:52 2017 mark.debug syslog: RET: Input/output error
Sun May 28 00:37:52 2017 mark.debug syslog: tidy: 4 of 15 nodes changed
Sun May 28 00:37:52 2017 mark.debug syslog: cache-size: 0 MiBytes.
Sun May 28 00:37:52 2017 mark.debug syslog: tidy: /dav/rec/astral.doc
Sun May 28 00:37:52 2017 mark.debug syslog: Running pre_send hooks
Sun May 28 00:37:52 2017 mark.debug syslog: Sending request headers:
Sun May 28 00:37:52 2017 mark.debug syslog: PUT /dav/rec/astral.doc HTTP/1.1
User-Agent: davfs2/1.5.4 neon/0.30.2
Connection: TE
TE: trailers
Host: dav.box.com
If-None-Match: *
Content-Length: 2051
Authorization: xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

Sun May 28 00:37:52 2017 mark.debug syslog: Sending request-line and headers:
Sun May 28 00:37:52 2017 mark.debug syslog: req: Connecting to
Sun May 28 00:37:52 2017 mark.debug syslog: 107.152.27.197
Sun May 28 00:37:52 2017 mark.debug syslog: :
Sun May 28 00:37:52 2017 mark.debug syslog: 443
Sun May 28 00:37:52 2017 mark.debug syslog:
Sun May 28 00:37:53 2017 mark.debug syslog: Sending request body:
Sun May 28 00:37:53 2017 mark.debug syslog: Request body provider failed with
Sun May 28 00:37:53 2017 mark.debug syslog: -
Sun May 28 00:37:53 2017 mark.debug syslog: 1
Sun May 28 00:37:53 2017 mark.debug syslog:
Sun May 28 00:37:53 2017 mark.debug syslog: sess: Closing connection.
Sun May 28 00:37:53 2017 mark.debug syslog: sess: Connection closed.
Sun May 28 00:37:53 2017 mark.debug syslog: Request ends, status
Sun May 28 00:37:53 2017 mark.debug syslog: 0
Sun May 28 00:37:53 2017 mark.debug syslog:  class
Sun May 28 00:37:53 2017 mark.debug syslog: 0
Sun May 28 00:37:53 2017 mark.debug syslog: xx, error line:
Sun May 28 00:37:53 2017 mark.debug syslog: Premature EOF in request body file
Sun May 28 00:37:53 2017 mark.debug syslog:
Sun May 28 00:37:53 2017 mark.debug syslog: Running destroy hooks.
Sun May 28 00:37:53 2017 mark.debug syslog: Request ends.
Sun May 28 00:37:53 2017 mark.debug syslog: tidy: neon error
Sun May 28 00:37:53 2017 mark.debug syslog:       Premature EOF in request body file
Sun May 28 00:38:04 2017 mark.debug syslog: SELECT: 0
```

diizzyy, 2017-06-02:
Does it work if you use a recent release of LEDE (openwrt fork)?
https://lede-project.org/

bpbiszkopt, 2017-06-03:
I don't know... It works in AA. 
Maybe I will check it in next week, because I need this works.

diizzyy, 2017-06-03:
CC and AA are both obsolete and unsupported so there won't be a fix for these releases.

fededim, 2017-06-07:
**Content-Length: 642494754447701833**....642 petabytes ?!? What kind of file are you going to create ? Maybe the error is here, I have just released a new version of davfs2 which fixes an old configuration error regarding large file support, probably it could be there the error, wait for PR #4458 to be merged and the new version of neon and davfs2 to be compiled. Retry and if it does not work let me know.

bpbiszkopt, 2017-06-08:
It was empty file...i dont know why it show so much value.
2 days ago i installed lede to check if this problem will still occurs ...but it works fine, without any problem, so im not coming back to CC.
But Thank You for answer and upgrading davfs2.

fededim, 2017-06-08:
I did the upgrade for another issue, but I was thinking that also this one was due to the forced Large File Support (since you wrote that you could create a directory and not a file it was something related to a file). Any way if you tell me that it works on Lede, the problem was another one. Please remember to close the issue.

closed 2017-06-08 by bpbiszkopt