openwrt/packages #4399: mpd: Pulls a lot of redundant libs?

diizzyy, 2017-05-28:
@thess 

I might be wrong here but as far as I can tell we're pulling a lot of redundant libs as from what I can tell it'll use ffmpeg for all supported formats? That might be even more beneficial even for the mini-package.

http://wiki.hydrogenaud.io/index.php?title=Music_Player_Daemon

That would also potentially mean that we could drop packages such as libaudiofile.

thess, 2017-05-28:
Could you be more specific -- and have you tested it with enough format variations?

I put quite a bit of work testing the 'mini' config a number of years ago and back then it was cut to the bone for the desired codecs and features. I'm sure things have aged. OK, I'm always up for a new config but don't have the time/desire to change anything at-the-moment. At least as of a couple of years ago, I could still fit mpd-full + ffmpeg-audio-dec w/ some pretty odd codecs into an 8/32 system. We may have lost that along the way - haven't tried lately.

diizzyy, 2017-06-01:
I was more thinking of the full version, +libaudiofile  +BUILD_PATENTED:libfaad2 +libogg +libsndfile +libvorbis should all be unnecessary if you link ffmpeg no?

thess, 2017-07-17:
If all you have is a hammer - everything looks like a nail :wink:

thess, 2017-08-27:
@diizzyy - I think we can close this for now, OK?

diizzyy, 2017-08-27:
Yes, I'll close it.

closed 2017-08-27 by diizzyy