openwrt/packages #4408: Upgrade Request: minidnla (ReadyMedia) to 1.2.0

ZacWolf, 2017-05-29:
Would it be possible to request an upgrade of package minidnla to latest ReadyMedia 1.2.0?

Just released a few days ago, but since it's patched so rarely...

1.2.0 - Released 17-May-2017
--------------------------------
- Add libavformat > 57 compatibility.
- Add TiVo Bonjour discovery support using Avahi for Bolt.
- Improve Samsung BD-J5500 support.
- Add quirk to fix video playback on Hyundai TVs.
- Add non-destructive update rescan support.
- Enhance bookmark support to work with Kodi.

diizzyy, 2017-06-01:
@ZacWolf  #4421

diizzyy, 2017-06-02:
Please test the pull request and close this issue unless you find any problems with the new version.

ZacWolf, 2017-06-03:
I install via Entware-NG so I've opened a case for this to sync with you upstream feed, so that it builds for my router. I'll close once I'm able to install via opkg.  THANKS!!!

diizzyy, 2017-06-24:
@ZacWolf 
Updated, please close this issue report.

diizzyy, 2017-07-13:
@hnyman @thess 
Fixed, please close

closed 2017-07-13 by champtar