openwrt/packages #4409: tvheadend - Update to 4.2

lylejenson, 2017-05-29:
Any chance of updating tvheadend to the latest version 4.2?

hnyman, 2017-05-30:
> Any chance of updating tvheadend to the latest version 4.2?

Any chance of you targeting your message to the maintainer of that package? Maintainers do not get automatically all messages...

cc to tvheadend maintainer @sairon

sairon, 2017-05-30:
Thanks for notifying, I did not notice the new version after the long period without any public releases. I'll take a look at it soon and update it if everything goes well.

roni261980, 2017-08-27:
anyone can update tvheadend to last stable release v4.2.3.33

diogosena, 2017-09-18:
@sairon any news on this?

9000000, 2018-04-03:
Why not updated 😥😥😥😥😥

kofec, 2018-04-03:
Hi if you have a while you can compile for yourself:
https://github.com/CZ-NIC/turris-os-packages/blob/test/multimedia/tvheadend/Makefile
I found this repo today

9000000, 2018-04-08:
team lede where are you ? 
,...............................................................!!!¡!!!!!!!!!!!!!!!!!!!!!!!!!!!¡

hnyman, 2018-04-08:
> team lede where are you ?

If @sairon is not taking care of the maintainer role and version upgrades, then somebody else who is interested, could take over the maintainership and author the version upgrade Pull Request.

This is a community effort. So it might just as well be you, @9000000

9000000, 2018-04-19:
Share tvheadend 4.1 #5942

sjjgo, 2018-07-01:
Please look at the PR I have created to update this package: https://github.com/openwrt/packages/pull/6386

sjjgo, 2018-07-06:
@hnyman I'm not sure if @sairon is maintaining the package any more. I am happy to take this responsibility on if they no longer wish to be involved in looking after this package.

benteam10, 2018-07-10:
Here is Tvheadend 4.2.6 for LEDE 17.1.4     for architecture  AR7xxx 
i run it on a Archer C7 V2  MIPS 24Kc   
https://drive.google.com/open?id=1k1Bs-GCsIyG68wdNGnPL6B-ybdjyRda5

runs very well

i couldnt compile it for chaos calmer since the built system is not complete it doesnt have the packages anymore. 
if it starts and ask for a username and password  edit the /etc/init.d/tvheadend    in the line that shows the ARGS  -B   ad  -C  save and start it, is not going to ask you for credentials anymore.

hnyman, 2018-07-10:
closing this, as  PR #6386 contains the concrete proposal for the version bump.

closed 2018-07-10 by hnyman