openwrt/packages #4428: mwan3: interface with metric > 256 is not active in policy

kubco2, 2017-06-02:
Maintainer: @Adze1502 
Environment: ar71xx, TP-Link TL-WR1043, OpenWRT 15.05

Description:
If I have only one active online interface with metric > 256, Interface is initialized except policy rules and therefore it uses fallback as policy. If I set lower value, it properly setup policy rules.

It should work for any metric value allowed by linux, or at least print error message. In LuCI range 1-1000 is allowed, so it should be adjusted to unlimited or 256 according to mwan3 fix.

feckert, 2017-08-02:
@kubco2 adze1502 is no longer the maintainer of mwan3. I have agreed to do this work now. I will have a look what the problem is

feckert, 2017-08-11:
@kubco2 is your version openwrt 15.05? Could you test lede-17.01 because there were a lot of changes in the mwan3.

feckert, 2017-11-03:
@kubco2 ping?
could we close this issue?

kubco2, 2017-11-03:
Will test this weekend.

S pozdravom,
Jakub Janco

kubco2, 2017-11-28:
it has same problems. I created pull request for mwan3 part.

feckert, 2017-12-18:
@kubco2 I think we could close this issue?

closed 2017-12-18 by kubco2

feckert, 2017-12-18:
@kubco2 thanks for your changes on mwan3 :-)