openwrt/packages #4431: quagga-ospf6d: doesn't get routes from other routers

suedwestlicht, 2017-06-04:
Maintainer: @<github-user> (find it by checking history of the package Makefile)
I am sorry, I can't find quagga in "net".

Environment:
LEDE Reboot 17.01.1 r3316-7eb58cf109 / LuCI lede-17.01 branch (git-17.100.70571-29fabe2)
TP-Link TL-WR1043N/ND v4

Description:
ospfd of quagga works ok.

ospf6d of quagga recognizes its neighbours but doesn't get routes.
Package: quagga-ospf6d
Version: 1.1.0-1

It seems we have a bad quagga version.
Please see  https://lists.gt.net/quagga/users/32153

Would it be possible to package an older version of quagga?

diizzyy, 2017-06-04:
Please close this, wrong repository.
Correct: https://github.com/openwrt-routing/packages
fwiw, I'd say that acinonyx is the maintainer for quagga ;-)

suedwestlicht, 2017-06-04:
Thanks. I opened a ticket in openwrt-routing/packages

closed 2017-06-04 by suedwestlicht