openwrt/packages #4432: mosquitto-client-nossl is a empty package

andrewc12, 2017-06-05:
-------------------------------

Maintainer: @karlp 
Environment: all, all, 17.01.1 and snapshot

Description:
mosquitto-client-nossl is literally an empty package.

andrewc12, 2017-06-05:
From irc
< swalker> the install define was renamed in
https://github.com/openwrt/packages/commit/027e3ec4ec4f35dc5d1bdf530ae9c913db5f1d08#diff-0ac3ba1e1320dc6cbcdf528ccb6e87b8R182 but
https://github.com/openwrt/packages/blob/master/net/mosquitto/Makefile#L187 wasn't updated so 2 months rather than 3 years

andrewc12, 2017-06-05:
Changing
Package/mosquitto-client-nossl/install = $(Package/mosquitto-client/install)
to 
Package/mosquitto-client-nossl/install = $(Package/mosquitto-client-ssl/install)
on https://github.com/openwrt/packages/blob/master/net/mosquitto/Makefile#L187 fixes it.
I have compiled it myself.

karlp, 2017-06-05:
blah, PROVIDES limitations fallout.  Will fix it tomorrow.

closed 2017-06-07 by karlp

karlp, 2017-06-07:
(also fixed in lede-17.01 branch)