openwrt/packages #4435: shadowsocks-libev 3.0.6-1 don't have ss-local nor ss-server?

steve3d, 2017-06-05:
I know that entware-ng is mostly based on this, after shadowsocks-libev upgraded from 2.5.1 to 3.0.6, I found there is no `ss-local` program but `/opt/etc/init.d/S22shadowsocks` still want to start the `ss-local`.

So I'm sure this is a big problem, without the `ss-local` program in shadowsocks-libev, the whole package are rendered as totally unusable.

Please see the following output by `opkg files`

```
admin@RT-AC68U-10E8:/jffs/bin# opkg files shadowsocks-libev
Package shadowsocks-libev (3.0.6-1) is installed on root and has the following files:
/opt/etc/shadowsocks.json
/opt/bin/ss-tunnel
/opt/etc/init.d/S22shadowsocks
/opt/bin/ss-rules
/opt/bin/ss-redir
admin@RT-AC68U-10E8:/jffs/bin# 
```

See? there is no `ss-local` nor `ss-server`, althought that `ss-server` is not important, but `ss-local` is the most important program in this package. This is like the gcc packages without the `gcc` program.

Hope it gets fixed soon.

diizzyy, 2017-06-07:
@steve3d 
https://github.com/openwrt/packages/blob/master/net/shadowsocks-libev/Makefile#L62
Can you provide a pull request?

steve3d, 2017-06-07:
sure, will do it later today.

hkjuniordev, 2017-06-09:
Please fix this as soon as possible, shadowsocks without ss-local and ss-server is completely useless! Thanks!

zeroxia, 2017-06-17:
It looks like the shadowsocks-libev package maintained by openwrt just does not have ss-local or ss-server (or at least this is not the first time it misses these two executables), why does it elect to do so? Just to save disk space?

steve3d, 2017-06-18:
I've already provided a PR in #4455 , but the maintainer of this packages seems having trouble to decide what to do.

yousong, 2017-06-26:
Hi, @steve3d , I am rewriting packaging method of shadowsocks-libev and opened a pr https://github.com/openwrt/packages/pull/4513 for this.  `ss-local` has been added there but not `ss-server`.  If you still need it, please give further comments there.

shounak-de, 2017-06-26:
@yousong `ss-server` is the only binary that I use. Please include it.

yousong, 2017-06-27:
hi, @shounak-de , now uci support for ss-server is added in the pull request.  Please give it a try and see if it can fulfil your needs.

I am curious about the setup.  Is the ss-server instance running on your home router and intended to provide a secure channel to connect back to your home network?

shounak-de, 2017-06-27:
I'd need to wait till [entware-ng] (https://github.com/Entware-ng/Entware-ng) picks up your changes. It is a downstream repo that I use for my router.

My setup is quite simple. I use shadowsocks as my VPN tunnel to my home network. Server on router, clients on phone and other PCs. There are some clear advantages over a similar openvpn setup, and that is why I chose shadowsocks.

yousong, 2017-07-02:
I am closing this now as pr #4513 was just merged.  Thanks for the feedback.

closed 2017-07-02 by yousong

la88962, 2017-07-17:
Please include ss-server