openwrt/packages #4453: busybox: Customize vi compile fail

hynnet, 2017-06-07:
Please make sure that the issue subject starts with `<package-name>: ` so that it's easily identifiable.

This repo here is ONLY for packages maintained in this repo.  For base packages residing in the same repo as the build system and maintained by core devs, please consider opening tickets there for more timely responses

 - OpenWrt: https://dev.openwrt.org/newticket
 - LEDE: https://bugs.lede-project.org/
 - Most LuCI packages: https://github.com/openwrt/luci/issues

Thanks for your contribution
Please remove this text (before ---) and fill the following template
-------------------------------

Maintainer: @<github-user> (find it by checking history of the package Makefile)
Environment: (put here arch, model, OpenWRT/LEDE version)

Description:
     Customize busybox options  Editors ->  vi  compile fail!
make menuconfig
```
busybox
[*]   Customize busybox options
Editors 
 [*] vi                                                                                                    x x
         (1024) Maximum screen width in vi                                                                         x x
         [*]   Allow vi to display 8-bit chars (otherwise shows dots)                                              x x
         [*]   Enable ":" colon commands (no "ex" mode)                                                            x x
         [*]   Enable yank/put commands and mark cmds                                                              x x
         [*]   Enable search and replace cmds                                                                      x x
         [*]     Enable regex in search and replace                                                                x x
         [*]   Catch signals                                                                                       x x
         [*]   Remember previous cmd and "." cmd                                                                   x x
         [*]   Enable -R option and "view" mode                                                                    x x
         [*]   Enable set-able options, ai ic showmatch                                                            x x
         [*]   Support for :set                                                                                    x x
         [*]   Handle window resize                                                                                x x
         [*]   Use 'tell me cursor position' ESC sequence to measure window                                        x x
         [*]   Support undo command 'u'                                                                            x x
         [*]     Enable undo operation queuing                                                                     x x
         (32)      Maximum undo character queue size                                                               x x
         [*]   Allow vi and awk to execute shell commands    

```

make -j1 V=99
```
make[3]: Entering directory '/home/hyn/lede/source-mt7620/package/utils/busybox'
rm -f /home/hyn/lede/source-mt7620/build_dir/target-mipsel_24kc_musl/busybox-1.26.2/.built
touch /home/hyn/lede/source-mt7620/build_dir/target-mipsel_24kc_musl/busybox-1.26.2/.built_check
make  -C /home/hyn/lede/source-mt7620/build_dir/target-mipsel_24kc_musl/busybox-1.26.2 CC="mipsel-openwrt-linux-musl-gcc" CROSS_COMPILE="mipsel-openwrt-linux-musl-" KBUILD_HAVE_NLS=no EXTRA_CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -msoft-float -mips16 -minterlink-mips16 -iremap/home/hyn/lede/source-mt7620/build_dir/target-mipsel_24kc_musl/busybox-1.26.2:busybox-1.26.2 -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" ARCH="mipsel" SKIP_STRIP=y LDLIBS="m crypt"  all
make[4]: Entering directory '/home/hyn/lede/source-mt7620/build_dir/target-mipsel_24kc_musl/busybox-1.26.2'
  CC      editors/vi.o
editors/vi.c: In function 'char_search':
editors/vi.c:1902:2: error: 're_syntax_options' undeclared (first use in this function)
  re_syntax_options = RE_SYNTAX_POSIX_EXTENDED;
  ^
editors/vi.c:1902:2: note: each undeclared identifier is reported only once for each function it appears in
editors/vi.c:1902:22: error: 'RE_SYNTAX_POSIX_EXTENDED' undeclared (first use in this function)
  re_syntax_options = RE_SYNTAX_POSIX_EXTENDED;
                      ^
editors/vi.c:1904:50: error: 'RE_ICASE' undeclared (first use in this function)
   re_syntax_options = RE_SYNTAX_POSIX_EXTENDED | RE_ICASE;
                                                  ^
editors/vi.c:1907:8: warning: implicit declaration of function 're_compile_pattern' [-Wimplicit-function-declaration]
  err = re_compile_pattern(pat, strlen(pat), &preg);
        ^
editors/vi.c:1907:6: warning: assignment makes pointer from integer without a cast [-Wint-conversion]
  err = re_compile_pattern(pat, strlen(pat), &preg);
      ^
editors/vi.c:1935:6: warning: implicit declaration of function 're_search' [-Wimplicit-function-declaration]
  i = re_search(&preg, q, size, /*start:*/ 0, range, /*struct re_registers*:*/ NULL);
      ^
scripts/Makefile.build:197: recipe for target 'editors/vi.o' failed
make[5]: *** [editors/vi.o] Error 1
Makefile:742: recipe for target 'editors' failed
make[4]: *** [editors] Error 2
make[4]: Leaving directory '/home/hyn/lede/source-mt7620/build_dir/target-mipsel_24kc_musl/busybox-1.26.2'

```

hnyman, 2017-06-07:
Busybox is in the main Openwrt/LEDE repos. You should open a report in the respective bug tracker(s), not here.

But in general, there is no guarantee that all busybox options are compatible with the build system. If you customize busybox options, you are pretty much on your own. Just try to fix the problem and then submit a patch.

closed 2017-06-07 by hnyman

aenertia, 2017-09-10:
This is something to do with missing regex libs inside the lede build tree find also fails to build if you enable regex matches

risa2000, 2019-05-07:
The bug is still present in the recent master  (bc85640cdc9de180da00fde1532584abf19db365), but only for `vi` regex search option. `find` option with the regex builds ok.

satmandu, 2019-11-15:
Also mentioned here: https://forum.openwrt.org/t/busybox-not-compiling/

satmandu, 2019-11-15:
Also mentioned here: https://dev.archive.openwrt.org/ticket/21741.html