openwrt/packages #4460: minidlna: segfault when parsing nfo

luizluca, 2017-06-08:
Maintainer: @medaved 
Environment: LEDE 17.01.2 on ar71xx (mips)

Description:
When parsing this NFO [gotham.32122.hdtv-lol.txt](https://github.com/openwrt/packages/files/1060069/gotham.32122.hdtv-lol.txt), minidlna segfaults at:

```
[94469.503912] do_page_fault(): sending SIGSEGV to minidlna for invalid write access to 7760f5c4
[94469.512632] epc = 0040ea21 in minidlna[400000+29000]
[94469.517676] ra  = 00410821 in minidlna[400000+29000]
```

Or, from gdb:

```
Thread 2 "minidlna" received signal SIGSEGV, Segmentation fault.
parse_nfo (path=path@entry=0x772eb96c "/mnt/usb-dados/torrent/Gotham.S03E21E22.HDTV.x264-LOL[rarbg]/gotham.32122.hdtv-lol.nfo", m=m@entry=0x772eb77c) at metadata.c:161
```

It seems to be a stack overflow. v1.2.0 has a fix for this https://sourceforge.net/p/minidlna/git/ci/46c692fc591c5d927f68783a2b6b687114421f3b/tree/metadata.c?diff=4f926639b24f8bbf1fb5b88f6e9ebc8cdb3649e8

luizluca, 2017-06-08:
Also, PR #4421 would fix this in trunk and, if backported, 17.01.x

diizzyy, 2017-06-15:
Pull req #4420 merged, not sure if we're going to see another 17.01 release.

luizluca, 2017-06-15:
And you mean #4421 and not #4420 ;-)

closed 2017-06-15 by luizluca