openwrt/packages #4472: Nmap: missing scripting support

hoxia, 2017-06-11:
Maintainer: @nunojpg 
Environment: all, trunk

Description:
Nmap was compiled without scripting support, with a disabled liblua dependency.

This can be observed by attempting to run one of these commands:
```
# nmap -sC
Nmap 7.40 ( https://nmap.org )
Usage: nmap [Scan Type(s)] [Options] {target specification}
[...]
SEE THE MAN PAGE (https://nmap.org/book/man.html) FOR MORE OPTIONS AND EXAMPLES
Scantype C not supported

# nmap --script
nmap: unrecognized option '--script'
See the output of nmap -h for a summary of options.
```

Also, --help output is missing the "SCRIPT SCAN" section present on nmap built with scripting support.

The [makefile](https://github.com/openwrt/packages/blob/master/net/nmap/Makefile) explicitly disables scripting, stating:
> liblua is still not recognized/blindly accepted. Impossible to compile with support

Would it be possible to build the current package, or an additional package, with scripting support? The reference to the liblua issue hasn't changed in the makefile in at least a couple years and a major revision, perhaps it could be revisited?

Originally reported to [Entware-ng](https://github.com/Entware-ng/Entware-ng/issues/678).

nunojpg, 2017-06-12:
@hoxia Yes, that limitation exists. I would be glad to add liblua support, please just provide a PR.

Closing, discussion will continue on PR if it is opened.

anonimou0, 2018-07-09:
@nunojpg 
I've interacted with developers and they have a patch to have nmap with scripting engine working on OpenWrt.
Here is the discussion.
http://seclists.org/nmap-dev/2018/q2/6
After adding that patch, I could compile nmap with NSE. Hope that helps.

nunojpg, 2018-07-09:
@anonimou0 yes, that is great. I will reopen this ticket and test it.

cotequeiroz, 2021-03-18:
This seems to have been fixed by 067c93b028b123184737f09751f73e46a5d44824.  I've checked it with:
```
# nmap --script
nmap: option requires an argument: script
See the output of nmap -h for a summary of options.
```

closed 2021-03-18 by cotequeiroz