openwrt/packages #4480: unbound: server access-control allows wan subnet

poranje, 2017-06-14:
Maintainer: @EricLuehrsen 
Environment: LEDE-17.01 

Description:
/etc/config/unbound has the option localservice set to '1'.
The generated /var/lib/unbound/unbound.conf contains an server:access-control stance that also allows the WAN its subnet. This seems not right.

EricLuehrsen, 2017-06-14:
This is common with dnsmasq (binary) behavior for the same option. A client must be on the defined subnet attached to the machine. This done for each interface bound to (wildcard default). Firewall should prevent access, and even if not, a client would need to be inside your ISP "neighborhood." Furthermore WAN doesnt always mean global. It may simply be upstream connection in a routed but site-local-secure network (multiple subnet). Upstream DNS may query down-route as `forward:` zone clauses. Because the goal is dnsmasq/unbound transparency, the feature should remain as is.

poranje, 2017-06-14:
Okay, thanks for the explanation.
Paul

closed 2017-06-14 by poranje