openwrt/packages #4487: unbound: syntax error when UCI dnsmasq[x].port not set

poranje, 2017-06-17:
Maintainer: @EricLuehrsen
Environment: OpenWRT/LEDE SNAPSHOT r4410-8f4085e2fd

Description:
When running with unbound.dhcp_link set to dnsmasq,
  then for each dnsmasq instance x that has dnsmasq[x], 
    dnsmasq_local_zone() throws message "sh: out of range".

The offending statement is:
`if [ -n "$fwd_domain" -a -n "$fwd_port" -a ! "$fwd_port" -eq 53 ] `
It seems the guard does not work within one single test ([ ... ]) statemanet.

When the test is divided into two tests:
`if [ -n "$fwd_domain" -a -n "$fwd_port" ] && [ $fwd_port -ne 53 ] `
the problem vanishes.

mhei, 2017-06-17:
To prevent splitting into two tests, you could also use something like this:
`if [ -n "$fwd_domain" -a -n "$fwd_port" -a ! "${fwd_port:-53}" -eq 53 ]`

poranje, 2017-06-17:
BTW, wouldn't a test like either [ $fwd_port -ne 53 ] or [ $fwd_port" != "53" ]  not be "better" ?
It is either an algebraic test -ne or a string compare; for testing UCI values a string compare looks simplest. (apart from using a default variable initialiser).

EricLuehrsen, 2017-06-17:
Port is a propper number and its cleanest to be in one test. I like: `if [ -n "$fwd_domain" -a -n "$fwd_port" -a ! "${fwd_port:-53}" -eq 53 ]`

pprindeville, 2017-06-17:
I don't understand this part:

`-n "$fwd_port" -a "${fwd_port:-53}" -ne 53`

specifically the `${fwd_port:-53}`...  If we've just tested for `-n "$fwd_port"` then we _know_ it's non-mpty and hence the `${fwd_port:-53}` is redundant...  We just need `$fwd_port` as it's non-empty.

EricLuehrsen, 2017-06-17:
> we've just tested for -n "$fwd_port" then we know it's non-mpty

If it where like an optimizing compiler, yes. It should evaluate from left-to-right and avoid unnecessary operations (or in this case guarded). Apparently the script interpreter isn't thinking that hard. So something to stub out the text-substitution phase is a good idea.

poranje, 2017-06-18:
On the other hand, though port is supposed to be a proper number, as the value tested is user supplied, it could possibly be some non-number string. When testing user input treating it as just a string and only after validation as being of some type may be a better practice.

Not really relevant, but ...

EricLuehrsen, 2017-06-19:
If we just change `$fwd_port` to `${fwd_port:-53}` then that is the smallest change. We just want to walk past a acceptably empty UCI entry (for another app's UCI). Some of the core OpenWrt/LEDE folks have recommended against fool proofing. Its and endless circle and eventually affects start time. LuCI does field checking for web access by the common user.

EricLuehrsen, 2017-06-23:
Can be closed. Fix implemented.

closed 2017-06-23 by hnyman