openwrt/packages #4488: MWAN3 routable loopback not working

cjjdaq, 2017-06-17:
The routable loopback (self) function not working.
after adding, some websites cannot be opened. For example: https://www.cnblogs.com

root@LEDE-X64:~# curl -v https://www.cnblogs.com
* Rebuilt URL to: https://www.cnblogs.com/
* Trying 101.37.225.58...
* TCP_NODELAY set
* Connected to www.cnblogs.com (101.37.225.58) port 443 (#0)
* ALPN, offering http/1.1
* Cipher selection: ALL:!EXPORT:!EXPORT40:!EXPORT56:!aNULL:!LOW:!RC4:@STRENGTH
* successfully set certificate verify locations:
* CAfile: none
CApath: /etc/ssl/certs
* TLSv1.2 (OUT), TLS header, Certificate Status (22):
* TLSv1.2 (OUT), TLS handshake, Client hello (1):

No more information....

config interface 'self'
option ifname 'lo'
option proto 'static'
option ipaddr '192.168.2.1'
option netmask '255.255.255.255'
option gateway '192.168.2.1'

config rule 'https'
option enable '1'
option dest_ip '0.0.0.0/0'
option dest_port '443'
option proto 'tcp'
option sticky '1'
option use_policy 'balanced'

I used a ad filter program named koolproxy on lede。
iptables -t nat -I PREROUTING -p tcp -m multiport --dport 80,443 -j REDIRECT --to 3500
3500 is the port of the koolproxy。
After transparent proxy，multi Wan, koolproxy will only be transmitted from the default route。I need koolproxy traffic load balancing。
In China, most people who use openwrt and Lede use ad filtering programs called koolproxy or adbyby. They are all transparent agents.
I can't speak English. These words are translated by translation software. I'm sorry。

feckert, 2017-08-02:
@cjjdaq could you please dump your iptables configuration with `iptables-save > /tmp/iptables.txt` and paste it to `https://pastebin.com/` so i could check your firewall configuration. I think it has to do with the iptables MARK target.

feckert, 2017-08-14:
May could help to read this issue #4685

feckert, 2017-11-03:
@cjjdaq ping?
could we close this issue?
Any further research for this issue

feckert, 2018-11-21:
@hnyman Since newest mwan3 this is not needed any more and this could be also closed