openwrt/packages #4491: btrfs-progs: needs a new maintainer

dangowrt, 2017-06-18:
I haven't been using btrfs for a very long time and hence removed myself from being the maintainer of btrfs-progs. btrfs is being used by the [Turis Omnia](https://omnia.turris.cz/en/) hence maybe @miska @shenek @Cynerd are good candidates to take over maintainership of btrfs-progs.

diizzyy, 2017-06-23:
I think we should decide on a placeholder for that entry.
Needs maintainer <no-maintainer@packages.openwrt.org> or whatever

mhei, 2017-11-25:
Maintainership was taken over by @neheb in #5154, so we can close this.

dangowrt, 2019-03-04:
https://github.com/openwrt/packages/pull/8339

BKPepe, 2019-03-05:
I haven't noticed this issue until now. 

As you said Turris Omnia indeed uses BTRFS and it will be good if we (speaking as a member of Turris Team) can take maintainer of this package. Thank you guys for suggestions!

We spoke together and I'm not sure if I currently understand issues related to BTRFS and I think it will be really hard for me to fix some issues, which may appear about it. That's why I have suggested @Cynerd to take package maintainership of this package and most probably he will prepare a pull request. :-)

dangowrt, 2019-03-11:
@BKPepe Thank you for getting involved.
Note that the bug was closed for quite a while @neheb was doing the job and only re-opened now as he dropped maintainership.
@Cynerd Please make a PR adding yourself (full first and last name) as the `PKG_MAINTAINER:=` of the package.

PR #8396 (2019-03-13): btrfs-progs: update to 4.20.2
Maintainer: me
Compile tested: Turris
Run tested: Turris {mox,omnia}

Description:
Update to latest fixup for btrfs-progs. I am also retaking maintenance of this package.

Closes #4491
closed 2019-03-13 by hnyman