openwrt/packages #4492: Problem with dependencies of some packages with build mechanism (probably)

build000, 2017-06-19:
As in the topic - I do not know to the end whether it has to do with the LEDE core or go deeper - so I'm posting here.

```
dom@machina:~/Pulpit/Projekty/WITI-LEDE-REBOOT-small$ git pull
remote: Counting objects: 7, done.
remote: Compressing objects: 100% (7/7), done.
remote: Total 7 (delta 4), reused 0 (delta 0)
Unpacking objects: 100% (7/7), done.
From https://git.lede-project.org/source
   80d9ec5..c885482  master     -> origin/master
Updating 80d9ec5..c885482
Fast-forward
 package/network/config/netifd/Makefile | 6 +++---
 1 file changed, 3 insertions(+), 3 deletions(-)
dom@machina:~/Pulpit/Projekty/WITI-LEDE-REBOOT-small$ make package/symlinks V=s
./scripts/feeds update -a
Updating feed 'packages' from 'https://git.lede-project.org/feed/packages.git' ...
remote: Counting objects: 83, done.
remote: Compressing objects: 100% (78/78), done.
remote: Total 83 (delta 47), reused 0 (delta 0)
Unpacking objects: 100% (83/83), done.
From https://git.lede-project.org/feed/packages
   3de228f..cd79492  master     -> origin/master
Updating 3de228f..cd79492
Fast-forward
 libs/glpk/Makefile                                             |  5 +++--
 libs/libevdev/Makefile                                         |  7 ++-----
 libs/libinput/Makefile                                         |  9 +++------
 libs/libinput/patches/001-no-locale.patch                      | 40 ++++++++++++++++++++++++++++------------
 libs/libinput/patches/002-static_assert.patch                  |  2 +-
 libs/libowfat/Makefile                                         | 30 ++++++++++++++++++++++--------
 libs/libowfat/patches/001-fixbuild.patch                       | 10 +++++++++-
 libs/libp11/Makefile                                           |  6 +++---
 libs/libp11/patches/001-fix-install.patch                      | 51 ++++++++++++++++++---------------------------------
 libs/libudev-fbsd/Makefile                                     | 50 ++++++++++++++++++++++++++++++++++++++++++++++++++
 libs/libudev-fbsd/patches/001-fix-unused.patch                 | 42 ++++++++++++++++++++++++++++++++++++++++++
 libs/libudev-fbsd/patches/002-replace-nitems.patch             | 11 +++++++++++
 libs/libudev-fbsd/patches/003-add-pkg-config-description.patch | 20 ++++++++++++++++++++
 libs/postgresql/Makefile                                       |  6 +++---
 net/cgi-io/Makefile                                            |  2 +-
 net/cgi-io/src/CMakeLists.txt                                  |  3 +++
 net/fping/Makefile                                             |  4 +++-
 net/gnurl/Makefile                                             |  4 ++--
 net/gnurl/patches/200-no_docs_tests.patch                      | 22 ----------------------
 net/opentracker/Makefile                                       | 10 ++++------
 utils/mbtools/Makefile                                         |  3 ++-
 utils/openzwave/Makefile                                       |  3 ++-
 utils/pcsc-lite/Makefile                                       |  6 +++---
 23 files changed, 235 insertions(+), 111 deletions(-)
 create mode 100644 libs/libudev-fbsd/Makefile
 create mode 100644 libs/libudev-fbsd/patches/001-fix-unused.patch
 create mode 100644 libs/libudev-fbsd/patches/002-replace-nitems.patch
 create mode 100644 libs/libudev-fbsd/patches/003-add-pkg-config-description.patch
 delete mode 100644 net/gnurl/patches/200-no_docs_tests.patch
Create index file './feeds/packages.index' 
Collecting package info: done
Updating feed 'luci' from 'https://git.lede-project.org/project/luci.git' ...
Already up-to-date.
Create index file './feeds/luci.index' 
Updating feed 'routing' from 'https://git.lede-project.org/feed/routing.git' ...
Already up-to-date.
Create index file './feeds/routing.index' 
Updating feed 'telephony' from 'https://git.lede-project.org/feed/telephony.git' ...
Already up-to-date.
Create index file './feeds/telephony.index' 
./scripts/feeds install -a
Collecting package info: done
Installing all packages from feed packages.
Installing package 'libudev-fbsd' from packages
Installing all packages from feed luci.
Installing all packages from feed routing.
Installing all packages from feed telephony.
dom@machina:~/Pulpit/Projekty/WITI-LEDE-REBOOT-small$ make menuconfig
Collecting package info: done
tmp/.config-package.in:94212:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:94212:	symbol PACKAGE_uvcdynctrl depends on PACKAGE_UVCDYNCTRL_UDEV_RULE
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:94231:	symbol PACKAGE_UVCDYNCTRL_UDEV_RULE depends on PACKAGE_uvcdynctrl
configuration written to .config

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

dom@machina:~/Pulpit/Projekty/WITI-LEDE-REBOOT-small$ scripts/getver.sh
r4427-c885482
```

kdarbyshirebryant, 2017-06-19:
Me too.  after a scripts/feeds install -a which installed the new ''libudev-fbsd'' package I get the following from make menuconfig:

make menuconfig
tmp/.config-package.in:94319:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:94319:	symbol PACKAGE_uvcdynctrl depends on PACKAGE_UVCDYNCTRL_UDEV_RULE
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:94338:	symbol PACKAGE_UVCDYNCTRL_UDEV_RULE depends on PACKAGE_uvcdynctrl

*** End of the configuration.
*** Execute 'make' to start the build or try 'make help'.

Removing the new package resolves the error.
./scripts/feeds uninstall libudev-fbsd

hnyman, 2017-06-19:
It is pretty evident that it is caused by the new package libudev-fbsd by @dangowrt 
Installing the new package to feeds triggers the error at the next "make defconfig":
```
$ make defconfig
#
# configuration written to .config
#

$ ./scripts/feeds install -a
Installing all packages from feed packages.
Installing package 'libudev-fbsd' from packages
Installing all packages from feed luci.
Installing all packages from feed routing.
Installing all packages from feed telephony.

$ make defconfig
Collecting package info: done
tmp/.config-package.in:94135:error: recursive dependency detected!
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:94135:	symbol PACKAGE_uvcdynctrl depends on PACKAGE_UVCDYNCTRL_UDEV_RULE
For a resolution refer to Documentation/kbuild/kconfig-language.txt
subsection "Kconfig recursive dependency limitations"
tmp/.config-package.in:94154:	symbol PACKAGE_UVCDYNCTRL_UDEV_RULE depends on PACKAGE_uvcdynctrl
#
# configuration written to .config
#
```

But I am a bit uncertain if it is caused by 
* the conditional udev dependency in uvcdynctrl or 
* the declaration conflict in libudev-fbsd as it says both "PROVIDES libudev" and also "CONFLICTS libudev"

https://github.com/openwrt/packages/blob/master/libs/libudev-fbsd/Makefile#L31
```
  DEPENDS:=libevdev
  PROVIDES:=libudev
  CONFLICTS:=libudev eudev udev
```
https://github.com/openwrt/packages/blob/master/utils/uvcdynctrl/Makefile#L26
```
DEPENDS:=+libwebcam +PACKAGE_UVCDYNCTRL_UDEV_RULE:udev
```

I remember seeing something about PROVIDES not being able to handle gracefully the exact name of an existing package. So maybe there should be an abstract "libudev-lib" that all provider packages would PROVIDE and user packages would DEPEND on. But not my specialty :-(

dangowrt, 2017-06-19:
As the udev daemon itself cannot and will never run properly on OpenWrt (nor LEDE), adding that rule simply doesn't make sense. You should remove the dependency on udev, as even if eudev is installed, it won't be able to handle that rule properly. If you need it, convert it to procd's json syntax and add that, no need to udev at all.
In the long-run, we should remove eudev and the 'stock' libudev as it cannot and will never work in coexistence with procd.

hnyman, 2017-06-19:
cc @roger-

dangowrt, 2017-06-19:
Should be fixed by https://github.com/openwrt/packages/commit/3d50afdf3c32ced1e34bab98567975aba49c7360 and https://github.com/openwrt/packages/commit/3d50afdf3c32ced1e34bab98567975aba49c7360 . The ineffective udev rule should be re-written in procd hotplug's json format if actually needed.

closed 2017-06-19 by dangowrt