openwrt/packages #4512: clamav: install conflict w/ luci-app-statistics on libltdl.so

aasgit, 2017-06-25:
Maintainer: @lucize 
Environment: armv7l, Linksys WRT3200ACM / LEDE Reboot (17.01.2, r3435-65eec8bd5f)

Description:

If trying to install ClamAV when luci-app-statistcs is already installed this happens (and vice versa: going to open a luci-app-statistics bug too).

```
root@lede-wyn:~# opkg install clamav
Installing clamav (0.98.7-1) to root...
Downloading http://downloads.lede-project.org/releases/17.01.2/packages/arm_cortex-a9_vfpv3/packages/clamav_0.98.7-1_arm_cortex-a9_vfpv3.ipk
Collected errors:
 * check_data_file_clashes: Package clamav wants to install file /usr/lib/libltdl.so.7
	But that file is already provided by package  * libltdl
 * check_data_file_clashes: Package clamav wants to install file /usr/lib/libltdl.so.7.3.0
	But that file is already provided by package  * libltdl
 * opkg_install_cmd: Cannot install package clamav.
```

lucize, 2017-06-25:
did you tried with the updated version ? please use the snapshots, I can't control what is going into the stable branch

```

root@LEDE:/tmp# opkg install clamav_0.99.2-1_x86_64.ipk
Installing clamav (0.99.2-1) to root...
Installing uclibcxx (0.2.4-3) to root...
Downloading http://downloads.lede-project.org/snapshots/packages/x86_64/base/uclibcxx_0.2.4-3_x86_64.ipk
Installing libpcre2 (10.23-1) to root...
Downloading http://downloads.lede-project.org/snapshots/packages/x86_64/packages/libpcre2_10.23-1_x86_64.ipk
Installing musl-fts (1.2.7-1) to root...
Downloading http://downloads.lede-project.org/snapshots/packages/x86_64/packages/musl-fts_1.2.7-1_x86_64.ipk
Configuring libpcre2.
Configuring uclibcxx.
Configuring musl-fts.
Configuring clamav.
root@LEDE:/tmp# opkg install luci-app-statistics
Installing luci-app-statistics (git-17.174.41033-de2e92a-1) to root...
Downloading http://downloads.lede-project.org/snapshots/packages/x86_64/luci/luci-app-statistics_git-17.174.41033-de2e92a-1_all.ipk
Configuring luci-app-statistics.
root@LEDE:/tmp#

```
Regards

aasgit, 2017-06-26:
hi @lucize 

thank you for the quick reply.

I'm on stable at the minute but I plan to install a snapshot ASAP since stable doesn't support wifi on my router yet. No ETA tho. I'm still making sure everything else is setup correctly and works. I also need a good script to backup/restore settings and packages before that too.

However, If you think this is solved in current and want to close it, then ok for me.

bobmseagithub, 2017-07-07:
I just suggested a workaround for this in [https://github.com/openwrt/luci/issues/1218](url)

lucize, 2017-07-07:
the updated package doesn't contain any reference to libltdl in ipk file

aasgit, 2017-07-09:
confirmed working on a custom built snapshot with Image Builder:

```
cat /etc/os-release
NAME="LEDE"
VERSION="SNAPSHOT, Reboot"
ID="lede"
ID_LIKE="lede openwrt"
PRETTY_NAME="LEDE Reboot SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="http://lede-project.org/"
BUG_URL="http://bugs.lede-project.org/"
SUPPORT_URL="http://forum.lede-project.org/"
BUILD_ID="r4542-0b493ed"
LEDE_BOARD="mvebu/generic"
LEDE_ARCH="arm_cortex-a9_vfpv3"
LEDE_TAINTS="no-all"
LEDE_DEVICE_MANUFACTURER="LEDE"
LEDE_DEVICE_MANUFACTURER_URL="http://lede-project.org/"
LEDE_DEVICE_PRODUCT="Generic"
LEDE_DEVICE_REVISION="v0"
LEDE_RELEASE="LEDE Reboot SNAPSHOT r4542-0b493ed"
```

closed 2017-07-09 by aasgit

diogosimao, 2018-03-21:
I have looked into this page [ClamAV ArchLinux](https://wiki.archlinux.org/index.php/ClamAV) and I "tried" the **Testing the software** section instructions, not exactly as it says, because on openwrt it works in another way. No info was logged by clamav when I accessed **EICAR test file** url.

I have installed the latest package as suggested on [GitHub Issue](https://github.com/openwrt/packages/issues/4512). I could find the latest package on [Snapshots](http://downloads.openwrt.org/snapshots/) -- e.g.: http://downloads.openwrt.org/snapshots/packages/mips_24kc/packages/

Is ClamAV installed on LEDE OpenWRT directed and created to deal with this kind of threats? I mean: if I access a malicious URL via wireless on my smartphone in my home network for instance, should it be blocked by ClamAV?

lucize, 2018-03-21:
Clamav usually runs against files, it's not scanning real-time memory, you should have a proxy set-up, and run clamav against the proxy, search for this on Google