openwrt/packages #4521: adblock: 2.8.0 introduces rapid DNS restarts due to netifd-procd interaction

EricLuehrsen, 2017-06-29:
@dibdot 
adblock 2.8.0 (e660813798832ffd8d7aad0c53161318c93eb024) added `wan6` to the procd trigger interfaces by default. Unfortunately, a design short fall in procd triggers prevent IFUP trigger filtering on "real" changes, and netifd registers IFUP for minor data like route lifetime on ip6. This causes adblock and the dns backend to restart every 1-10 minutes, depending on upstream RA signals. This busts the DNS cache and requires configuration reloads. See https://bugs.lede-project.org/index.php?do=details&task_id=713. I would suggest the default be only `wan`.

jow-, 2017-06-29:
Instead of working around the issue it might make more sense to figure out why adblock service reloads have side effects. A reload, even a frequent one, should be harmless.

dibdot, 2017-06-29:
@EricLuehrsen Thanks for heads up - I've reverted latest default config change.
@jow- the adblock reload itself is harmless, the frequent / rapid DNS restarts (at the end of the adblock processing) are the problem.

dibdot, 2017-06-29:
@jow- in adblock 2.8.2 I've made the DNS restart conditional, see #4525

EricLuehrsen, 2017-07-01:
@dibdot thanks for the clever fix.
@jow- the root cause still really should be fixed. procd does not have the same filtering hotplug has. netifd update announcements seem poorly conditioned; announce ip6 route update every few minutes when the lifetime is long and barely anything changed.

closed 2017-07-01 by EricLuehrsen