openwrt/packages #4522: php7-fpm: reduce size?

jammmet, 2017-06-29:
Maintainer: @mhei
Environment: ramips, MT7620, trunk

Description:
Not a bug per-se, but a question!
I am building `php7-fpm` for use with `nginx` as well as `php7-cli`. I end up with three packages that take up a lot of space on my box (plus a smaller assortment of php extensions):

`php7-cgi_7.1.6-2_mipsel_24kc` -> 1.4 MB
`php7-cli_7.1.6-2_mipsel_24kc` -> 1.4 MB
`php7-fpm_7.1.6-2_mipsel_24kc` -> 1.5MB

Do I effectively have 3 copies of the php engine? Is there a way I can reduce it to one?

mhei, 2017-06-30:
Yes, your analysis is correct. I'm not sure, but I think I remember that there would be a possibility to run th cli variant in cgi mode, or vice versa by giving a magic command line argument. I don't remember whether there were any drawbacks...
Could you elaborate more on your usecase, why you need all three variants?

jammmet, 2017-07-03:
Thanks for looking into this @mhei. We have quite a few php scripts we call from the command line, so cli support is important (although technically, these could be made to the local webserver). We have no obvious use for `php-cgi` except that `php-fpm DEPENDS+= +php7-cgi`.

mhei, 2017-08-19:
Sorry, I did not have had enough time to look into this deeper. Just want to let you know that it's still on my ToDo list...

mhei, 2017-09-26:
Finally found time to look into this for FPM... I don't understand why Ubuntu depends on CLI, but there are some more dependencies which might be due to default use-cases. For my test setup, I configured nginx to use PHP FPM and I did not install the CGI package. Everything worked so far, so I dropped the dep, see referenced commit.

mhei, 2017-10-13:
I'm closing this issue: for fpm variant the referenced commit reduces the footprint. While more optimizations might be possible as mentioned, I'd like to keep things as is until we have a really strong use-case to solve this. But for now I expect everybody who _really_ _wants_ PHP to have enough filesystem space available...

closed 2017-10-13 by mhei

pprindeville, 2017-10-13:
Just to play devil's advocate: one other reason for wanting to install minimal PHP support is to reduce attack surface...