openwrt/packages #4529: python-setuptools Do not compile

nmaas87, 2017-07-02:
Maintainer: @commodo 
Environment: OpenWRT Trunk, current

Description:
Compiling OpenWRT for Raspberry Pi 3 (brcm2708 / brcm2710) with python-setuptools results in following error:

```
if [ -f /home/user/OpenWRT/openwrt/staging_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/pkginfo/python.default.install.clean ]; then rm -f /home/user/OpenWRT/openwrt/staging_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/pkginfo/python.default.install /home/user/OpenWRT/openwrt/staging_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/pkginfo/python.default.install.clean; fi; echo "python-openssl" >> /home/user/OpenWRT/openwrt/staging_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/pkginfo/python.default.install
mkdir -p /home/user/OpenWRT/openwrt/bin/brcm2708/packages /home/user/OpenWRT/openwrt/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/Python-2.7.13/ipkg-brcm2708/python-setuptools/CONTROL /home/user/OpenWRT/openwrt/staging_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/pkginfo
install -d -m0755 /home/user/OpenWRT/openwrt/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/Python-2.7.13/ipkg-brcm2708/python-setuptools/usr/bin /home/user/OpenWRT/openwrt/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/Python-2.7.13/ipkg-brcm2708/python-setuptools/usr/lib/python2.7/site-packages
# Adjust shebang to proper python location on target
sed "1s/.*/#\!\/usr\/bin\/python2.7/" -i /home/user/OpenWRT/openwrt/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/Python-2.7.13/install-setuptools/bin/*
sed: can't read /home/user/OpenWRT/openwrt/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/Python-2.7.13/install-setuptools/bin/*: No such file or directory
Makefile:310: recipe for target '/home/user/OpenWRT/openwrt/bin/brcm2708/packages/packages/python-setuptools_28.8.0_brcm2708.ipk' failed
make[3]: *** [/home/user/OpenWRT/openwrt/bin/brcm2708/packages/packages/python-setuptools_28.8.0_brcm2708.ipk] Error 2
make[3]: Leaving directory '/home/user/OpenWRT/openwrt/feeds/packages/lang/python/python'
package/Makefile:196: recipe for target 'package/feeds/packages/python/compile' failed
make[2]: *** [package/feeds/packages/python/compile] Error 2
make[2]: Leaving directory '/home/user/OpenWRT/openwrt'
package/Makefile:193: recipe for target '/home/user/OpenWRT/openwrt/staging_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/stamp/.package_compile' failed
make[1]: *** [/home/user/OpenWRT/openwrt/staging_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/user/OpenWRT/openwrt'
/home/user/OpenWRT/openwrt/include/toplevel.mk:192: die Regel für Ziel „world“ scheiterte
make: *** [world] Fehler 2
```
Without python-setuptools, OpenWRT does compile successfully. Please leave me a message if I could be of help somehow :). Thanks a lot!

commodo, 2017-07-02:
Is this on a Mac ?
Looks like sed is complaining.

Will check it out.

Thanks

nmaas87, 2017-07-02:
@commodo Nope, it is on a recent Ubuntu server
````
user@Server:~$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 16.04.2 LTS
Release:        16.04
Codename:       xenial
user@Server:~$ uname -a
Linux Server 4.4.0-83-generic #106-Ubuntu SMP Mon Jun 26 17:54:43 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux
user@Server:~$
````

Thank you very much :)

commodo, 2017-07-02:
Oh.
Just got in front of a computer and noticed the sed error a bit better.

I'm on XUbuntu 14.04; will try on that first ; to build the brcm2708 / brcm2710 subtarget.
If I don't see the err, will try a Ubuntu 16.04 VM.

commodo, 2017-07-02:
it built on xubuntu 14.04
will try on a vm with ubuntu 16.04 tomorrow

quick q: is this err [you're seeing] on a clean checkout ?
or, did you build a part of the packages and tried to enable python-setuptools after that ?
reason i'm asking: i'm also seeing [on a few packages] that I need to add some dependencies on some `CONFIG_PACKAGE_xxx` symbols to re-trigger a build

pprindeville, 2017-07-02:
By the way, if there does end up being a commit to fix this, can we also address this low-hanging fruit:

`sed "1s/.*/#\!\/usr\/bin\/python2.7/"  ...`

could be more easily expressed as:

`sed "1s@.*@#\!/usr/bin/python2.7/@" ...`

so that a lot of unnecessary escaping can be avoided.

Thanks

nmaas87, 2017-07-03:
@commodo You're right I did build the trunk with rpi1-3 and x86/x64 earlier on with some additional packages and then did remove those because I could not find the error until I could nail it down to setuptools (at least thats what I think it is, no offense :( ).
Also, I need to mention that I get the very disturbing warning
````
fatal: No names found, cannot describe anything.
````
if I do any make command. I never saw that earlier and just thought it would be because of the fact that I did checkout trunk freshly and that something was a bit flunky on that end :/

commodo, 2017-07-03:
@pprindeville 
ack

@nmaas87 
no offense taken;
will look a bit more into this
maybe i need to adjust some `CONFIG_PACKAGE_` symbols

commodo, 2017-07-03:
@nmaas87 
i understand now the issue; will spin-up a PR to fix it

thanks for reporting

closed 2017-07-03 by jow-

nmaas87, 2017-07-03:
I deleted the whole OpenWRT folder, checked out everything from scratch and started to build it again.
I ended up at this point:
````
arm-openwrt-linux-muslgnueabi-gcc -shared -L/home/user/OpenWRT/openwrt/staging_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/usr/lib -L/home/user/OpenWRT/openwrt/staging_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/lib -L/home/user/OpenWRT/openwrt/staging_dir/toolchain-arm_cortex-a53+neon-vfpv4_gcc-5.3.0_musl-1.1.16_eabi/usr/lib -L/home/user/OpenWRT/openwrt/staging_dir/toolchain-arm_cortex-a53+neon-vfpv4_gcc-5.3.0_musl-1.1.16_eabi/lib -znow -zrelro -L/home/user/OpenWRT/openwrt/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/Python-2.7.13 -L/home/user/OpenWRT/openwrt/staging_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/usr/lib -L/home/user/OpenWRT/openwrt/staging_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/lib -L/home/user/OpenWRT/openwrt/staging_dir/toolchain-arm_cortex-a53+neon-vfpv4_gcc-5.3.0_musl-1.1.16_eabi/usr/lib -L/home/user/OpenWRT/openwrt/staging_dir/toolchain-arm_cortex-a53+neon-vfpv4_gcc-5.3.0_musl-1.1.16_eabi/lib -znow -zrelro -L/home/user/OpenWRT/openwrt/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/Python-2.7.13 -fno-strict-aliasing -Os -pipe -march=armv8-a -mtune=cortex-a53 -mfpu=neon-vfpv4 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap /home/user/OpenWRT/openwrt/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/Python-2.7.13:Python-2.7.13 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DNDEBUG -fno-inline -DNDEBUG -Os -pipe -march=armv8-a -mtune=cortex-a53 -mfpu=neon-vfpv4 -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -iremap /home/user/OpenWRT/openwrt/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/Python-2.7.13:Python-2.7.13 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro build/temp.linux2-2.7/home/user/OpenWRT/openwrt/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/Python-2.7.13/Modules/_ctypes/_ctypes.o build/temp.linux2-2.7/home/user/OpenWRT/openwrt/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/Python-2.7.13/Modules/_ctypes/callbacks.o build/temp.linux2-2.7/home/user/OpenWRT/openwrt/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/Python-2.7.13/Modules/_ctypes/callproc.o build/temp.linux2-2.7/home/user/OpenWRT/openwrt/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/Python-2.7.13/Modules/_ctypes/stgdict.o build/temp.linux2-2.7/home/user/OpenWRT/openwrt/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/Python-2.7.13/Modules/_ctypes/cfield.o -L/home/user/OpenWRT/openwrt/staging_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/usr/lib -L/home/user/OpenWRT/openwrt/staging_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/lib -L/home/user/OpenWRT/openwrt/staging_dir/toolchain-arm_cortex-a53+neon-vfpv4_gcc-5.3.0_musl-1.1.16_eabi/usr/lib -L/home/user/OpenWRT/openwrt/staging_dir/toolchain-arm_cortex-a53+neon-vfpv4_gcc-5.3.0_musl-1.1.16_eabi/lib -L/home/user/OpenWRT/openwrt/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/Python-2.7.13 -L/home/user/OpenWRT/openwrt/staging_dir/toolchain-arm_cortex-a53+neon-vfpv4_gcc-5.3.0_musl-1.1.16_eabi/arm-openwrt-linux-muslgnueabi/lib -L. -lffi -lpython2.7 -o build/lib.linux2-2.7/_ctypes.so

Python build finished, but the necessary bits to build these modules were not found:
_tkinter           bsddb185           dl
imageop            nis                sunaudiodev
To find the necessary bits, look in setup.py in detect_modules() for the module's name.

running build_scripts
creating build/scripts-2.7
copying /home/user/OpenWRT/openwrt/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/Python-2.7.13/Tools/scripts/pydoc -> build/scripts-2.7
copying /home/user/OpenWRT/openwrt/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/Python-2.7.13/Tools/scripts/idle -> build/scripts-2.7
copying /home/user/OpenWRT/openwrt/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/Python-2.7.13/Tools/scripts/2to3 -> build/scripts-2.7
copying /home/user/OpenWRT/openwrt/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/Python-2.7.13/Lib/smtpd.py -> build/scripts-2.7
/usr/bin/install -c -m 644 ./Tools/gdb/libpython.py python-gdb.py
make[4]: Leaving directory '/home/user/OpenWRT/openwrt/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/Python-2.7.13'
# Use host pip to install python-setuptools
/home/user/OpenWRT/openwrt/staging_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/host/bin/pip install --ignore-installed --root=/home/user/OpenWRT/openwrt/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/Python-2.7.13/install-setuptools --prefix=. /home/user/OpenWRT/openwrt/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/Python-2.7.13/Lib/ensurepip/_bundled/setuptools-*.whl
bash: /home/user/OpenWRT/openwrt/staging_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/host/bin/pip: No such file or directory
Makefile:313: recipe for target '/home/user/OpenWRT/openwrt/build_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/Python-2.7.13/.built' failed
[... cut ...]
package/Makefile:193: recipe for target '/home/user/OpenWRT/openwrt/staging_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/stamp/.package_compile' failed
make[1]: *** [/home/user/OpenWRT/openwrt/staging_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/user/OpenWRT/openwrt'
/home/user/OpenWRT/openwrt/include/toplevel.mk:192: die Regel für Ziel „world“ scheiterte
make: *** [world] Fehler 2
````

Does this still belong to the same error?

Regards

commodo, 2017-07-03:
hmm;
not sure;

to be honest i tested my PR on LEDE trunk; i forgot to test on OpenWrt
will retry on OpenWrt

i do hope the re-merge happens soon ; testing on both distros is a bit annoying

commodo, 2017-07-03:
i wonder if you could give me your `.config` file ?
so that i can try it on my machine ; that way at least there's a more common baseline

nmaas87, 2017-07-04:
@commodo 
Sure :)!
I create my .config via a small script, you just have to run this in your openwrt root folder. It adds my own package repo, updates trunk and packages and creates the needed config. Just give it a shot :).

````
# only for the first time to add repo
cp feeds.conf.default feeds.conf
echo "src-git wiringPi https://github.com/nmaas87/rpi-openwrt-packages.git" >> feeds.conf
# start
echo "Upgrade repo and packages"
git pull
./scripts/feeds update -a
./scripts/feeds install -a
echo "TRUNK BUILD RPI03"
echo CONFIG_TARGET_brcm2708=y > .config
echo CONFIG_TARGET_brcm2708_bcm2710=y >> .config
echo '
CONFIG_PACKAGE_wiringPi=y
CONFIG_PACKAGE_python-wiringpi2=y
CONFIG_PACKAGE_paho_mqtt=y
CONFIG_PACKAGE_i2c-tools-clone=y
CONFIG_PACKAGE_python-smbus=y
CONFIG_PACKAGE_libi2c=y
CONFIG_PACKAGE_ssd1306_i2c=y
CONFIG_PACKAGE_oled-exp=y
CONFIG_PACKAGE_pwm-exp=y
CONFIG_PACKAGE_relay-exp=y
CONFIG_PACKAGE_liboniondebug=y
CONFIG_PACKAGE_libonioni2c=y
CONFIG_PACKAGE_libonionmcp23008=y
CONFIG_PACKAGE_libonionoledexp=y
CONFIG_PACKAGE_libonionpwmexp=y
CONFIG_PACKAGE_libonionrelayexp=y
CONFIG_PACKAGE_pyOledExp=y
CONFIG_PACKAGE_pyOmegaExpansion=y
CONFIG_PACKAGE_pyOnionI2C=y
CONFIG_PACKAGE_pyPwmExp=y
CONFIG_PACKAGE_pyRelayExp=y
' >> .config
make defconfig
echo CONFIG_TARGET_ROOTFS_TARGZ=y >> .config
make -j4 V=s
echo "TRUNK DONE BUILD RPI03"
````

commodo, 2017-07-04:
@nmaas87 
aside from a few hiccups [with downloading stuff], it built for me
this time i tried on a XUbuntu 16.04 machine
maybe it's something that's specific to your host ;
some packages that are installed on your system and not on mine

```
Generating index for package ./python-unittest_2.7.13-5_brcm2708.ipk
Generating index for package ./python-xml_2.7.13-5_brcm2708.ipk
Generating index for package ./python_2.7.13-5_brcm2708.ipk
Generating index for package ./i2c-tools-clone_2017-04-07-3.1.2+_brcm2708.ipk
Generating index for package ./libi2c_2017-04-07-3.1.2+_brcm2708.ipk
Generating index for package ./liboniondebug_0.4-1_brcm2708.ipk
Generating index for package ./libonioni2c_0.4-1_brcm2708.ipk
Generating index for package ./libonionmcp23008_0.4-1_brcm2708.ipk
Generating index for package ./libonionoledexp_0.4-1_brcm2708.ipk
Generating index for package ./libonionpwmexp_0.4-1_brcm2708.ipk
Generating index for package ./libonionrelayexp_0.4-1_brcm2708.ipk
Generating index for package ./oled-exp_0.4-1_brcm2708.ipk
Generating index for package ./paho_mqtt_1.3.0-1_brcm2708.ipk
Generating index for package ./pwm-exp_0.4-1_brcm2708.ipk
Generating index for package ./pyOledExp_0.4-1_brcm2708.ipk
Generating index for package ./pyOmegaExpansion_0.4-1_brcm2708.ipk
Generating index for package ./pyOnionI2C_0.4-1_brcm2708.ipk
Generating index for package ./pyPwmExp_0.4-1_brcm2708.ipk
Generating index for package ./pyRelayExp_0.4-1_brcm2708.ipk
Generating index for package ./python-smbus_2017-04-07-3.1.2+_brcm2708.ipk
Generating index for package ./python-wiringpi2_2.44.0-1_brcm2708.ipk
Generating index for package ./relay-exp_0.4-1_brcm2708.ipk
Generating index for package ./ssd1306_i2c_0.3-1_brcm2708.ipk
Generating index for package ./wiringPi_2017-03-03-2.44_brcm2708.ipk
Signing package index...
make[2]: Leaving directory '/home/sandu/work/openwrt'
make[1]: Leaving directory '/home/sandu/work/openwrt'
TRUNK DONE BUILD RPI03
```

host-side builds [ and especially the host-side build for Python/Python3] are a bit picky
i put your script into a file and ran that;
i had to re-run it a few times

hmm, do you cache tarballs ?
for my case specifically, before any build I do  ` ln -s ~/work/dl-cache dl` ;
so that the build does not waste time with trying to download [and maybe fail]
[ i need to delete my ` ~/work/dl-cache` folder  every now-n-then to get rid of ancient stuff, and multiple kernel versions ]

and, i am not sure if this is still valid; but there used to be a time [in older OpenWrt] that parallel builds would sometimes randomly fail; it was never deterministic ;
i don't know if this is the case here

so:
1) after a build fails, and you try to do `make V=s`, does the build continue/work ? 
2) would it be an idea to try `make V=s` for a full build ? I do know that it takes some time, but, atm, I don't have many ideas to ask/suggest

nmaas87, 2017-07-04:
@commodo Thanks for all your help, I could finally hammer it down to the real reason - there was some problem with pip! :( For reasons unknown in ````cd ~/OpenWRT/openwrt/staging_dir/target-arm_cortex-a53+neon-vfpv4_musl-1.1.16_eabi/host/bin```` - he did not generate / install / link pip and that tampered with the compile process. I went for the sledgehammer method and did link-in pip ````ln -s /usr/local/bin/pip pip```` - after that, it worked (at least it went past the problematic stage of python modules). I'll try to let it compile until the end, then clear the whole folder and start from scratch. I found pip to be installed in /usr/local/bin as well as some links to local user files - I'll report back if uninstalling python-pip and cleaning up has resolved the issue in the morning.

commodo, 2017-07-04:
hmm ; interesting ; 
i don't have pip installed on my machine ; 
would be interesting to see if that changes anything

i did see [from your initial logs] that pip is not installed host-side ;
i was hoping i could also reproduce it on my side, to see the cause ;

nmaas87, 2017-07-05:
Seems like it worked and it built successfully. I am sorry that I cannot help any further in finding that error, but seems like it tried to rely on local installed pip - which was broken and therefore could not advance in the build process. Maybe forcing it to use its "internal" pip only would be an good idea - as it would not need to rely on some maybe broken installation on the host pc.

commodo, 2017-07-05:
well, this information is good as a starting point ;
thanks for finding it :)

maybe some path or build variables need to be updated/enforced

nmaas87, 2017-07-05:
Thank you very much for bearing with me ;)

commodo, 2017-07-05:
no worries :)
it wasn't difficult as you make it sound

commodo, 2017-07-09:
@nmaas87 

as a quick update from me;
i tried installing a ubuntu server VM and installed pip on the machine, to see if it interferes somehow
for me, it did not ;
i tried symlinking pip to `/usr/local/bin`
still did not fail

this does not rule out completely the fact that there isn't a hidden issue in there
but [ as I said before ] tracking down host side failures for python [especially] can be tricky [ because of all the setup.py magic ]

let's see ; maybe later there will be a more clear thing that would identify this