openwrt/packages #4534: AdBlock: Whitelist not working?

zivkovic, 2017-07-02:
Maintainer: Dirk Brenken (dev@brenken.org)
Environment: (Tp-Link Archer C7 v2, LEDE Reboot 17.01.1 r3316-7eb58cf109)

Description:

I have added "phonearena.com" to the whitelist, but adblock still blocks the domain on the router.
I am using the luci plugin for adblock.

Those are selected adblock lists:
http://prntscr.com/fqsact

And here you can see the whitelist:
http://prntscr.com/fqsao1

Thank you!

dibdot, 2017-07-02:
@zivkovic After you've added a new whitelist entry via "Submit" you have to re-run adblock as well ... switch to the overview page and hit "Save & Apply".

zivkovic, 2017-07-02:
I have dont that & rebooted the router after I clicked "Save & Apply" and it still wasn't working. Any other ideas?

dibdot, 2017-07-02:
No. Please post the output of cat /etc/adblock/adblock.whitelist and post the global section of /etc/config/adblock. By the way, which version of adblock did you use?

zivkovic, 2017-07-02:
Okay, here are the requested files:
`root@LEDE:~# cat /etc/adblock/adblock.whitelist

phonearena.com
root@LEDE:~#
`
Global section of /etc/config/adblock:
`config adblock 'global'
        option adb_enabled '1'
        option adb_debug '0'
        option adb_forcesrt '0'
        option adb_forcedns '0'
        option adb_iface 'wan'
        option adb_triggerdelay '2'
        option adb_whitelist '/etc/adblock/adblock.whitelist'
        option adb_whitelist_rset '\$1 ~/^([A-Za-z0-9_-]+\.){1,}[A-Za-z]+/{print tolower(\"^\"\$1\"\\\|[.]\"\$1)}'
        option adb_backup '0'
        option adb_backupdir '/mnt'
        option adb_rtfile '/tmp/adb_runtime.json'
`

dibdot, 2017-07-02:
your whitelist ruleset is borked, take this one:

option adb_whitelist_rset '\$1 ~/^([A-Za-z0-9_-]+\.){1,}[A-Za-z]+/{print tolower(\"^\"\$1\"\\\|[.]\"\$1)}'

zivkovic, 2017-07-02:
Okay, I have changed that setting, rebooted, still the same.

dibdot, 2017-07-02:
No, that makes no harm. I have no further idea, I can't reproduce this issue. 

Again, which version of adblock did you use?!

zivkovic, 2017-07-02:
Im using 2.6.2 version of adblock.

dibdot, 2017-07-02:
Please check with current trunk version (2.8.2) again - thanks.

zivkovic, 2017-07-02:
How would I go about that? opkg upgrade adblock does not upgrade as it seems to be the latest?

dibdot, 2017-07-02:
See the download links [here](https://forum.lede-project.org/t/adblock-2-x-support-thread/507)

zivkovic, 2017-07-02:
Volia, updating to latest versions worked. Thank you!

closed 2017-07-02 by zivkovic