openwrt/packages #4537: nginx: lua-nginx compile error

jammmet, 2017-07-03:
Maintainer: @heil 
Environment: ramips, MT7620, trunk

Description:

enabling Lua module on with `lua-nginx` cause compile to fail. Output below:

```
nginx-1.12.0/lua-nginx/src/ngx_http_lua_headers.c
nginx-1.12.0/lua-nginx/src/ngx_http_lua_headers.c: In function 'ngx_http_lua_ngx_req_raw_header':
nginx-1.12.0/lua-nginx/src/ngx_http_lua_headers.c:145:15: error: incompatible types when assigning to type 'ngx_buf_t * {aka struct ngx_buf_s *}' from type 'ngx_chain_t {aka struct ngx_chain_s}'
             b = hc->busy[i];
               ^
nginx-1.12.0/lua-nginx/src/ngx_http_lua_headers.c:221:15: error: incompatible types when assigning to type 'ngx_buf_t * {aka struct ngx_buf_s *}' from type 'ngx_chain_t {aka struct ngx_chain_s}'
             b = hc->busy[i];
               ^
make[5]: *** [objs/addon/src/ngx_http_lua_headers.o] Error 1
```

ljubomirb, 2017-07-06:
Makefile is somewhat old, and within makefile, there is a reference to lua nginx module git that at the time had bug, and has been fixed since.
To compile, change VERSION like here  #https://github.com/openresty/lua-nginx-module/issues/1016

jammmet, 2017-07-17:
@ljubomirb that's what I did and can verify it's compiling and working with an updated `lua-nginx'module`

ljubomirb, 2017-07-17:
@jammmet I've created new pull request #4594
If you have time, check that, and see if it works for you. I've also put nchan package, for which I think is very handy. For me everything compiles, but nchan is not working on ar71xxx (but nginx does!?). I'm not sure if it's firewall issue, or some other patch conflicting, or... It could be well be that it works on other platforms.

closed 2018-06-12 by heil