openwrt/packages #4548: python compile error

build000, 2017-07-05:
```
dom@machina:~/Pulpit/Projekty/WITI-LEDE-REBOOT-small$ scripts/getver.sh
r4529-05d6e92

(...)
make[3]: Entering directory '/home/dom/Pulpit/Projekty/WITI-LEDE-REBOOT-small/feeds/packages/lang/python/python'
rm -rf /home/dom/Pulpit/Projekty/WITI-LEDE-REBOOT-small/tmp/stage-python
mkdir -p /home/dom/Pulpit/Projekty/WITI-LEDE-REBOOT-small/tmp/stage-python/host /home/dom/Pulpit/Projekty/WITI-LEDE-REBOOT-small/staging_dir/target-mipsel_24kc_musl/packages /home/dom/Pulpit/Projekty/WITI-LEDE-REBOOT-small/staging_dir/host/packages
install -d -m0755 /home/dom/Pulpit/Projekty/WITI-LEDE-REBOOT-small/tmp/stage-python/usr/include/ /home/dom/Pulpit/Projekty/WITI-LEDE-REBOOT-small/tmp/stage-python/usr/lib/ /home/dom/Pulpit/Projekty/WITI-LEDE-REBOOT-small/tmp/stage-python/usr/lib/pkgconfig
install -d -m0755 /home/dom/Pulpit/Projekty/WITI-LEDE-REBOOT-small/tmp/stage-python/usr/lib/python2.7/
install -m0644 ./files/python-package.mk ./files/python-host.mk ./files/python-version.mk ./files/python-package-install.sh /home/dom/Pulpit/Projekty/WITI-LEDE-REBOOT-small/staging_dir/target-mipsel_24kc_musl/mk/
install: target '/home/dom/Pulpit/Projekty/WITI-LEDE-REBOOT-small/staging_dir/target-mipsel_24kc_musl/mk/' is not a directory: No such file or directory
Makefile:313: recipe for target '/home/dom/Pulpit/Projekty/WITI-LEDE-REBOOT-small/staging_dir/target-mipsel_24kc_musl/stamp/.python_installed' failed
make[3]: *** [/home/dom/Pulpit/Projekty/WITI-LEDE-REBOOT-small/staging_dir/target-mipsel_24kc_musl/stamp/.python_installed] Error 1
make[3]: Leaving directory '/home/dom/Pulpit/Projekty/WITI-LEDE-REBOOT-small/feeds/packages/lang/python/python'
package/Makefile:109: recipe for target 'package/feeds/packages/python/python/compile' failed
make[2]: *** [package/feeds/packages/python/python/compile] Error 2
make[2]: Leaving directory '/home/dom/Pulpit/Projekty/WITI-LEDE-REBOOT-small'
package/Makefile:105: recipe for target '/home/dom/Pulpit/Projekty/WITI-LEDE-REBOOT-small/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: *** [/home/dom/Pulpit/Projekty/WITI-LEDE-REBOOT-small/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/dom/Pulpit/Projekty/WITI-LEDE-REBOOT-small'
/home/dom/Pulpit/Projekty/WITI-LEDE-REBOOT-small/include/toplevel.mk:207: polecenia dla obiektu 'world' nie powiodły się
make: *** [world] Błąd 2
```

build000, 2017-07-05:
I do not know what was wrong and why - I made "make dirclean && meke menuconfig" and then make ... and went without a problem - something must have happened in the system why did not handle procek at -j8 - seems to be waiting for me to vacuum the computer .. .hehe.
sry ( :+1: )

commodo, 2017-07-05:
hmm ;
so there could be some fault that's occurring here [maybe intermitently]
will check it out

pprindeville, 2017-07-05:
This was closed prematurely.

I'm also seeing this issue following rebasing last night.

```
make[3]: Entering directory '/home/philipp/bertram/lede/feeds/packages/lang/python/python'
rm -rf /home/philipp/bertram/lede/tmp/stage-python
mkdir -p /home/philipp/bertram/lede/tmp/stage-python/host /home/philipp/bertram/lede/staging_dir/target-x86_64_musl_powercode-bmu/packages /home/philipp/bertram/lede/staging_dir/host/packages
install -d -m0755 /home/philipp/bertram/lede/tmp/stage-python/usr/include/ /home/philipp/bertram/lede/tmp/stage-python/usr/lib/ /home/philipp/bertram/lede/tmp/stage-python/usr/lib/pkgconfig
install -d -m0755 /home/philipp/bertram/lede/tmp/stage-python/usr/lib/python2.7/
install -m0644 ./files/python-package.mk ./files/python-host.mk ./files/python-version.mk ./files/python-package-install.sh /home/philipp/bertram/lede/staging_dir/target-x86_64_musl_powercode-bmu/mk/
install: target '/home/philipp/bertram/lede/staging_dir/target-x86_64_musl_powercode-bmu/mk/' is not a directory: No such file or directory
Makefile:310: recipe for target '/home/philipp/bertram/lede/staging_dir/target-x86_64_musl_powercode-bmu/stamp/.python_installed' failed
make[3]: *** [/home/philipp/bertram/lede/staging_dir/target-x86_64_musl_powercode-bmu/stamp/.python_installed] Error 1
make[3]: Leaving directory '/home/philipp/bertram/lede/feeds/packages/lang/python/python'
package/Makefile:109: recipe for target 'package/feeds/packages/python/python/compile' failed
make[2]: *** [package/feeds/packages/python/python/compile] Error 2
make[2]: Leaving directory '/home/philipp/bertram/lede'
package/Makefile:105: recipe for target '/home/philipp/bertram/lede/staging_dir/target-x86_64_musl_powercode-bmu/stamp/.package_compile' failed
make[1]: *** [/home/philipp/bertram/lede/staging_dir/target-x86_64_musl_powercode-bmu/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/philipp/bertram/lede'
/home/philipp/bertram/lede/include/toplevel.mk:207: recipe for target 'world' failed
make: *** [world] Error 2
```

yousong, 2017-07-06:
It's probably caused by commit e88e35249f06a8fe7c543572fb424b752fec22d0 removing the mkdir `$(STAGING_DIR)/mk/`

commodo, 2017-07-06:
ack 
i was also eye-balling that change last night ; but decided to leave it for next morning 

will fix

closed 2017-07-06 by yousong