openwrt/packages #4549: clamav: not in mvebu snapshot repos?

aasgit, 2017-07-05:
Maintainer: @lucize 
Environment: mvebu/generic ; arm_cortex-a9_vfpv3; profile linksys-wrt3200acm

Description:

I'm building a custom image with the official Image Builder from https://downloads.lede-project.org/snapshots/targets/mvebu/generic/ using the following 'make image' command:

```
make image PROFILE="linksys-wrt3200acm" PACKAGES="adblock bash bind-dig block-mount ca-bundle ca-certificates coreutils-base64 coreutils-sha512sum curl diffutils dropbear e2fsprogs fdisk file gpgv htop iftop lftp luci luci-app-adblock luci-app-clamav luci-app-ddns luci-app-fwknopd luci-app-openvpn luci-app-sqm luci-app-transmission luci-app-uhttpd luci-app-watchcat luci-base luci-ssl-openssl netdata nginx odhcp6c openvpn-openssl procps-ng-vmstat rsync shadow-groupadd shadow-useradd shadow-userdel shadow-usermod sudo swap-utils sysstat tmux transmission-cli-openssl transmission-daemon-openssl transmission-remote-openssl transmission-web unzip usbutils vim-full vim-runtime xzgrep" FILES="files/"
```
...and every time the 'clamav' package is missing:

```
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-clamav:
 *      clamav *
```

I tried with basically every snapshot today, the latest being this (minutes ago):

```
lede-imagebuilder-mvebu.Linux-x86_64.tar.xz| 2bb550ab1aa808210631a4c0f2d99873d42c43e8983f47263ea9f6a3bf902777 | 31797.7 KB| Wed Jul 5 22:00:56 2017
```
..but clamav is not there. Was it removed? Because stable 17.01.2 mvebu does have it available with opkg.

However, 'luci-app-clamav' requires 'clamav' as a dependency (even if it installs).

The bottom line is that the build doesn't fail, although it has yet to be tested for install. But this missing package fails the main purpose for custom image building (e.g: packages + custom configs, beside wifi that doesn't work on stable).

I hope this is helpful enough.

I'm available for tests (although I would avoid flashing from an incomplete image).

Please and thank you :)

pprindeville, 2017-07-05:
Really dumb question, but what happens if you put `clamav` into the list of `PACKAGES`?

aasgit, 2017-07-06:
@pprindeville it's even worse:

Downloading obviously fails:

```
Installing luci-app-clamav (git-17.186.70712-4f127c3-1) to root...
Downloading http://downloads.lede-project.org/snapshots/packages/arm_cortex-a9_vfpv3/luci/luci-app-clamav_git-17.186.70712-4f127c3-1_all.ipk
Unknown package 'clamav'.
```
...and it won't even build:

```
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-clamav:
 *      clamav *
 * opkg_install_cmd: Cannot install package clamav.
make[2]: *** [Makefile:150: package_install] Error 255
make[1]: *** [Makefile:115: _call_image] Error 2
make: *** [Makefile:202: image] Error 2
```

I hope this helps

lucize, 2017-07-06:
on snapshot faillogs I see it on x86 fail with
`configure: error: OpenSSL not found.`
maybe this is the problem

aasgit, 2017-07-06:
that's something :)

lucize, 2017-07-06:
I'll see into it if ssl is actually needed or if it can be disabled, otherwise I think I need to add openssl to the dependencies

aasgit, 2017-07-06:
thank you very much @lucize , really appreciated :)

aasgit, 2017-07-06:
ps: apparently it was introduced as a dependency in 2014: http://blog.clamav.net/2014/02/introducing-openssl-as-dependency-to.html

lucize, 2017-07-06:
it already has +libopenssl in DEPENDS, maybe I need to add  --with-openssl[=DIR]  argument
I'll test it

lucize, 2017-07-06:
I made a minimal install and tried to build it, it would fail without specifying the path in the makefile so #4554 should fix it

closed 2017-07-06 by hnyman

aasgit, 2017-07-09:
@lucize thank you dude :) it works smoothly now 👍

PR #4554 (2017-07-06): clamav: Fix OpenSSL dependency

Maintainer: me 
Compile/Run tested: x86 Lede r4531-c2a4f14

Description:
Should fix #4549 because buildbots are failing at finding OpenSSL path