openwrt/packages #4564: wireguard-tools and kmod-wireguard: Please update wireguard for LEDE to the latest version

myly10, 2017-07-08:
Maintainer: @danrl

The wireguard in LEDE package repo is out of date and cannot be interconnected with other wireguard servers running the latest release. So I'm raising this issue to request for package updating.

Thanks!

danrl, 2017-07-08:
WireGuard is fast-moving, experimental software. Please consider running the latest snapshot if you rely on experimental features.

9shearer, 2017-07-09:
I'm in a bit of a dilemma regarding this.
My router (Netgear Nighthawk R7000, using bcm53xx) isn't fully supported in the latest LEDE snapshot (no WiFi driver). On the other hand, I would really like to use WireGuard.
Therefore the question: is it possible to install the latest **stable LEDE** and then update the WireGuard-related package(s) to the **latest snapshot** level? I see the kernel is only some minor versions away, would that work?

danrl, 2017-07-09:
You can try that, but I have my doubts. After all, it is a kernel module so changing the kernel would then leave you with the wifi problem IIRC.

zorun, 2017-07-09:
I think it would make sense to bump the version in `lede-17.01`, for the next stable release of LEDE.  But because of 658ffbb3fd9488ccaea3042e4e64f16b6be318ad the config would not be backward-compatible, and it would also require a bump of the luci app.

@9shearer  In the meantime, you can compile the latest version of wireguard yourself (take the wireguard version from master and compile it with LEDE 17.01.2)

9shearer, 2017-07-09:
@zorun that's actually a pretty good idea. I'm now conjuring the courage to flash my (so far well-functioning) router. Thanks.

zorun, 2017-07-09:
@9shearer actually, if you use the exact same version of LEDE (use the git tag of the release you are running on your router), you should be able to compile the wireguard package and install it without reflashing your router.

9shearer, 2017-07-09:
@zorun that's the thing, I am now running xwrt-vortex, which works generally fine for my purposes. Which is also I am hesitating to embark on this LEDE journey.

9shearer, 2017-07-16:
Further question: what is the current kmod-wireguard version in LEDE 17.0.1.2? The project website only states "see kernel", which corresponds to 4.4.71 (i.e. the June release).
Otherwise stated: is the kmod-wireguard setup in 17.0.1.2 built from a Wireguard source version before or after May 17th, 2017? 

Thanks!

danrl, 2017-07-16:
https://github.com/openwrt/packages/blob/lede-17.01/net/wireguard/Makefile
Jan 15 2017 is the snapshot from.

9shearer, 2017-07-16:
Right, found that out myself. I've been struggling for the past few hours to get that updated, but no chance. I suppose my cross-compiling skills are not as good as I thought. Any suggestions (unless someone would do the world a favor and updated the package in the stable branch)?

closed 2018-08-05 by hnyman