openwrt/packages #4567: ip-full: fails to symlink /sbin/ip -> /sbin/ip-full

aasgit, 2017-07-10:
Maintainer: I could not find the package/maintainer sorry.
Environment: mvebu/generic ; arm_cortex-a9_vfpv3; profile linksys-wrt3200acm

Description:

I'm building a custom image with the official Image Builder from https://downloads.lede-project.org/snapshots/targets/mvebu/generic/ using the following 'make image' command:

```
make image PROFILE="linksys-wrt3200acm" PACKAGES="adblock bash bind-dig block-mount ca-bundle ca-certificates coreutils-base64 coreutils-sha512sum curl diffutils dropbear e2fsprogs fdisk file git git-http gpgv htop iftop ip-full lftp luci luci-app-adblock luci-app-clamav luci-app-ddns luci-app-fwknopd luci-app-openvpn luci-app-sqm luci-app-transmission luci-app-uhttpd luci-app-watchcat luci-base luci-ssl-openssl netdata nginx odhcp6c openvpn-openssl ppp-mod-pppoa procps-ng-vmstat rsync shadow-groupadd shadow-useradd shadow-userdel shadow-usermod sudo swap-utils sysstat tmux transmission-cli-openssl transmission-daemon-openssl transmission-remote-openssl transmission-web unzip usbutils vim-full vim-runtime xzgrep" FILES="files/"
```
...and every time 'ip-full' symlink fails:

```
Collected errors:
 * pkg_alternatives_update_path: failed symlinking /home/andrea/LEDE/ImageBuilding/lede-buildwrapper/lede-imagebuilder-mvebu.Linux-x86_64/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/root-mvebu/sbin/ip -> /sbin/ip-full
```

I tried with a few snapshot today, the latest being this (minutes ago):

```
lede-imagebuilder-mvebu.Linux-x86_64.tar.xz| 89a3fed9d0553b3dc06c47a775f54d160fbfa142c0c61fbae05fab7e06d44ed7 | 33763.1 KB| Mon Jul 10 11:32:05 2017
```

The build doesn't fail. But it would be nice to have ip-full properly symlinked.

I hope this is helpful enough.

I'm available for tests (although I would avoid flashing from an incomplete image).

Please and thank you :)

aasgit, 2017-07-10:
ps: if it helps, installing it manually on an existing system works fine.

currently installed custom image where this is true:

```
cat /etc/os-release
NAME="LEDE"
VERSION="SNAPSHOT, Reboot"
ID="lede"
ID_LIKE="lede openwrt"
PRETTY_NAME="LEDE Reboot SNAPSHOT"
VERSION_ID="snapshot"
HOME_URL="http://lede-project.org/"
BUG_URL="http://bugs.lede-project.org/"
SUPPORT_URL="http://forum.lede-project.org/"
BUILD_ID="r4542-0b493ed"
LEDE_BOARD="mvebu/generic"
LEDE_ARCH="arm_cortex-a9_vfpv3"
LEDE_TAINTS="no-all"
LEDE_DEVICE_MANUFACTURER="LEDE"
LEDE_DEVICE_MANUFACTURER_URL="http://lede-project.org/"
LEDE_DEVICE_PRODUCT="Generic"
LEDE_DEVICE_REVISION="v0"
LEDE_RELEASE="LEDE Reboot SNAPSHOT r4542-0b493ed"
```

hnyman, 2017-07-10:
Wrong repo. ip-full is part iproute2 in the main source repo.
https://github.com/lede-project/source/blob/master/package/network/utils/iproute2/Makefile

So you should file this in the main bug tracker.
https://bugs.lede-project.org/

This packages feed has no role with ip-full.

closed 2017-07-10 by hnyman

aasgit, 2017-07-10:
apologies and thanks for the info.

yousong, 2017-07-11:
Issue confirmed.  I will prepare a patch and Thanks for reporting anyway.

aasgit, 2017-07-11:
@yousong should I file a bug on flyspray?

yousong, 2017-07-11:
No need.  The commit message will refer to this issue report, though it's indeed in the wrong tracker...  The patch was just posted to lede-dev mailing list and if all goes well I will push it to the opkg repo, then update the lede master branch.